ptrkrysik/gr-gsm #365: Corrupt decoding

topdog6945, 2018-01-01:
I've been experiencing issues decoding cfiles so I downloaded the 'test_data' repository and run the command in the wiki: grgsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s $((100000000/174)) -a 725 -m SDCCH8 -t 1

[image]

i get SMS instead of System Information Type 5 messages. Same results on Kali and Parrot OS.

velichkov, 2018-01-02:
Hi @topdog6945,

Which version of Wireshark are you using and could you provide the pcap file?

velichkov, 2018-01-02:
Hi @topdog6945 ,

The difference between what you have decoded and what is shown in the wiki most probably comes from different wireshark versions . From https://code.wireshark.org/review/#/c/21767/

 > LAPDm: Support the LAPDm B4 Frame Format
The B4 Frame Format is used on the downlink SACCH and has no length field.

Wireshark versions after this change (>= v2.4.0) decode frame 2 as SMS and not as  "System Information Type 5" because of the length byte. Right now I can't say whether this change is correct or not.

topdog6945, 2018-01-04:
Thanks. It decodes the si5 message correctly with an older version of wireshark.

velichkov, 2018-01-04:
This bug was already reported in wireshark https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=14105

As this is not a gr-gsm problem, please close this issue.

closed 2018-01-04 by topdog6945