ptrkrysik/gr-gsm #367: problem when deocoding hopping SDCCH8

moemusaa, 2018-01-13:
Hi , everyone 
i have a problem for decoding hopping SDCCH8 channel.

first listen to BCCH C0 i have this

[image]

from immediate assignment : 
MAIO : 0
HSN   : 25
TSEQ : 2
Timeslot : 2

and From SI type 1 , i found List of ARFCNs = 648 673 676 683

then i configure Cx blocker as previous information 

[image]

but when i run the flow graph the output is nothing 

[image]

can any one tell me why this happen , why  i cant see any usfel information
" sometimes i have just one or two useful information such as SI type 5 "

moemusaa, 2018-01-15:
I solve the first problem , by adding main ARFCN " which is 648 " into Cx block inside MA .
i decrypt the SDCCH  , i have CM service request with Sender TMSI , " SMS CP-ACK" ,  and msg label as connect 

[image]

the problem here i cant find the SMS , also why this "malformed packet" always show 

please help :)

velichkov, 2018-01-15:
Hi @moemusaa,

Try with wireshark version 2.2.6 or older. See https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=14105

moemusaa, 2018-01-16:
thank @velichkov for replay
But when i try to decode non hopping channel its going fine with me , and i can see SMS CP-DATA with no problem
the problem only appear in hopping channel

ptrkrysik, 2018-05-28:
@moemusaa - can you show your mobile allocation from immediate assignment command?

closed 2018-06-02 by ptrkrysik