ptrkrysik/gr-gsm #368: Unable to sniff DCS1800

imranrajjad, 2018-01-15:
I am using Ubuntu 16.4 LTS in a Vmware. I am able to use grgsm_livemon with GSM900 band but getting nothing on DCS1800.

```
Using kalibrate-rtl, My RTL-SDR Usb dongle is able to pick up
        chan: 562 (1815.2MHz + 30.717kHz)       power: 2044453.78
        chan: 563 (1815.4MHz + 15.690kHz)       power: 5252914.14
        chan: 568 (1816.4MHz + 31.510kHz)       power: 1711955.07
        chan: 875 (1877.8MHz + 22.040kHz)       power: 1791337.37
        chan: 881 (1879.0MHz - 27.201kHz)       power: 2543330.99
```

and also from PCS

```
        chan: 526 (1933.0MHz + 11.459kHz)       power: 1242186.94
        chan: 531 (1934.0MHz + 27.453kHz)       power: 1060618.44
        chan: 544 (1936.6MHz - 9.228kHz)        power: 1015822.19
        chan: 664 (1960.6MHz + 32.379kHz)       power: 1405732.06
        chan: 680 (1963.8MHz - 35.678kHz)       power: 5036392.35

```

however the grgsm_scanner returns nothing neither does grgsm_livemon.

doing an rtl_test on my device gives

```
Using device 1: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Supported gain values (5): -9.9 -4.0 7.1 17.9 19.2 
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
```

Also my dongle is using indoor GSM antennas.

regards,
Imran

moemusaa, 2018-01-15:
try to use grgsm_scanner instead of kal 
also try to figure the frequencies using gqrx to make sure there are DCS1800 in air

imranrajjad, 2018-01-15:
thanks for the reply

grgsm_scanner shows nothing..even on GSM900 band.

I am able to view DCS1800 in SDRSharp. I cannot use GQRX because for some reason it will not tune above 950Mhz

regards,
Imran

imranrajjad, 2018-01-15:
[image]

moemusaa, 2018-01-15:
im also have this problem with grgsm_scanner , but after update the project " download it and build it again" it work fine with me

imranrajjad, 2018-01-15:
installed from a fresh git clone yesterday

ptrkrysik, 2018-01-16:
@imranrajjad, thanks for reporting the problem. My first suspect here is
automatic frequency offset control. I already found that there is no
callback to set carrier frequency in gnuradio-companion xml file
describing clock_offset_control block. Carrier frequency is used in the
computation of the offset so its wrong value results with wrong offset
value.

This explains why there is problem with grgsm_livemon.

But this probably doesn't explain why there is problem in grgsm_scanner
as it doesn't use gnuradio-companion generated code. Maybe the carrier
freq value is not update there for some other reason.

imranrajjad, 2018-01-16:
true that, I have gone through the grgsm_scanner python code and was not able to find anything. What is the default de-modulator being used for gsm traffic in case of DCS1800? GMSK?

regards,
Imran

ptrkrysik, 2018-01-17:
Everything is the same as on GSM900, only the frequency is different. So
I think that it might be that carrier frequency is not set somewhere
where it should be.

imranrajjad, 2018-01-17:
would it be possible for you to review a binary file of samples collected through rtl_sdr on the dcs frequency?

ptrkrysik, 2018-01-30:
I think I found the bug in the grgsm_livemon. If you can please install
new version from most recent development branch and check if it works
for you.

Regarding grgsm_scanner I don't know yet where the problem is (I have to
admit I wasn't looking very actively for it yet), but recording of the
signal might not help in the debugging here because of the dynamic
nature of the program.

imranrajjad, 2018-01-30:
I will definitely check once I upgrade my SDR, currently it occurred to me that my SDR does not sample above 2M(aliexpress page said It samples at 4.3M) which is not enough for DCS and PCS frequencies.

ptrkrysik, 2018-01-30:
@imranrajjad, from your post it seems you are mixing sampling frequency
with carrier frequency (2MHz seems to be sampling frequency and DCS,PCS
are frequency bands).

Do you maybe have one of the dongles that don't work on carrier
frequencies higher than ~1.7GHz?

imranrajjad, 2018-01-31:
The one is have works from 10khz to 950Mhz and 1.7Ghz to 2.1Ghz (as per SDR Sharp). However upon running rtl_test, the max samples per second are not good enough for frequencies above 1Ghz.

Keskebeu, 2018-02-21:
Work it out with correcting ppm and ppm offset. I was able to capture plenty with by samply changing the configurations.

With my b200 i was able to set both at 40 and I was sniffing from far away. Depending on what receiver you are working with you should play with that and see the spectrum changing in livemon. You don't need to use scanning at all. Just saying

closed 2018-05-28 by ptrkrysik