ptrkrysik/gr-gsm #369: Timeslot in Wireshark after decoding in grgsm_decode differs with go.sh

kent200786, 2018-01-18:
Hi. When I decode .cfile in grgsm_decode, in Wireshark I see SI5 frame in Timeslot 1. After application of a script go.sh with the parameter 1S of this frame I don't see, but when i applied script go.sh on various timeslots, the necessary frame has been found in other timeslot.
In what can be a problem?
Debian 8.5, RTL-SDR, Gnuradio 3.7.5-5, Gr-gsm make with Pybombs

velichkov, 2018-04-14:
Hi @kent200786,

> After application of a script go.sh with the parameter 1S of this frame I don't see, but when i applied script go.sh on various timeslots, the necessary frame has been found in other timeslot.

In `gr-gsm` project there is no `go.sh` script so where do you get it and what does this script do?

> In what can be a problem?

Could you share the .cfile and the exact commands that you have used?

ptrkrysik, 2018-06-02:
I'm closing the issue due to lack of information needed to reproduce what was observed by @kent200786 .

closed 2018-06-02 by ptrkrysik