ptrkrysik/gr-gsm #374: installation problem

BushKape, 2018-02-12:
When entering a command for the installation, it gives errors
`sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default[sudo] пароль для sunduk: 
Попробуйте ещё раз.
[sudo] пароль для sunduk: 
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
Alias `default_prx' already exists, overwrite Y/[N]? y
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Fetcher - INFO - Already fetched: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gnuradio/build
Configuring: (100%) [=======================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.2.0-8ubuntu3) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   chrono
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.10.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- 
-- Configuring testing-support support...
--   Dependency CPPUNIT_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
CMake Warning at CMakeLists.txt:380 (message):
  

  Warning: CppUnit version is 1.14.0 which requires C++11 for building.
  Trying to set CMake internally to use C++11 ...

-- 
-- Configuring VOLK support...
--   VOLK submodule is not checked out.
--   To check out the VOLK submodule, use:
--     git pull --recurse-submodules=on
--     git submodule update
--   External VOLK disabled.
--   Override with -DENABLE_INTERNAL_VOLK=ON/OFF
-- 
CMake Error at CMakeLists.txt:450 (message):
  VOLK required but not found.

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.`
ubuntu 17.10

velichkov, 2018-02-12:
Hi @BushKape,

Try installing libvolk1-dev

```
apt-get install libvolk1-dev
```

BushKape, 2018-02-13:
hi @velichkov 
after executing the command: libvolk1-dev the package is already installed the newest version (1.3-1build2)
`PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
Alias `default_prx' already exists, overwrite Y/[N]? y
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Fetcher - INFO - Already fetched: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gnuradio/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.2.0-8ubuntu3) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   chrono
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.10.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- 
-- Configuring testing-support support...
--   Dependency CPPUNIT_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
CMake Warning at CMakeLists.txt:380 (message):
  

  Warning: CppUnit version is 1.14.0 which requires C++11 for building.
  Trying to set CMake internally to use C++11 ...

-- 
-- Configuring VOLK support...
--   VOLK submodule is not checked out.
--   To check out the VOLK submodule, use:
--     git pull --recurse-submodules=on
--     git submodule update
--   External VOLK disabled.
--   Override with -DENABLE_INTERNAL_VOLK=ON/OFF
-- 
CMake Error at CMakeLists.txt:450 (message):
  VOLK required but not found.

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.`
nothing changed

velichkov, 2018-02-13:
Hi @BushKape,

Check that `/usr/local/src/gnuradio/volk` directory is not empty and if it is execute
```
cd /usr/local/src/gnuradio/
git submodule update --recursive
```
or as stated in the above output
```
git pull --recurse-submodules=on
git submodule update
```
As this is a gnuradio installation problem consider reporting it to the [gnuradio mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) as well

velichkov, 2018-02-13:
I was able to reproduce this issue and a [fix is already available](https://github.com/gnuradio/pybombs/commit/5a340b45) but a version with this fix is still not released so you need to update it manually from the github repo.

```
pip install git+https://github.com/gnuradio/pybombs.git
```

See also https://github.com/gnuradio/pybombs/issues/460

BushKape, 2018-02-13:
@velichkov  did as described in the topic  gnuradio/pybombs#460, it did not help me.
updated sudo pip install git+https://github.com/gnuradio/pybombs.git

> `The directory '/home/sunduk/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/sunduk/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting git+https://github.com/gnuradio/pybombs.git
  Cloning https://github.com/gnuradio/pybombs.git to /tmp/pip-7CGCqN-build
  Requirement already satisfied (use --upgrade to upgrade): PyBOMBS==2.3.3a0 from git+https://github.com/gnuradio/pybombs.git in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: future in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: requests in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: ruamel.yaml in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: six in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: idna<2.7,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: urllib3<1.23,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: ruamel.ordereddict; platform_python_implementation == "CPython" and python_version <= "2.7" in /usr/local/lib/python2.7/dist-packages (from ruamel.yaml->PyBOMBS==2.3.3a0)`

> 
errors remain
cd /usr/local/src/gnuradio/
git submodule update --recursive

> 
fatal: submodule «volk» could not add alternative: path «/home/sunduk/.pybombs/gitcache/modules/volk/» does not exist
Could not clone «volk». A retry is planned
fatal:  submodule «volk»  could not add alternative: path «/home/sunduk/.pybombs/gitcache/modules/volk/» does not exist
Could not clone  «volk» on the second attempt, cancellation

>

velichkov, 2018-02-13:
> The directory '/home/sunduk/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.

How have you installed pybombs?
Try without sudo, e.g. only `pip install git+https://github.com/gnuradio/pybombs.git`

BushKape, 2018-02-13:
@velichkov 
pip install git+https://github.com/gnuradio/pybombs.git

> 
pip install git+https://github.com/gnuradio/pybombs.git
Collecting git+https://github.com/gnuradio/pybombs.git
  Cloning https://github.com/gnuradio/pybombs.git to /tmp/pip-SY4W9O-build
Collecting future (from PyBOMBS==2.3.3a0)
  Downloading future-0.16.0.tar.gz (824kB)
    100% |████████████████████████████████| 829kB 707kB/s 
Collecting requests (from PyBOMBS==2.3.3a0)
  Downloading requests-2.18.4-py2.py3-none-any.whl (88kB)
    100% |████████████████████████████████| 92kB 1.3MB/s 
Collecting ruamel.yaml (from PyBOMBS==2.3.3a0)
  Downloading ruamel.yaml-0.15.35-cp27-cp27mu-manylinux1_x86_64.whl (534kB)
    100% |████████████████████████████████| 542kB 887kB/s 
Collecting setuptools (from PyBOMBS==2.3.3a0)
  Downloading setuptools-38.5.1-py2.py3-none-any.whl (489kB)
    100% |████████████████████████████████| 491kB 894kB/s 
Collecting six (from PyBOMBS==2.3.3a0)
  Downloading six-1.11.0-py2.py3-none-any.whl
Collecting idna<2.7,>=2.5 (from requests->PyBOMBS==2.3.3a0)
  Downloading idna-2.6-py2.py3-none-any.whl (56kB)
    100% |████████████████████████████████| 61kB 2.1MB/s 
Collecting urllib3<1.23,>=1.21.1 (from requests->PyBOMBS==2.3.3a0)
  Downloading urllib3-1.22-py2.py3-none-any.whl (132kB)
    100% |████████████████████████████████| 133kB 1.2MB/s 
Collecting certifi>=2017.4.17 (from requests->PyBOMBS==2.3.3a0)
  Downloading certifi-2018.1.18-py2.py3-none-any.whl (151kB)
    100% |████████████████████████████████| 153kB 1.2MB/s 
Collecting chardet<3.1.0,>=3.0.2 (from requests->PyBOMBS==2.3.3a0)
  Downloading chardet-3.0.4-py2.py3-none-any.whl (133kB)
    100% |████████████████████████████████| 143kB 1.7MB/s 
Collecting ruamel.ordereddict; platform_python_implementation == "CPython" and python_version <= "2.7" (from ruamel.yaml->PyBOMBS==2.3.3a0)
  Downloading ruamel.ordereddict-0.4.13-cp27-cp27mu-manylinux1_x86_64.whl (99kB)
    100% |████████████████████████████████| 102kB 1.9MB/s 
Building wheels for collected packages: future
  Running setup.py bdist_wheel for future ... done
  Stored in directory: /home/sunduk/.cache/pip/wheels/c2/50/7c/0d83b4baac4f63ff7a765bd16390d2ab43c93587fac9d6017a
Successfully built future
Installing collected packages: future, idna, urllib3, certifi, chardet, requests, ruamel.ordereddict, ruamel.yaml, setuptools, six, PyBOMBS
  Running setup.py install for PyBOMBS ... done
Successfully installed PyBOMBS-2.3.3a0 certifi-2018.1.18 chardet-3.0.4 future-0.16.0 idna-2.6 requests-2.18.4 ruamel.ordereddict-0.4.13 ruamel.yaml-0.15.35 setuptools-38.5.1 six-1.11.0 urllib3-1.22

> 
installed pypombs as here https://github.com/ptrkrysik/gr-gsm/wiki/Installation

velichkov, 2018-02-13:
Ok, now with the latest pybombs version does the `sudo pybobms prefix init ...` fails with the same error?

BushKape, 2018-02-13:
@velichkov 

> `sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
Alias `default_prx' already exists, overwrite Y/[N]? y
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Fetcher - INFO - Already fetched: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gnuradio/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.2.0-8ubuntu3) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   chrono
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.10.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- 
-- Configuring testing-support support...
--   Dependency CPPUNIT_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
CMake Warning at CMakeLists.txt:380 (message):
  

  Warning: CppUnit version is 1.14.0 which requires C++11 for building.
  Trying to set CMake internally to use C++11 ...

-- 
-- Configuring VOLK support...
--   VOLK submodule is not checked out.
--   To check out the VOLK submodule, use:
--     git pull --recurse-submodules=on
--     git submodule update
--   External VOLK disabled.
--   Override with -DENABLE_INTERNAL_VOLK=ON/OFF
-- 
CMake Error at CMakeLists.txt:450 (message):
  VOLK required but not found.

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
`

> 
pypombs installed after the update

velichkov, 2018-02-13:
> PyBOMBS.Fetcher - INFO - Already fetched: gnuradio

It seems you have to manually clean up before proceeding further (use sudo if needed)
```
rm /usr/local/src/gnuradio -rf
pybombs -v fetch gnuradio
```
P.S.
When posting logs please enclose them in ``` as described [here](https://help.github.com/articles/creating-and-highlighting-code-blocks/)

BushKape, 2018-02-13:
@velichkov it worked, thanks a lot.
```sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default
[sudo] пароль для sunduk: 
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
Alias `default_prx' already exists, overwrite Y/[N]? y
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Configuring: (100%) [=======================================================================================================================]
Building:    (100%) [=======================================================================================================================]
Installing:  (100%) [=======================================================================================================================]
PyBOMBS.install_manager - INFO - Installation successful.```

but there was another mistake

```sudo pybombs install gr-gsm
[sudo] пароль для sunduk: 
Попробуйте ещё раз.
[sudo] пароль для sunduk: 
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gr-gsm
   |
   \- gr-osmosdr
      |
      +- airspy
      |
      +- hackrf
      |
      \- soapysdr
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: soapysdr
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/soapysdr/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
-- PYTHONINTERP_FOUND: TRUE - 2.7.14
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.14
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7
-- PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
[... 55 lines cut ...]
-- ######################################################
-- 
 * Library, runtime library v0.7.0-g2a266d60
 * Tests, library unit tests
 * Docs, doxygen documentation
 * Python, python bindings v2.7.14
 * Python3, python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 

-- SoapySDR version: v0.7.0-g2a266d60
-- ABI/so version: v0.7
-- Install prefix: /usr/local
-- Configuring incomplete, errors occurred!
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package soapysdr:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package soapysdr. Aborting.```

After reading, I realized this is a build problem. So many topics are devoted to it, is there a solution?

velichkov, 2018-02-14:
> So many topics are devoted to it, is there a solution?

Read the whole output and try to find the most specific error message, do not just take the first or last one.

>   install TARGETS given no LIBRARY DESTINATION for module target
  "_SoapySDR3".

Now when you google this you are going to find at least 2 other github issues.

https://github.com/pothosware/SoapySDR/issues/132
https://github.com/gnuradio/gr-recipes/issues/91

The first one contains 2 possible solutions and the first seems to be the easiest. 

> add -DENABLE_PYTHON3=OFF to the cmake arguments. 

You need to add it to the `~/.pybombs/recipes/gr-recipes/sopaysdr.lwr` 

```patch
diff --git a/soapysdr.lwr b/soapysdr.lwr
index 8e5426290c..0713e14648 100644
--- a/soapysdr.lwr
+++ b/soapysdr.lwr
@@ -25,3 +25,5 @@ gitbranch: master
 inherit: cmake
 # Let's always build this from source, not binaries
 source: git+git://github.com/pothosware/SoapySDR.git
+vars:
+  config_opt: ' -DENABLE_PYTHON3=OFF '
```

BushKape, 2018-02-14:
@velichkov when fully added
```
PyBOMBS.get_recipe - ERROR - Error fetching recipe `soapysdr':
Error loading /home/sunduk/.pybombs/recipes/gr-recipes/soapysdr.lwr: while scanning a simple key
  in "<byte string>", line 24, column 1:
    diff --git a/soapysdr.lwr b/soap ... 
    ^ (line: 24)
could not find expected ':'
  in "<byte string>", line 25, column 1:
    index 8e5426290c..0713e14648 100644
    ^ (line: 25)
```
when adding only a trip 
```
+vars:
+  config_opt: ' -DENABLE_PYTHON3=OFF '
```
```
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package soapysdr:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package soapysdr. Aborting.
```

velichkov, 2018-02-14:
> when fully added

What exactly you have done?

> when adding only a trip
>```
>+vars:
>+  config_opt: ' -DENABLE_PYTHON3=OFF '
>```

You need to add these two lines (without the plus sign) at the bottom of the file.
And like before you probably have to cleanup before executing `pybombs install`

```
rm -rf /usr/local/src/soapysdr/build
```
or
```
rm -rf /usr/local/src/soapysdr
pybombs inv soapysdr none

BushKape, 2018-02-14:
@velichkov thanks helped

velichkov, 2018-02-14:
> @velichkov thanks helped

Cool, then please close the issue.

BushKape, 2018-02-14:
closes #374

BushKape, 2018-02-15:
@velichkov I apologize, but I again have problems, I'm trying to go to the folder for /usr/local/src/gr-gsm/apps, command 
```
cd /usr/local/src/gr-gsm/apps 
bash: cd: /usr/local/scr/gr-gsm: No such file or directory
```
I do this for
```
sudo gnuradio-companion grgsm_livemon.grc
```
I'm trying to execute a command
```
grgsm_livemon
grgsm_livemon:command not found
```
use usrp n210
established drivers and gnuradio, as here https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux
what am I doing wrong again?

velichkov, 2018-02-16:
>I apologize, but I again have problems,

The better approach is to open a new github issue for the new problem.

> I'm trying to go to the folder for /usr/local/src/gr-gsm/apps, command
>```
> cd /usr/local/src/gr-gsm/apps 
> bash: cd: /usr/local/scr/gr-gsm: No such file or directory
>```

I don't know why this directory does not exist on your system. If the `sudo pybombs install gr-gsm` has finished successfully then this directory should exist.

>I do this for
>```
> sudo gnuradio-companion grgsm_livemon.grc
>```

You could clone the gr-gsm repository in any other directory and open the grgsm_livemon.grc from there.
```
cd $HOME
git clone https://github.com/ptrkrysik/gr-gsm.git
cd gr-gsm/apps/
gnuradio-companion grgsm_livemon.grc
```
> I'm trying to execute a command
>```
> grgsm_livemon
> grgsm_livemon:command not found
>```

Have you executed `sudo ldconfig` after `sudo pybombs install gr-gsm` has finished as described in the [wiki](https://github.com/ptrkrysik/gr-gsm/wiki/Installation). The grgsm_livemon should be in `/usr/local/bin` so you should be able to execute it directly if `/usr/local/bin` is in your $PATH environment variable or you could execute it with as `/usr/local/bin/grgsm_livemon` or

```
cd /usr/local/bin/
./grgsm_livemon
```

BushKape, 2018-02-16:
@velichkov after cloning
```
gnuradio-companion grgsm_livemon.grc
<<< Welcome to GNU Radio Companion 3.7.12git-329-gf783dd54 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/sunduk/gr-gsm/apps/grgsm_livemon.grc"
Block key "gsm_bcch_ccch_demapper" not found
Block key "gsm_clock_offset_control" not found
Block key "gsm_control_channels_decoder" not found
Block key "gsm_control_channels_decoder" not found
Block key "gsm_decryption" not found
Block key "gsm_input" not found
Block key "gsm_message_printer" not found
Block key "gsm_receiver" not found
Block key "gsm_sdcch8_demapper" not found
Block key "rtlsdr_source" not found
```

BushKape, 2018-02-16:
After the installation, the command executed
```
sudo ldconfig
```

velichkov, 2018-02-16:
Try to reinstall gr-gsm by executing `sudo pybombs rebuild gr-gsm` and `sudo ldconfig` and if still fails provide the output of
```
cat $HOME/.pybombs/config.yml
ls -al /usr/local/src/
pybombs prefix info
pybombs inv
find /usr/local/share/gnuradio/grc/ -name "*gsm*"
```
Are you able to start it using the full path `/usr/local/bin/grgsm_livemon`?

BushKape, 2018-02-16:
@velichkov reinstall did not help
 ```
cat $HOME/.pybombs/config.yml
!!omap
- config:
    git-cache: /home/sunduk/.pybombs/gitcache
    elevate_pre_args:
    - sudo
    - -H
    makewidth: 4
    packagers: apt,pymod,pip,pkgconfig,cmd
    default_prefix: default_prx
- recipes:
    gr-etcetera: git+https://github.com/gnuradio/gr-etcetera.git
    gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
- prefix_aliases:
    default_prx: /usr/local
    gr-gsm: /usr/local
sunduk@sunduk:~$ ls -al /usr/local/src/
итого 48
drwxr-xr-x 12 root root 4096 фев 14 05:42 .
drwxr-xr-x 13 root root 4096 фев 13 22:55 ..
drwxr-xr-x 13 root root 4096 фев 12 20:06 apache-thrift
drwxr-xr-x 11 root root 4096 фев 13 23:06 bladeRF
drwxr-xr-x 33 root root 4096 фев 13 21:55 gnuradio
drwxr-xr-x 13 root root 4096 фев 13 23:05 gr-iqbal
drwxr-xr-x 12 root root 4096 фев 12 21:30 libosmocore
drwxr-xr-x  8 root root 4096 фев 13 23:04 libosmo-dsp
drwxr-xr-x  8 root root 4096 фев 13 23:03 osmo-sdr
drwxr-xr-x  8 root root 4096 фев 13 23:01 rtl-sdr
drwxr-xr-x 15 root root 4096 фев 14 05:42 soapysdr
drwxr-xr-x  9 root root 4096 фев 12 20:10 uhd
sunduk@sunduk:~$ pybombs prefix info
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
[Default Prefix]: default_prx 
Available Prefixes :
gr-gsm - /usr/local
default_prx - /usr/local
sunduk@sunduk:~$ pybombs inv
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
Showing package state:
uhd:                installed
apache-thrift:      installed
libosmocore:        installed
gnuradio:           installed
rtl-sdr:            installed
osmo-sdr:           installed
libosmo-dsp:        installed
gr-iqbal:           installed
bladeRF:            installed
soapysdr:           fetched
sunduk@sunduk:~$ find /usr/local/share/gnuradio/grc/ -name "*gsm*"
/usr/local/share/gnuradio/grc/blocks/vocoder_gsm_fr_encode_sp.xml
/usr/local/share/gnuradio/grc/blocks/vocoder_gsm_fr_decode_ps.xml
```
no such file or directory
```
/usr/local/bin/grgsm_livemon
```

velichkov, 2018-02-16:
The above output clearly indicates that the gr-gsm is not installed and the installation probably fails when compiling the soapysdr. Provide the full output of `sudo pybombs -v install gr-gsm`

BushKape, 2018-02-16:
@velichkov 
 ```
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/sunduk/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: ordereddict([('git-cache', '/home/sunduk/.pybombs/gitcache'), ('elevate_pre_args', ['sudo', '-H']), ('makewidth', 4), ('packagers', 'apt,pymod,pip,pkgconfig,cmd'), ('default_prefix', 'default_prx')])
PyBOMBS.ConfigManager - DEBUG - Template directory: /home/sunduk/.local/lib/python2.7/site-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/sunduk/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias `default_prx'.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/usr/local)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /usr/local/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /usr/local/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /usr/local/src
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /usr/local/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /usr/local/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /usr/local/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/usr/local/.pybombs/recipes', '/home/sunduk/.pybombs/recipes/gr-recipes', '/home/sunduk/.pybombs/recipes/gr-etcetera', '/home/sunduk/.local/lib/python2.7/site-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/sunduk/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/sunduk/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 178 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/sunduk/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/sunduk/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 97 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/sunduk/.local/lib/python2.7/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/sunduk/.local/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.PackageManager - DEBUG - pymod is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['apt', 'pymod', 'pip', 'pkgconfig', 'cmd']
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.PackageManager - DEBUG - Checking if package gr-gsm is installable...
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Checking if package gr-gsm is installed...
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is not installed.
PyBOMBS.install_manager - DEBUG - Testing binary install for package gr-gsm.
PyBOMBS.PackageManager - DEBUG - install(gr-gsm, install_type=binary, static=False)
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Returning no packagers -- package is requesting source build, but binary build is requested.
PyBOMBS.PackageManager - DEBUG - Checking if package gnuradio is installable...
PyBOMBS.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Checking if package gnuradio is installed...
PyBOMBS.PackageManager - DEBUG - Package gnuradio is requesting a source build.
[... cut ...]
  File "/home/sunduk/.local/lib/python2.7/site-packages/pybombs/package_manager.py", line 283, in _std_package_operation
    result = getattr(pkgr, operation)(rec, **kwargs)
  File "/home/sunduk/.local/lib/python2.7/site-packages/pybombs/packagers/source.py", line 109, in install
    fail_if_builddir_missing=update,
  File "/home/sunduk/.local/lib/python2.7/site-packages/pybombs/packagers/source.py", line 235, in run_build
    recipe.vars['builddir'] = builddir
TypeError: 'NoneType' object does not support item assignment
 ```

velichkov, 2018-02-16:
Make sure that the content of `~/.pybombs/recipes/gr-recipes/sopaysdr.lwr` is  exactly 
```
#
# This file is part of PyBOMBS
#
# PyBOMBS is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3, or (at your option)
# any later version.
#
# PyBOMBS is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with PyBOMBS; see the file COPYING.  If not, write to
# the Free Software Foundation, Inc., 51 Franklin Street,
# Boston, MA 02110-1301, USA.
#

category: hardware
depends:
- swig
description: Vendor and platform neutral SDR support library
gitbranch: master
inherit: cmake
# Let's always build this from source, not binaries
source: git+git://github.com/pothosware/SoapySDR.git
vars:
  config_opt: ' -DENABLE_PYTHON3=OFF '
```
and then clean up and install again
```
sudo rm -rf /usr/local/src/soapysdr/build
sudo pybombs -v install gr-gsm
```

BushKape, 2018-02-16:
@velichkov Now everything is established, but the problem is now in another
 ```
/usr/local/src/gr-gsm/apps$ sudo gnuradio-companion grgsm_livemon.grc
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/usr/local/bin/gnuradio-companion", line 97, in <module>
    check_gtk()
  File "/usr/local/bin/gnuradio-companion", line 64, in check_gtk
    die(err, "Failed to initialize GTK. If you are running over ssh, "
  File "/usr/local/bin/gnuradio-companion", line 42, in die
    import gtk
  File "/usr/lib/python2.7/dist-packages/gtk-2.0/gtk/__init__.py", line 69, in <module>
    _init()
  File "/usr/lib/python2.7/dist-packages/gtk-2.0/gtk/__init__.py", line 57, in _init
    warnings.warn(str(e), _gtk.Warning)
gtk.GtkWarning: could not open display
 ```

velichkov, 2018-02-16:
> /usr/local/src/gr-gsm/apps$ sudo gnuradio-companion grgsm_livemon.grc

Try without sudo

BushKape, 2018-02-17:
@velichkov works without sudo
 ```
gnuradio-companion grgsm_livemon.grc
<<< Welcome to GNU Radio Companion 3.7.12git-329-gf783dd54 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks

Loading: "/usr/local/src/gr-gsm/apps/grgsm_livemon.grc"
/usr/local/lib/python2.7/dist-packages/scipy/linalg/basic.py:17: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._solve_toeplitz import levinson
/usr/local/lib/python2.7/dist-packages/scipy/linalg/__init__.py:202: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._decomp_update import *
/usr/local/lib/python2.7/dist-packages/scipy/linalg/__init__.py:202: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._decomp_update import *
/usr/local/lib/python2.7/dist-packages/scipy/linalg/__init__.py:202: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._decomp_update import *
/usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:640: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._ufuncs import *
/usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:640: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._ufuncs import *
/usr/local/lib/python2.7/dist-packages/scipy/special/_ellip_harm.py:7: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._ellip_harm_2 import _ellipsoid, _ellipsoid_norm
/usr/local/lib/python2.7/dist-packages/scipy/sparse/lil.py:16: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from . import _csparsetools
/usr/local/lib/python2.7/dist-packages/scipy/optimize/_trlib/__init__.py:1: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._trlib import TRLIBQuadraticSubproblem
/usr/local/lib/python2.7/dist-packages/scipy/optimize/_minimize.py:32: ImportWarning: Not importing directory '/usr/local/lib/python2.7/dist-packages/scipy/optimize/lbfgsb': missing __init__.py
  from .lbfgsb import _minimize_lbfgsb
/usr/local/lib/python2.7/dist-packages/scipy/optimize/_numdiff.py:8: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._group_columns import group_dense, group_sparse
/usr/local/lib/python2.7/dist-packages/scipy/interpolate/_bsplines.py:9: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from . import _bspl
/usr/local/lib/python2.7/dist-packages/scipy/spatial/__init__.py:94: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from .ckdtree import *
/usr/local/lib/python2.7/dist-packages/scipy/spatial/__init__.py:95: ImportWarning: Not importing directory '/usr/local/lib/python2.7/dist-packages/scipy/spatial/qhull': missing __init__.py
  from .qhull import *
/usr/local/lib/python2.7/dist-packages/scipy/spatial/__init__.py:95: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from .qhull import *
/usr/local/lib/python2.7/dist-packages/scipy/spatial/_spherical_voronoi.py:18: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from . import _voronoi
/usr/local/lib/python2.7/dist-packages/scipy/spatial/distance.py:121: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from . import _hausdorff
/usr/local/lib/python2.7/dist-packages/scipy/ndimage/measurements.py:36: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from . import _ni_label
linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.003.HEAD-0-gef157678

>>> Done

Generating: '/tmp/grgsm_livemon.py'

Executing: /usr/bin/python2 -u /tmp/grgsm_livemon.py

linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.003.HEAD-0-gef157678

gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-329-gf783dd54
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes
-- Detecting internal GPSDO.... Found an internal GPSDO: Jackson-Labs, FireFly , Firmware Rev 0.929

UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam
-- Setting references to the internal GPSDO
-- Using subdev spec 'A:0'.

 ```

BushKape, 2018-02-17:
@velichkov I give the command
 ```
 wireshark -k -f udp -Y gsmtap -i lo
 ```
Decoding does not occur

velichkov, 2018-02-19:
You need to run both `grgsm_livemon` and `wireshark` in parallel so you need to start wireshark in one console and keep it running, open another console and start `grgsm_livemon` or `grgsm_decode`

See also
https://github.com/ptrkrysik/gr-gsm/wiki/Usage#analyzing-gsm-messages-in-the-wireshark
https://osqa-ask.wireshark.org/questions/7976/wireshark-setup-linux-for-nonroot-user
https://github.com/ptrkrysik/gr-gsm/wiki/Usage
https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To

ptrkrysik, 2018-02-20:
@BushKape, from the gr-gsm project's readme: "Mailing list is a place for general discussions, questions about the usage and **installation**" . Please use the mailing list to ask about your problems with gr-gsm installation.

@velichkov, many thanks for the support you provide to the gr-gsm users. I appreciate it a lot.

closed 2018-02-20 by ptrkrysik

RadCet, 2018-05-22:
@BushKape Can i ask u some question? Did u success running gr-gsm in N210? And i also want to buy a N210 product and run gr-gsm on it but when i mail Ettus they said i should buy N210 with sbx daughterboard. Did it really need ?