ptrkrysik/gr-gsm #375: AttributeError: 'module' object has no attribute 'util'

vutuyetmai130398, 2018-02-21:
I am studying GSM sniffing and want to go the grgsm_livemon on ubuntu terminal and I have some problems. That is:
~$ grgsm_livemon
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-release

Traceback (most recent call last):
File "/usr/local/bin/grgsm_livemon", line 380, in 
main()
File "/usr/local/bin/grgsm_livemon", line 368, in main
tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
File "/usr/local/bin/grgsm_livemon", line 66, in init
qtgui.util.check_set_qss()
AttributeError: 'module' object has no attribute 'util'

How to involve them?
Please help me!
Thanks,

velichkov, 2018-02-21:
Hi @vutuyetmai130398,

Which version of Ubuntu you have and how did you install `gnuradio` and `gr-gsm` - using the pybombs method, from the repository packages or manually?

Try to open `apps/grgsm_livemon.grc` file in `gnuradio-companion` and then click Run -> Execute (F6).

vutuyetmai130398, 2018-02-21:
I use ubuntu 16.04.
I install gnuradio and gr-gsm by the pybombs method and manually. But both of them it is not be executed when typing grgsm_livemon?
So i dont know how to fix them?
I sucessed in opening apps/grgsm_livemon.grc

velichkov, 2018-02-21:
> and manually. 

Have you installed the latest `master` or some other version?

> But both of them it is not be executed when typing grgsm_livemon?

Same or different errors?

> I sucessed in opening apps/grgsm_livemon.grc

If you see the GUI you could run it this way, it's the same app run in two different ways.

vutuyetmai130398, 2018-02-21:
I install gr-gsm by the pybomb method and from here https://github.com/ptrkrysik/gr-gsm/wiki/Manual-compilation-and-installation. But the lastest i install by the pybombs method.
Both of them have the same errors.

velichkov, 2018-02-21:
> But the lastest i install by the pybombs method.
> Both of them have the same errors.

Ok, in the terminal go to the directory that contains the `grgsm_livemon.grc` file, execute the following commands and provide the full output

```
git describe
git branch
git status
git diff
diff -u /usr/local/bin/grgsm_livemon apps/grgsm_livemon
diff -u apps/grgsm_livemon apps/grgsm_livemon.py
```

vutuyetmai130398, 2018-02-22:
Thanks Mr.velichkov . I will try.

velichkov, 2018-03-23:
Hi @mhgue,

> git describe => v0.41.1-98-gc097ce7
> Ubuntu 16.04.4 LTS

```
$dpkg -l gnuradio
ii  gnuradio                        3.7.9.1-2ubuntu1     amd64                GNU Radio Software Radio Toolkit
```

This gnuradio version is not compatible with the version used to generate the `grgsm_livemon` script and that's why it is failing.

Could you pull the latest changes and try with it. Your version is few commits behind the current `master` and `development` branches and does not contain my changes from #378. In the new version `grgsm_livemon` is regenerated automatically on every compile using the `gnuradio` on your system.

You could also try to regenerated it manually, in `gnuradio-companion` open the apps/grgsm_livemon.grc file and then click Run -> Execute (F6).

gronget, 2018-03-23:
hey
i could not also run grgsm_livemon through terminal but could run and execute it in gnuradio_companion... But i would love to run it directly on terminal....i happen to be having same version of gnuradio   3.7.9.1.................what version do you recommend to work fine

velichkov, 2018-03-23:
Hi @gronget,

> i happen to be having same version of gnuradio 3.7.9.1.................what version do you recommend to work fine

It should works with gnuradio 3.7.9.1, the script just needs to be regenerated. Could you try the latest `master` 7476868?

gronget, 2018-03-23:
is it possible to identify packets that are for a specific phone in wireshark......
was thinking of maybe by its tmsi but somehow the tmsi in encrypted after authentication...

gronget, 2018-03-23:
and the other question is about timeslots....iam kinda confused 
is a mobile phone assigned a specific timeslot for the entire time its attached to BTS and all the communication happens on that timslot

gronget, 2018-03-23:
i now have grgsm_livemon working fine after getting the latest    master 7476868

velichkov, 2018-03-23:
Hi @gronget,

> i now have grgsm_livemon working fine after getting the latest master 7476868

That's good.

> is it possible to identify packets that are for a specific phone in wireshark......
was thinking of maybe by its tmsi but somehow the tmsi in encrypted after authentication...

No an easy way as far as I know. 

> is a mobile phone assigned a specific timeslot for the entire time its attached to BTS and all the communication happens on that timslot

A timeslot is assigned per call.

Please post general questions to the [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

mhgue, 2018-03-24:
Hi @velichkov,
looks perfectly ok using the 7476868, loading in `gnuradio-companion` too.
Thanks for your commit.