ptrkrysik/gr-gsm #376: swig not work throught i have reinstall swig 2.0

vequeemnhe, 2018-02-21:
i have reinstall swig 2.0 debian but not work.... help i.
i install on Kali linux 2.0, GNU 3.7.11 and gcc version 7.2.0 (Debian 7.2.0-19) 
<code>[ 64%] Generating doxygen xml for grgsm_swig_doc docs
warning: Tag `SYMBOL_CACHE_SIZE' at line 289 of file `/root/Documents/Hacking/RF Hacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `SHOW_DIRECTORIES' at line 477 of file `/root/Documents/Hacking/RF Hacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `HTML_ALIGN_MEMBERS' at line 803 of file `/root/Documents/Hacking/RF Hacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `USE_INLINE_TREES' at line 956 of file `/root/Documents/Hacking/RF Hacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `XML_SCHEMA' at line 1149 of file `/root/Documents/Hacking/RF Hacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `XML_DTD' at line 1155 of file `/root/Documents/Hacking/RF Hacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: doxygen no longer ships with the FreeSans font.
You may want to clear or change DOT_FONTNAME.
Otherwise you run the risk that the wrong font is being used for dot generated graphs.
error: tag OUTPUT_DIRECTORY: Output directory `/root/Documents/Hacking/RFHacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs' does not exist and cannot be created
swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make:64: recipe for target 'swig/grgsm_swig_doc_swig_docs/xml/index.xml' failed
make[2]: *** [swig/grgsm_swig_doc_swig_docs/xml/index.xml] Error 1
CMakeFiles/Makefile2:602: recipe for target 'swig/CMakeFiles/grgsm_swig_swig_doc.dir/all' failed
make[1]: *** [swig/CMakeFiles/grgsm_swig_swig_doc.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
</code>
throught i have change version all or use "doxygen -u" but not work....
Thanks you very much

vequeemnhe, 2018-02-21:
i have clear "SYMBOL_CACHE_SIZE, SHOW_DIRECTORIES, HTML_ALIGN_MEMBERS, USE_INLINE_TREES, XML_SCHEMA, XML_DTD" but i get message 

> error: tag OUTPUT_DIRECTORY: Output directory `/root/Documents/Hacking/RFHacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs' does not exist and cannot be created
swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make:64: recipe for target 'swig/grgsm_swig_doc_swig_docs/xml/index.xml' failed
make[2]: *** [swig/grgsm_swig_doc_swig_docs/xml/index.xml] Error 1
CMakeFiles/Makefile2:602: recipe for target 'swig/CMakeFiles/grgsm_swig_swig_doc.dir/all' failed
make[1]: *** [swig/CMakeFiles/grgsm_swig_swig_doc.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

velichkov, 2018-02-21:
Hi @vequeemnhe,

> i install on Kali linux 2.0, GNU 3.7.11 and gcc version 7.2.0 (Debian 7.2.0-19)

Although you have recent versions of gnuradio and gcc, the Kali version is quite old. Consider upgrading to the latest Kali 2018.1 Rolling. 

> error: tag OUTPUT_DIRECTORY: Output directory `/root/Documents/Hacking/RFHacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs' does not exist and cannot be created

Try to manually create this directory, e.g `mkdir /root/Documents/Hacking/RFHacking/gr-gsm/build/swig/grgsm_swig_doc_swig_docs`

Which version of doxygen is installed `dpkg -l doxygen`?

You could also try to disable the documentation by running 
```
cmake -DENABLE_DOXYGEN=OFF ..
make
```

vequeemnhe, 2018-02-22:
first, i thank velichkov help me...
it work with command 

> cmake -DENABLE_DOXYGEN=OFF ..
make
thanks you very much.....

closed 2018-04-11 by ptrkrysik

zahirem, 2018-04-14:
how can i fix this?
[ 87%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_time_spec_t_get_system_time(PyObject*, PyObject*)’:
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:20234:34: error: ‘get_system_time’ is not a member of ‘uhd::time_spec_t’
       result = uhd::time_spec_t::get_system_time();
                                  ^~~~~~~~~~~~~~~
gr-uhd/swig/CMakeFiles/_uhd_swig.dir/build.make:70: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:13300: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all' failed
make[1]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

velichkov, 2018-04-14:
Hi @zahirem,

Please do not reuse closed issues. 
Please post to the [mailing list](mailto:gr-gsm@googlegroups.com) the above error messages and write which linux distribution and version you are using. See also https://github.com/ptrkrysik/gr-gsm#mailing-list

velichkov, 2018-04-14:
Hi @zahirem,
> ```
> /usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_time_spec_t_get_system_time(PyObject*, PyObject*)’: [...]

This turns out to be a known `gnuradio` issue, see  https://github.com/gnuradio/gnuradio/issues/1706 and https://lists.gnu.org/archive/html/discuss-gnuradio/2018-03/msg00265.html