ptrkrysik/gr-gsm #383: *** Error in `/usr/bin/python2': corrupted double-linked list: 0x01c22b08 ***

imranrajjad, 2018-03-20:
I am trying to get gr-gsm to work on Pi3. After following the complete documentation, I am unable to start grgsm_headless. Below is the error I get

```
pi@EL-raspberrypi:~ $ grgsm_livemon_headless -f 936.4M -g 40.2 -p 37
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
*** Error in `/usr/bin/python2': corrupted double-linked list: 0x01c22b08 ***
Aborted

```

Any idea what might be broken?

thanks

Gheek, 2018-04-18:
Exact the same Problem for me; seems to be a issue related to the build process.

imranrajjad, 2018-05-11:
guys use Ubuntu-mate

closed 2018-05-11 by imranrajjad

velichkov, 2018-11-23:
Hi @AgeOfMarcus 

See https://github.com/ptrkrysik/gr-gsm/issues/435#issuecomment-440987611

> To run the grgsm code you also need to specify --args rtl, otherwise you get error message, compare to desktop version where this is not needed.