ptrkrysik/gr-gsm #384: grgsm_livemon frequency slider

konrom, 2018-03-22:
Im running grgsm on kali linux with hackrf (latest firmware) and when im running grgsm_livemon i cant adjust the frequency or well i can adjust it but it doesnt actually affect anything hence i cant  fine tune it to c0 channel frequency.

velichkov, 2018-03-23:
Hi @konrom,

> on kali linux

Which version?

> i cant adjust the frequency or well i can adjust it but it doesnt actually affect anything hence i cant fine tune it to c0 channel frequency.

This is very strange. Are you able to tune to the desired frequency using the command line option?

konrom, 2018-03-25:
Yes CLI worked the error seems to be with the QT install i think, installing Gnuradio live SDR instead fixed it

closed 2018-03-25 by konrom