ptrkrysik/gr-gsm #386: Generating grgsm_livemon -> CMake Error

fipso, 2018-03-24:
Hello,
I was trying to install gr-gsm using pyBombs.

```

\- gr-gsm
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gr-gsm
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gr-gsm/build
Building:    (100%) [===============================================================================================================================================================================================================================]
[ 63%] Built target grgsm
[ 63%] Built target grgsm_swig_swig_doc
[ 65%] Built target _grgsm_swig_swig_tag
[ 68%] Built target grgsm_swig_swig_2d0df
[ 72%] Built target _grgsm_swig
[ 76%] Built target pygen_swig_c227a
[ 79%] Built target pygen_python_bc8f2
[ 82%] Built target pygen_python_misc_utils_6b490
[ 85%] Built target pygen_python_receiver_89d28
[ 88%] Built target pygen_python_demapping_89c33
[ 89%] Generating grgsm_livemon
CMake Error: cmake version 2.8.12.2
Usage: /usr/bin/cmake -E [command] [arguments ...]
Available commands: 
  chdir dir cmd [args]...   - run command in a given directory
  compare_files file1 file2 - check if file1 is same as file2
  copy file destination     - copy file to destination (either file or directory)
  copy_directory source destination   - copy directory 'source' content to directory 'destination'
  copy_if_different in-file out-file  - copy file if input has changed
  echo [string]...          - displays arguments as text
  echo_append [string]...   - displays arguments as text but no new line
  environment               - display the current environment
  make_directory dir        - create a directory
  md5sum file1 [...]        - compute md5sum of files
  remove [-f] file1 file2 ... - remove the file(s), use -f to force it
  remove_directory dir      - remove a directory and its contents
  rename oldname newname    - rename a file or directory (on one volume)
  tar [cxt][vfz][cvfj] file.tar [file/dir1 file/dir2 ...]
                            - create or extract a tar or zip archive
  time command [args] ...   - run command and return elapsed time
  touch file                - touch a file.
  touch_nocreate file       - touch a file but do not create it.
Available on UNIX only:
  create_symlink old new    - create a symbolic link new -> old

make[2]: *** [apps/grgsm_livemon] Error 1
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-gsm:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gr-gsm. Aborting.

```

fipso, 2018-03-24:
I think the problem is somewhere in /usr/local/src/gr-gsm/apps/CMakeLists.txt

```

SET(PYTHONPATH
        ${CMAKE_SOURCE_DIR}/python
        ${CMAKE_SOURCE_DIR}/python/misc_utils
        ${CMAKE_SOURCE_DIR}/python/demapping
        ${CMAKE_SOURCE_DIR}/python/receiver
        ${CMAKE_BINARY_DIR}/swig
        $ENV{PYTHONPATH}
        )
string(REPLACE ";" ":" PYTHONPATH "${PYTHONPATH}")

ADD_CUSTOM_COMMAND(
        OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/grgsm_livemon
        COMMAND "${CMAKE_COMMAND}"
                -E env PYTHONPATH="${PYTHONPATH}" GRC_BLOCKS_PATH=${CMAKE_SOURCE_DIR}/grc
                ${PC_GNURADIO_RUNTIME_PREFIX}/${GR_RUNTIME_DIR}/grcc -d ${CMAKE_CURRENT_BINARY_DIR}
                ${CMAKE_CURRENT_SOURCE_DIR}/grgsm_livemon.grc
        COMMAND "${CMAKE_COMMAND}" -E rename grgsm_livemon.py grgsm_livemon
        DEPENDS grgsm_livemon.grc
        )
ADD_CUSTOM_COMMAND(
        OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/grgsm_livemon_headless
        COMMAND "${CMAKE_COMMAND}"
                -E env PYTHONPATH="${PYTHONPATH}" GRC_BLOCKS_PATH=${CMAKE_SOURCE_DIR}/grc
                ${PC_GNURADIO_RUNTIME_PREFIX}/${GR_RUNTIME_DIR}/grcc -d ${CMAKE_CURRENT_BINARY_DIR}
                ${CMAKE_CURRENT_SOURCE_DIR}/grgsm_livemon_headless.grc
        COMMAND "${CMAKE_COMMAND}" -E rename grgsm_livemon_headless.py grgsm_livemon_headless
        DEPENDS grgsm_livemon_headless.grc
        )

```

fipso, 2018-03-24:
The error happens when cmake is trying to execute:

```

/usr/bin/cmake -E env PYTHONPATH="/usr/local/src/gr-gsm/python:/usr/local/src/gr-gsm/python/misc_utils:/usr/local/src/gr-gsm/python/demapping:/usr/local/src/gr-gsm/python/receiver:/usr/local/src/gr-gsm/build/swig:/usr/local/python:/usr/local/lib/python2.6/site-packages:/usr/local/lib64/python2.6/site-packages:/usr/local/lib/python2.6/dist-packages:/usr/local/lib64/python2.6/dist-packages:/usr/local/lib/python2.7/site-packages:/usr/local/lib64/python2.7/site-packages:/usr/local/lib/python2.7/dist-packages:/usr/local/lib64/python2.7/dist-packages:/usr/local/python/:/usr/local/lib/python2.6/site-packages:/usr/local/lib64/python2.6/site-packages:/usr/local/lib/python2.6/dist-packages:/usr/local/lib64/python2.6/dist-packages:/usr/local/lib/python2.7/site-packages:/usr/local/lib64/python2.7/site-packages:/usr/local/lib/python2.7/dist-packages:/usr/local/lib64/python2.7/dist-packages:YTHONPATH"

```

velichkov, 2018-03-24:
Hi @fipso,

> [ 89%] Generating grgsm_livemon
CMake Error: cmake version 2.8.12.2

Which Linux distribution and version are you using?
It looks like your cmake is too old and does not support the "cmake -E env" command. Could you try to update it to a newer version?

fipso, 2018-03-25:
I am using Ubuntu 14.04

fipso, 2018-03-25:
I installed cmake 3.2
Its working now. Thanks

closed 2018-03-25 by fipso