ptrkrysik/gr-gsm #389: make error "rtlsdr_source" not found

colorlight, 2018-04-05:
I follow the mannual install guide, but when I build the gr-gsm, some error comes out
```
[ 84%] Built target pygen_python_misc_utils_e5856
[ 87%] Built target pygen_python_receiver_4cba7
[ 89%] Built target pygen_python_demapping_ff0b9
[ 91%] Generating grgsm_livemon
>>> Error: Block key "rtlsdr_source" not found
Validation failed:

Block - blocks_rotator_cc_0 - Rotator(blocks_rotator_cc):
        Sink - in(0):
                Port is not connected.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:79: 
```
I install rlt-sdr and librtlsdr-dev with apt-get install
Why does it cannot find rtlsdr_source

colorlight, 2018-04-05:
I solve it myself, should install gr-osmosdr before building

closed 2018-04-05 by colorlight

tomasdev, 2019-02-06:
Weird, I already have gr-osmosdr installed but I still get this error.

velichkov, 2019-02-06:
Hi @tomasdev,

> Weird, I already have gr-osmosdr installed but I still get this error.

How did you install it - manually or from the repository?
Try running `sudo ldconfig`

tomasdev, 2019-02-06:
I solved it by uninstalling it and installing it again. Story of my life.

kumaraguru1735, 2019-03-28:
I already have gr-osmosdr installed but I still get this error.

[ 90%] Generating grgsm_livemon
Block key "rtlsdr_source" not found
Validation failed:

Block - blocks_rotator_cc_0 - Rotator(blocks_rotator_cc):
	Sink - in(0):
		Port is not connected.
Error during file compilation.
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:1323: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

velichkov, 2019-03-28:
Hi @kumaraguru1735,

> I already have gr-osmosdr 

How did you install it and which OS and version are you using?

Have you tried running `sudo ldconfig` as suggested in https://github.com/ptrkrysik/gr-gsm/issues/389#issuecomment-461069562

catalinalb, 2019-12-27:
Hi I just ran into the same issue: 
Ubuntu 18.04 installing with pybombs. gr-osmosdr has finally passed (with gitbranch gr-3.7)  also I have the gqrx installed. Still when I want to install gr-gsm:

(i did install sctp_bindx from synaptic)

[ 90%] Generating grgsm_livemon
Gtk-Message: 08:45:41.934: Failed to load module "canberra-gtk-module"
Block key "rtlsdr_source" not found
Validation failed:

Block - blocks_rotator_cc_0 - Rotator(blocks_rotator_cc):
	Sink - in(0):
		Port is not connected.

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
	Param - Cell allocation(cell_allocation):
		Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
		name 'arfcn' is not defined

Block - import_1 - Import(import):
	Param - Import(import):
		Import "from grgsm import arfcn" failed.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1317: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 2

Any ideas?

I want to use it mainly with hackrf so I don't need rtlsdr...

THank you

velichkov, 2019-12-27:
Hi @catalinalb,

> Ubuntu 18.04 installing with pybombs. gr-osmosdr has finally passed (with gitbranch gr-3.7) 

What prefix did you install into - `/usr/local/`?
Why `pybombs` and why not from the ubuntu or from the [Gnuradio 3.7 PPA](https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases-3.7) repositories?

> Block key "rtlsdr_source" not found

Try running `sudo ldconfig`

> I want to use it mainly with hackrf so I don't need rtlsdr...

This `rtlsdr_source` is needed for all SDR devices and not only for the rtlsdr dongles.

icecliffs, 2020-01-13:
Wow. Thanks

BigSkidderHyPhen, 2021-08-25:
> Hi I just ran into the same issue:
> Ubuntu 18.04 installing with pybombs. gr-osmosdr has finally passed (with gitbranch gr-3.7) also I have the gqrx installed. Still when I want to install gr-gsm: [...]

bruh im dealing with the same problem now. have u solved the issue?

root@okokok:~/gr-gsm/apps# grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'