ptrkrysik/gr-gsm #392: Unable to install - Build Failed

andrazjelenc, 2018-04-10:
Hello,
I've been following installation guide on Ubuntu 14.04 LTS, fresh install in VmWare.
I am facing this problem when installing gr-gsm:
`In file included from /usr/local/include/uhd/types/dict.hpp:148:0,
                 from /usr/local/include/uhd/types/device_addr.hpp:12,
                 from /usr/local/include/uhd/stream.hpp:13,
                 from /usr/local/include/uhd/device.hpp:12,
                 from /usr/local/include/uhd/usrp/multi_usrp.hpp:27,
                 from /usr/local/src/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_block.h:28,
                 from /usr/local/src/gnuradio/gr-uhd/lib/usrp_block_impl.h:26,
                 from /usr/local/src/gnuradio/gr-uhd/lib/usrp_block_impl.cc:23:
/usr/local/include/uhd/types/dict.ipp: In member function ‘bool uhd::dict<Key, Val>::operator==(const uhd::dict<Key, Val>&) const’:
/usr/local/include/uhd/types/dict.ipp:120:23: error: range-based ‘for’ loops are not allowed in C++98 mode
         for(auto &p : _map){
                       ^
/usr/local/include/uhd/types/dict.ipp:121:38: error: request for member ‘first’ in ‘p’, which is of non-class type ‘int’
             if (not (other.has_key(p.first) and other.get(p.first) == p.second)){
                                      ^
/usr/local/include/uhd/types/dict.ipp:121:61: error: request for member ‘first’ in ‘p’, which is of non-class type ‘int’
             if (not (other.has_key(p.first) and other.get(p.first) == p.second)){
                                                             ^
/usr/local/include/uhd/types/dict.ipp:121:73: error: request for member ‘second’ in ‘p’, which is of non-class type ‘int’
             if (not (other.has_key(p.first) and other.get(p.first) == p.second)){
                                                                         ^
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o] Error 1
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. `Aborting.``

Any ideas how to solve this?
Thanks!

anderstogern, 2018-04-10:
Hi,

I have this exact same problem!

I am installing using PyBOMBS.

I have tried with Mint 18.3 and Ubuntu 16.4, both in VMWare. I had it running fine in Mint 18.? in VMWare for a long time, but after a kernel upgrade (I think that must be it) gr-gsm stopped working.

For Ubuntu 16.4 it does not work with kernel 4.13.0-36-generic and newer, although I am not certain that it is the newer kernel that is causing trouble.

I would be very grateful if someone could help resolving this issue!

Cheers,
Anders

velichkov, 2018-04-10:
Hi @andrazjelenc,

> I've been following installation guide on Ubuntu 14.04 LTS, fresh install in VmWare.

Ubuntu 14.04 is pretty old, because this is a fresh install my advice is to install Ubuntu 16.04 LTS

```
/usr/local/include/uhd/types/dict.ipp: In member function ‘bool uhd::dict<Key, Val>::operator==(const uhd::dict<Key, Val>&) const’: 
 /usr/local/include/uhd/types/dict.ipp:120:23: error: range-based ‘for’ loops are not allowed in C++98 mode for(auto &p : _map){ ^ 
 /usr/local/include/uhd/types/dict.ipp:121:38: error: request for member ‘first’ in ‘p’, which is of non-class type ‘int’ if (not (other.has_key(p.first) and other.get(p.first) == p.second)){ ^ 
 /usr/local/include/uhd/types/dict.ipp:121:61: error: request for member ‘first’ in ‘p’, which is of non-class type ‘int’ if (not (other.has_key(p.first) and other.get(p.first) == p.second)){ ^
 /usr/local/include/uhd/types/dict.ipp:121:73: error: request for member ‘second’ in ‘p’, which is of non-class type ‘int’ if (not (other.has_key(p.first) and other.get(p.first) == p.second)){ ^
```

It seems the compiler is too old, could you provide the output of `cc --version`

velichkov, 2018-04-10:
Hi @anderstogern, 

> I am installing using PyBOMBS.

In Ubuntu 16.04 there is an easier method to install then pybombs, just execute all commands as in the docker file

https://github.com/ptrkrysik/gr-gsm/blob/e24860f9c018b761d3380d9ceaea0dff46ead6f4/tests/dockerfiles/Ubuntu_16_04.docker#L4-L32

> I have this exact same problem!
> I had it running fine in Mint 18.? in VMWare for a long time, but after a kernel upgrade (I think that must be it) gr-gsm stopped working.

It stopped compiling with the exact same error as in the original report?

> For Ubuntu 16.4 it does not work with kernel 4.13.0-36-generic and newer, although I am not certain that it is the newer kernel that is causing trouble.

What SDR do you use?

ptrkrysik, 2018-04-11:
Please follow the @velichkov advice. In case of further problems with installation please report the problem on the mailing list.

closed 2018-04-11 by ptrkrysik

anderstogern, 2018-04-11:
Hi @velichkov,

Thanks for your replies!

> It stopped compiling with the exact same error as in the original report?

Yes:

```
Building CXX object gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o
In file included from /usr/local/include/uhd/types/dict.hpp:148:0,
                 from /usr/local/include/uhd/types/device_addr.hpp:12,
                 from /usr/local/include/uhd/stream.hpp:13,
                 from /usr/local/include/uhd/device.hpp:12,
                 from /usr/local/include/uhd/usrp/multi_usrp.hpp:27,
                 from /usr/local/src/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_block.h:28,
                 from /usr/local/src/gnuradio/gr-uhd/lib/usrp_block_impl.h:26,
                 from /usr/local/src/gnuradio/gr-uhd/lib/usrp_block_impl.cc:23:
/usr/local/include/uhd/types/dict.ipp: In member function ‘bool uhd::dict<Key, Val>::operator==(const uhd::dict<Key, Val>&) const’:
/usr/local/include/uhd/types/dict.ipp:121:38: error: request for member ‘first’ in ‘p’, which is of non-class type ‘int’
             if (not (other.has_key(p.first) and other.get(p.first) == p.second)
                                      ^
/usr/local/include/uhd/types/dict.ipp:121:61: error: request for member ‘first’ in ‘p’, which is of non-class type ‘int’
             if (not (other.has_key(p.first) and other.get(p.first) == p.second)
                                                             ^
/usr/local/include/uhd/types/dict.ipp:121:73: error: request for member ‘second’ in ‘p’, which is of non-class type ‘int’
           if (not (other.has_key(p.first) and other.get(p.first) == p.second)){
                                                                       ^
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o] Error 1
gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/build.make:62: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o' failed
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
CMakeFiles/Makefile2:13533: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
make: *** [all] Error 2
Makefile:160: recipe for target 'all' failed
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

> What SDR do you use?

I am using a NooElec NESDR Nano 3.

Using the instructions you provided from the Docker file works for Ubuntu 16.04. Getting rid of PyBOMS is also much desired.

Now I am facing a problem with the grgsm_scanner application. It scans the first frequenc (926000000) fine, but hangs while scanning the second frequency 928000000). I have modified grgsm_scanner to be more verbose and it hangs while running the wideband_scanner:

```
sudo ./grgsm_scanner.py -v
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Scanning frequency 926000000
Before wideband_scanner()
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 NooElec NESDR Nano 3 SN: stx:978:1
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx2_64_mmx_orc
After wideband_scanner()
Starting recording..
set_min_output_buffer on block 2 to 4000000
set_min_output_buffer on block 4 to 4000000
set_min_output_buffer on block 3 to 4000000
Done recording
Scanning frequency 928000000
Before wideband_scanner()
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 NooElec NESDR Nano 3 SN: stx:978:1
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
After wideband_scanner()
Starting recording..
set_min_output_buffer on block 134 to 4000000
set_min_output_buffer on block 136 to 4000000
set_min_output_buffer on block 135 to 4000000
>>>>>HANGS HERE<<<<<
```

Pressing CTRL+C continues execution with the following output:

```
rtlsdr_read_async returned with -5
Done recording
rtlsdr_demod_write_reg failed with -7
r82xx_write: i2c wr failed=-9 reg=06 len=1
Scanning frequency 930000000
...
```

I am not sure exactly in which part of gnuradio it hangs, but it seems to be outside grgsm_scanner. Do you have any suggestions as to the cause of this?

Cheers,
Anders

anderstogern, 2018-04-12:
Hi again,

Further investigation shows that grgsm_livemon can scan the last frequency (928000000) just fine (although there is nothing on it) and I am able to receive data on other frequencies using grgsm_livemon.

So it seems to be a problem with the grgsm_scanner is using the gnuradio libraries. Unfortunately my current work relies on grgsm_scanner to work.

Does this spur any ideas as to what might be the cause of the hang?

Cheers,
Anders

andrazjelenc, 2018-04-14:
Hello,
I updated Ubuntu to 16.04 and used commands that are in Dockerfile and it worked, build and tests went straight through. Thank you for mentioning Dockerfile in tests.