ptrkrysik/gr-gsm #393: Add LimeSDR support

ptrkrysik, 2018-04-11:
Test applications on LimeSDR and add needed changes

9600, 2018-04-11:
Just to comment that, if possible, it would be better to take a more direct route and not go via gr-osmosdr -> UHD -> SoapySDR -> Lime Suite. E.g. gr-osmosdr -> SoapySDR -> Lime Suite would at least cut one abstraction out. Although I'm not sure if perhaps you need features provided by UHD?

ptrkrysik, 2018-04-11:
Currently only grgsm_trx app (which is in ptrkrysik/trx branch) needs
features provided by UHD.

Regarding the advice - I wouldn't try to use UHD to support LimeSDR if I
it's not necessary.

9600, 2018-08-11:
Just to note that we do now have blocks which integrate direct via LMS API.

https://github.com/myriadrf/gr-limesdr

I have enquired with a colleague to see whether more work will be needed for these to support grgsm_trx.

ptrkrysik, 2018-08-31:
@9600, for sure support for stream tags is missing. We are using them to:
* tell the Tx side where burst starts, how long it is ('length' tag) and when to transmit it ('tx_time' tag),
* on the Rx we get 'rx_time' tag that is needed to synchronize the Rx and Tx.

We are also developing hopping that is based on UHD timed commands that are controlled with 'tx_command' tags that have 'freq' and 'time' fields.

ptrkrysik, 2026-11-09:
Transmitting (TRX) is not supported by gr-gsm. Receiving with a LimeSDR works through gr-osmosdr/SoapySDR.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik