ptrkrysik/gr-gsm #394: grgsm_scanner hangs

anderstogern, 2018-04-12:
Hi,

I have installed gnuradio, gr-gsm, etc. using apt in Ubuntu 16.04, which seems to work just fine.

I am facing a problem with the grgsm_scanner application. It scans the first frequency (926000000) fine, but hangs while scanning the second frequency (928000000). I have modified grgsm_scanner to be more verbose and it hangs while running the wideband_scanner:

```
sudo ./grgsm_scanner.py -v
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Scanning frequency 926000000
Before wideband_scanner()
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 NooElec NESDR Nano 3 SN: stx:978:1
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx2_64_mmx_orc
After wideband_scanner()
Starting recording..
set_min_output_buffer on block 2 to 4000000
set_min_output_buffer on block 4 to 4000000
set_min_output_buffer on block 3 to 4000000
Done recording
Scanning frequency 928000000
Before wideband_scanner()
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 NooElec NESDR Nano 3 SN: stx:978:1
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
After wideband_scanner()
Starting recording..
set_min_output_buffer on block 134 to 4000000
set_min_output_buffer on block 136 to 4000000
set_min_output_buffer on block 135 to 4000000
>>>>>HANGS HERE<<<<<
Pressing CTRL+C continues execution with the following output:

rtlsdr_read_async returned with -5
Done recording
rtlsdr_demod_write_reg failed with -7
r82xx_write: i2c wr failed=-9 reg=06 len=1
Scanning frequency 930000000
...
```

I am not sure exactly in which part of gnuradio it hangs, but it seems to be outside grgsm_scanner.

Further investigation shows that grgsm_livemon can scan the last frequency (928000000) just fine (although there is nothing on it) and I am able to receive data on other frequencies using grgsm_livemon.

So it seems to be a problem with the way grgsm_scanner is using the gnuradio libraries. Unfortunately my current work relies on grgsm_scanner to work.

Does this spur any ideas as to what might be the cause of the hang?

Cheers,
Anders

velichkov, 2018-04-14:
Hi @anderstogern,

If you try with a different band (`grgsm_scanner --band=DCS1800`) does it hangs again and on which iteration?

Try with different/smaller sample rate (`grgsm_scanner --samp-rate=0.8e6`).

To check if the hangs is related to something specific to 928000000 try to modify grgsm_scanner to just skip this frequency and continues with the next one 930000000

If the problem is outside of the python code you could try to attach to the running process with `gdb` and take a full backtrace, it should be something similar to

```
gdb -p <pid_of_grgsm_scanner>
(gdb) thread apply all bt
```

anderstogern, 2018-04-16:
Hi @velichkov,

Using band DCS1800, grgsm_scanner hangs at the same place as before, but when scanning frequency 1842000000, which is the 18th frequency (out of 37) it scans in the run. On band GSM900 it hangs while scanning the second frequency.

Using a sample rate of 0.8e6 does not change anything for GSM900. For DCS1800 the nnumber of iterations are increased to 93 and it hangs at iteration 47, while scanning frequency 1843000000 (which is close to where it hung previously).

Skipping frequency 928000000 for GSM900 and without changing sample rate continues to scan 930000000, which goes fine, but then it hangs while scanning 932000000. If I change the sample rate to 0.8e6 it hangs already on the first frequency (925400000).

I have attached GDB to the process running grgsm_scanner on band GSM900 without skipping any frequencies. The output is below, but I am not familiar with reading GDB-output. From what I can see thread 5 is performing the rtlsdr_read_async call, which according to my original post, is the first call to return after breaking execution using CTRL+C. Anyway, it seems like the process is hanging in a conditional wait.

```
Attaching to process 9336
[New LWP 9356]
[New LWP 9391]
[New LWP 9392]
[New LWP 9395]
[New LWP 9448]
[New LWP 9449]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007fa7756a45d3 in select () at ../sysdeps/unix/syscall-template.S:84
84  ../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) thread apply all bt

Thread 7 (Thread 0x7fa6ee7c4700 (LWP 9449)):
#0  pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007fa771ba4043 in boost::thread::join_noexcept() ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#2  0x00007fa772d9d705 in gr::thread::thread_group::join_all() ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9
#3  0x00007fa772e0e2d3 in gr::top_block_impl::wait_for_jobs() ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9
#4  0x00007fa772e0e878 in gr::top_block_impl::wait() ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9
#5  0x00007fa76e8ecd2f in ?? ()
   from /usr/lib/python2.7/dist-packages/gnuradio/gr/_runtime_swig.x86_64-linux-gnu.so
#6  0x00007fa76e8ece62 in ?? ()
   from /usr/lib/python2.7/dist-packages/gnuradio/gr/_runtime_swig.x86_64-linux-gnu.so
#7  0x00000000004bc3fa in call_function (oparg=<optimized out>, 
    pp_stack=0x7fa6ee7c3740) at ../Python/ceval.c:4350
#8  PyEval_EvalFrameEx () at ../Python/ceval.c:2987
#9  0x00000000004c136f in fast_function (nk=<optimized out>, 
    na=<optimized out>, n=1, pp_stack=0x7fa6ee7c3860, 
    func=<function at remote 0x7fa76e8426e0>) at ../Python/ceval.c:4435
#10 call_function (oparg=<optimized out>, pp_stack=0x7fa6ee7c3860)
    at ../Python/ceval.c:4370
#11 PyEval_EvalFrameEx () at ../Python/ceval.c:2987
#12 0x00000000004c136f in fast_function (nk=<optimized out>, 
    na=<optimized out>, n=1, pp_stack=0x7fa6ee7c3980, 
    func=<function at remote 0x7fa76b3dc8c0>) at ../Python/ceval.c:4435
#13 call_function (oparg=<optimized out>, pp_stack=0x7fa6ee7c3980)
    at ../Python/ceval.c:4370
#14 PyEval_EvalFrameEx () at ../Python/ceval.c:2987
#15 0x00000000004c136f in fast_function (nk=<optimized out>, 
    na=<optimized out>, n=1, pp_stack=0x7fa6ee7c3aa0, 
    func=<function at remote 0x7fa76e7fb0c8>) at ../Python/ceval.c:4435
#16 call_function (oparg=<optimized out>, pp_stack=0x7fa6ee7c3aa0)
    at ../Python/ceval.c:4370
#17 PyEval_EvalFrameEx () at ../Python/ceval.c:2987
#18 0x00000000004b9ab6 in PyEval_EvalCodeEx () at ../Python/ceval.c:3582
#19 0x00000000004d54b9 in function_call.lto_priv ()
    at ../Objects/funcobject.c:523
#20 0x00000000004eebee in PyObject_Call (kw=0x0, 
    arg=(<_top_block_waiter(_Thread__ident=140354942420736, _Thread__block=<_Condition(_Verbose__verbose=False, _Condition__lock=<thread.lock at remote 0x7fa775d238f0>, acquire=<built-in method acquire of thread.lock object at remote 0x7fa775d238f0>, _Condition__waiters=[], release=<built-in method release of thread.lock object at remote 0x7fa775d238f0>) at remote 0x7fa76195bd90>, _Thread__name='Thread-1', _Thread__daemonic=1, tb=<top_block_sptr(this=<SwigPyObject at remote 0x7fa7686592a0>) at remote 0x7fa76193df10>, _Thread__started=<_Event(_Verbose__verbose=False, _Event__flag=True, _Event__cond=<_Condition(_Verbose__verbose=False, _Condition__lock=<thread.lock at remote 0x7fa775d23910>, acquire=<built-in method acquire of thread.lock object at remote 0x7fa775d23910>, _Condition__waiters=[], release=<built-in method release of thread.lock object at remote 0x7fa775d23910>) at remote 0x7fa76195bcd0>) at remote 0x7fa76195bc50>, _Thread__stderr=<file at remote 0x7fa775d701e0>, event=<_Event(_Verbose__verbose=False, _Ev...(truncated), func=<function at remote 0x7fa76e7f9f50>) at ../Objects/abstract.c:2546
#21 instancemethod_call.lto_priv () at ../Objects/classobject.c:2602
#22 0x00000000004a577e in PyObject_Call () at ../Objects/abstract.c:2546
#23 0x00000000004c5e10 in PyEval_CallObjectWithKeywords ()
    at ../Python/ceval.c:4219
#24 0x0000000000589172 in t_bootstrap () at ../Modules/threadmodule.c:620
[... 122 lines cut ...]
#22 call_function (oparg=<optimized out>, pp_stack=0x7fff60604d10)
    at ../Python/ceval.c:4370
#23 PyEval_EvalFrameEx () at ../Python/ceval.c:2987
[... cut ...]
#30 run_mod.lto_priv () at ../Python/pythonrun.c:1376
#31 0x00000000004e5422 in PyRun_FileExFlags () at ../Python/pythonrun.c:1362
#32 0x00000000004e3cd6 in PyRun_SimpleFileExFlags ()
    at ../Python/pythonrun.c:948
#33 0x0000000000493ae2 in Py_Main () at ../Modules/main.c:640
#34 0x00007fa7755c7830 in __libc_start_main (main=0x4934c0 <main>, argc=3, 
    argv=0x7fff60605358, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fff60605348)
    at ../csu/libc-start.c:291
#35 0x00000000004933e9 in _start ()
```

velichkov, 2018-04-16:
Hi @anderstogern,

How long do you wait before deciding that it is hanged? 

Could you install the following `dbgsym` packages and take another backtrace:

gnuradio-dbgsym
gr-osmosdr-dbgsym
rtl-sdr-dbgsym
libusb-0.1-4-dbgsym

[Instructions](https://wiki.ubuntu.com/Debug%20Symbol%20Packages) how to enable the debug repositories.

anderstogern, 2018-04-17:
Hi @velichkov,

I was waiting several minutes!

During some more testing I noticed something strange: grgsm_scanner progressed further when I removed focus from the VM (to the host) immediately after starting the process.
This has led me to believe that the problem is with some parts of the VmWare Tools. To test this I have made the same installation in VirtualBox and here it runs perfectly, even with the guest additions installed!

So the problem is specific to running in VmWare with a newer version of the VmWare Tools installed!

I hope that this helps someone else, although the problem is not fixed the solution of using VirtualBox instead works for me.

@velichkov: If you want me to, I can run further diagnostics in VmWare?

Cheers,
Anders

velichkov, 2018-04-17:
Hi @anderstogern,

> the solution of using VirtualBox instead works for me.

It's great you have been able to find a solution.

> If you want me to, I can run further diagnostics in VmWare?

No need. Please only rename the issue to indicate that it hangs in a VmWare VM and close it.

Cheers,
Vasil

anderstogern, 2018-04-17:
Hi @velichkov,

Thank you very much for your assistance!

I have renamed the issue and will close it!

Cheers,
Anders

closed 2018-04-17 by anderstogern