ptrkrysik/gr-gsm #395: i can't find system information type 5 or 6

mhshamsinejad, 2018-04-12:
hi Piotr Krysik
I capture gsm traffic by using **"grgsm_capture -g 40 -a 16 -s 1e6 -c 970121.cfile -T 60"**
and I can find SDCCH8 Time slots number, ARFCNs, TMSI and other information from BCCH by using **"grgsm_decode -a 16 -s 1e6 -c 970121.cfile -m BCCH -t 0"**
then I'm using **"grgsm_decode -a 16 -s 1e6 -c 970121.cfile -m SDCCH8 -t 3"** for finding system information type 5 but can't find it. There exist other packets such as cipher mode. I'm trying over and over for capturing and decoding gsm frame, but I can't find **SI5** or **SI6** packet.
please help me.

velichkov, 2018-04-14:
Hi @mhshamsinejad,

Are you able to find SI5 and SI6 using the steps and the cfile file from [Decoding How To](../wiki/Usage:-Decoding-How-To)?

Could you share the cfile?

mhshamsinejad, 2018-04-14:
yes I can find it using the step from cfile from  [https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To](url)

but I can't find SI5 or SI6 from my capture cfile. can the **frequency hopping** cause this problem?! 
I try too chanalize my cfile with **grgsm_channelize** and use  **grgsm_hopping_example.grc**  but can't get any output in wireshark. could you explain how to get output in wireshark from frequency hopping cfile?

(I using rtl-sdr dvb-t dongle for capture and some BTS are with low distance ARFCNs from each other in captured cfile. for example ARFCN 48 , 46.

velichkov, 2018-04-14:
> can the frequency hopping cause this problem?!
> I try too chanalize my cfile with grgsm_channelize and use grgsm_hopping_example.grc but can't get any output in wireshark. could you explain how to get output in wireshark from frequency hopping cfile?

Sorry I don't have experience with hopping channels so I can't explain how to decode them.

ptrkrysik, 2018-05-27:
@ mhshamsinejad, please ask such questions on the mailing list.

closed 2018-05-27 by ptrkrysik