ptrkrysik/gr-gsm #396: Swig - no matching function for call on CentOS 7

E7ernal, 2018-04-20:
When trying to build gr-gsm, I'm getting the following error and dozens of similar complaints:

```[ 69%] Building CXX object swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
/home/david/Downloads/gr-gsm/swig/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_receiver_sptr_set_processor_affinity(PyObject*, PyObject*, PyObject*)’:
/home/david/Downloads/gr-gsm/swig/grgsm_swigPYTHON_wrap.cxx:8725:110: error: no matching function for call to ‘gr::gsm::receiver::set_processor_affinity(const std::vector<long unsigned int>&)’
       (*arg1)->set_processor_affinity((std::vector< std::size_t,std::allocator< std::size_t > > const &)*arg2);
                                                                                                              ^
/home/david/Downloads/gr-gsm/swig/grgsm_swigPYTHON_wrap.cxx:8725:110: note: candidate is:
In file included from /home/david/Downloads/gr-gsm/swig/grgsm_swigPYTHON_wrap.cxx:4850:0:
/usr/local/include/gnuradio/block.h:626:10: note: virtual void gr::block::set_processor_affinity(const std::vector<int>&)
     void set_processor_affinity(const std::vector<int> &mask);
```

I've upgraded my swig to the latest available version, but the errors persist. There seems to be an issue converting the types for processor_affinity calls, but I'm not enough of a swig guru to know how to fix it.

uname -a gives `3.10.0-693.el7.x86_64 #1 SMP Tue Aug 22 21:09:27 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

swig version is: ```SWIG Version 3.0.12

Compiled with g++ [x86_64-pc-linux-gnu]

Configured options: +pcre```

velichkov, 2018-04-23:
Hi @E7ernal,

I'm able to compile on CentOS7 using the gnuradio and cmake3 from the [epel repository](https://fedoraproject.org/wiki/EPEL)

```
gnuradio-devel-3.7.11-6.el7.x86_64
gnuradio-3.7.11-6.el7.x86_64
cmake3-3.6.3-1.el7.x86_64
cmake3-data-3.6.3-1.el7.noarch
swig-2.0.10-5.el7.x86_64
```

How did you install gnuradio and which version?
How did you install swig 3.0.12? Could you try to downgrade the swig to the one from the centos repository, I don't think that this problem is directly related to the swig version.

If this is an old installation that you try to update could you try to remove the old version first - `make uninstall`

Note that you need to use cmake3 because of #378 
```
mkdir build
cd build
cmake3 ..
make VERBOSE=1
```
Please try this with the above versions and attach the full output.

E7ernal, 2018-04-23:
I did not have cmake3 installed, but installing it doesn't seem to make a difference. All other packages are correct versions. I built swig 3.0.12 from source, but even targeting the older swig 2.0.10-5 from the repo, I get the same errors. I don't think this is a swig version issue either.

Output is below:

```
[david@gpudev gr-gsm]$ make VERBOSE=1
/usr/bin/cmake3 -H/home/david/Downloads/gr-gsm -B/home/david/Downloads/gr-gsm --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake3 -E cmake_progress_start /home/david/Downloads/gr-gsm/CMakeFiles /home/david/Downloads/gr-gsm/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `/home/david/Downloads/gr-gsm'
make -f lib/CMakeFiles/grgsm.dir/build.make lib/CMakeFiles/grgsm.dir/depend
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
cd /home/david/Downloads/gr-gsm && /usr/bin/cmake3 -E cmake_depends "Unix Makefiles" /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/lib /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/lib /home/david/Downloads/gr-gsm/lib/CMakeFiles/grgsm.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
make -f lib/CMakeFiles/grgsm.dir/build.make lib/CMakeFiles/grgsm.dir/build
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
make[2]: Nothing to be done for `lib/CMakeFiles/grgsm.dir/build'.
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
[ 62%] Built target grgsm
make -f swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make swig/CMakeFiles/grgsm_swig_swig_doc.dir/depend
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
cd /home/david/Downloads/gr-gsm && /usr/bin/cmake3 -E cmake_depends "Unix Makefiles" /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm/swig/CMakeFiles/grgsm_swig_swig_doc.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
make -f swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make swig/CMakeFiles/grgsm_swig_swig_doc.dir/build
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
make[2]: Nothing to be done for `swig/CMakeFiles/grgsm_swig_swig_doc.dir/build'.
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
[ 62%] Built target grgsm_swig_swig_doc
make -f swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build.make swig/CMakeFiles/_grgsm_swig_swig_tag.dir/depend
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
cd /home/david/Downloads/gr-gsm && /usr/bin/cmake3 -E cmake_depends "Unix Makefiles" /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm/swig/CMakeFiles/_grgsm_swig_swig_tag.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
make -f swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build.make swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
make[2]: Nothing to be done for `swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build'.
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
[ 64%] Built target _grgsm_swig_swig_tag
make -f swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build.make swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/depend
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
cd /home/david/Downloads/gr-gsm && /usr/bin/cmake3 -E cmake_depends "Unix Makefiles" /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm/swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
make -f swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build.make swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
make[2]: Nothing to be done for `swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build'.
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
[ 68%] Built target grgsm_swig_swig_2d0df
make -f swig/CMakeFiles/_grgsm_swig.dir/build.make swig/CMakeFiles/_grgsm_swig.dir/depend
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
cd /home/david/Downloads/gr-gsm && /usr/bin/cmake3 -E cmake_depends "Unix Makefiles" /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm /home/david/Downloads/gr-gsm/swig /home/david/Downloads/gr-gsm/swig/CMakeFiles/_grgsm_swig.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
make -f swig/CMakeFiles/_grgsm_swig.dir/build.make swig/CMakeFiles/_grgsm_swig.dir/build
make[2]: Entering directory `/home/david/Downloads/gr-gsm'
[ 69%] Building CXX object swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
cd /home/david/Downloads/gr-gsm/swig && /usr/bin/c++   -D_grgsm_swig_EXPORTS -I/home/david/Downloads/gr-gsm/lib -I/home/david/Downloads/gr-gsm/include -I/home/david/Downloads/gr-gsm/lib/decoding -I/usr/local/include -I/home/david/Downloads/gr-gsm/swig -I/usr/include/gnuradio/swig -I/usr/include/python2.7  -O3 -DNDEBUG -fPIC   -fvisibility=hidden -Wno-unused-but-set-variable -o CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o -c /home/david/Downloads/gr-gsm/swig/grgsm_swigPYTHON_wrap.cxx
[... cut ...]
   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
                     ^
/home/david/Downloads/gr-gsm/swig/grgsm_swigPYTHON_wrap.cxx:28:21: note:   no known conversion for argument 1 from ‘std::vector<int>’ to ‘const std::vector<long unsigned int>&’
make[2]: *** [swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o] Error 1
make[2]: Leaving directory `/home/david/Downloads/gr-gsm'
make[1]: *** [swig/CMakeFiles/_grgsm_swig.dir/all] Error 2
make[1]: Leaving directory `/home/david/Downloads/gr-gsm'
make: *** [all] Error 2
```

velichkov, 2018-04-23:
Hi,

It turns out a similar issue has been reported before #179 and some instructions in #180

Please provide the output of `/usr/bin/c++ --version` and attach as file the output of `rpm -qa`

Could you also try a clean rebuild
```
rm build
mkdir build
cd build
cmake3 ..
```

> I did not have cmake3 installed, but installing it doesn't seem to make a difference.

You are gonna need it later when the .grc flowgraphs are compiled.

How important is CentOS7 for you? If CentOS7 is not that important try installing gr-gsm on Fedora 26/27 or Ubuntu 16.04

velichkov, 2018-04-24:
Hi @E7ernal,

Could you attache the long outputs as files because the long posts makes following the discussion pretty hard.

> The build I did above was a clean build. 

It seems you are building in the source directory and not in a subdirectory which sometimes could cause problems and the output shows that it was not a clean build because not all c files were recompiled.

Could you execute the following and attache the full output as a file
```
find /usr/local/
cd /home/david/Downloads/gr-gsm/
git clean -fdx
git describe
git status
mkdir build
cd build
cmake3 ..
make VERBOSE=1
```

I was able to build a CentOS7 docker container in travis (1b90d2ea310701cbf7256d09fb4ed640c3fc9f6e)

See https://travis-ci.org/velichkov/gr-gsm/builds/370673004#L3082-L3087
The [list of installed packages](https://travis-ci.org/velichkov/gr-gsm/builds/370673004#L2386-L2802)
The used [CentOS_7.Dockerfile](https://github.com/ptrkrysik/gr-gsm/blob/1b90d2ea310701cbf7256d09fb4ed640c3fc9f6e/tests/dockerfiles/CentOS_7.Dockerfile)

In gnuradio that comes from the epel repository the Qt blocks were disabled and because of this the grgsm_livemon can't be compiled. See https://src.fedoraproject.org/rpms/gnuradio/blob/epel7/f/gnuradio.spec#_55-57

E7ernal, 2018-04-24:
I've attached files for each step with output. I did find that problem with the gnuradio which comes from the epel repository. If I'm not mistaken I had to build from source for that reason.

[output.txt](https://github.com/ptrkrysik/gr-gsm/files/1944699/output.txt)
[output2.txt](https://github.com/ptrkrysik/gr-gsm/files/1944700/output2.txt)
[output3.txt](https://github.com/ptrkrysik/gr-gsm/files/1944701/output3.txt)
[output4.txt](https://github.com/ptrkrysik/gr-gsm/files/1944702/output4.txt)
[output5.txt](https://github.com/ptrkrysik/gr-gsm/files/1944703/output5.txt)
[output6.txt](https://github.com/ptrkrysik/gr-gsm/files/1944704/output6.txt)

velichkov, 2018-04-25:
Hi @E7ernal,

You have installed two versions of gnuradio, this is always dangerous and possible source for many errors.

> gnuradio-devel-3.7.11-6.el7.x86_64
> gnuradio-3.7.11-6.el7.x86_64
> /usr/include/gnuradio/api.h 

The one from the epel repository.

> /usr/local/include/gnuradio
> /usr/local/.pybombs
/usr/local/.pybombs/inventory.yml
/usr/local/.pybombs/recipes
/usr/local/.pybombs/config.yml
/usr/local/setup_env.sh

And the one installed using pybombs.

Given the fact that the epel version does not support Qt blocks you most probably want to use the other one - which hopefully will have Qt blocks installed and runnig.

So my advise is to remove the gnuradio and gnuradio-devel RPMs - `yum erase gnuradio gnuradio-devel` and then do a clean build.

If for some reason you don't want to remove the RPMs you could also try to source the `/usr/local/setup_env.sh` before the clean rebuild

```
rm -rf build
source /usr/local/setup_env.sh
mkdir build && cd build
cmake3 ...
```

P.S.
You will also need to install gr-osmosdr - `pybombs install gr-osmosdr`. All dependencies are listed in the [pybombs gr-gsm receipt](https://github.com/gnuradio/gr-etcetera/blob/master/gr-gsm.lwr#L22-L27)