ptrkrysik/gr-gsm #397: grgsm_scanner - segfault

bossco79, 2018-04-21:
Hello! Faced such a problem during startup grgsm_scanner...
I tried to reinstall to a clean OS according to the instructions from the wiki - all the same! All other grgsm components work fine.
My OS info
# uname -a
Linux 4.15.0-kali2-amd64 #1 SMP Debian 4.15.11-1kali1 (2018-03-21) x86_64 GNU/Linux

root@:/usr/local/src/gr-gsm/apps# ./grgsm_scanner 
Segmentation fault
or
root@:~/IMSI-catcher# python scan-and-livemon 
Locating potential GSM base station frequencies (this can take a few minutes).
Segmentation fault

Dmesg:
[ 1439.034375] hackrf_source_c[5519]: segfault at 55babb54b000 ip 00007ff924eae060 sp 00007ff8b27f37b0 error 4 in libgnuradio-osmosdr-0.1.5git.so.0.0.0[7ff924e41000+d0000]
[ 1648.932165] hackrf_source_c[5708]: segfault at 5556a0b15000 ip 00007f9a52495060 sp 00007f99deffc7b0 error 4 in libgnuradio-osmosdr-0.1.5git.so.0.0.0[7f9a52428000+d0000]
[ 2275.911641] hackrf_source_c[9154]: segfault at 55ac93797000 ip 00007f570ac80060 sp 00007f5697ffe7b0 error 4 in libgnuradio-osmosdr-0.1.5git.so.0.0.0[7f570ac13000+d0000]

velichkov, 2018-04-21:
Hi @bossco79,

> [ 1439.034375] hackrf_source_c[5519]: segfault at 55babb54b000 ip 00007ff924eae060 sp 

It seems you are using HackRF. See #366 and try to comment out line 212 in grgsm_scanner

```python
         #self.rtlsdr_source.set_min_output_buffer(int(sample_rate*rec_len))
```

velichkov, 2018-05-19:
Hi @mazen-bassiouny,

> just commenting this line should solve the issue with hackrf 

I've linked a wrong issue, the correct one is #336 and as you could see in it, this have helped at least two hackrf users.

> i already did that but the same (sementation fault ) is any further action i can do ?

I don't have hackrf so I can't really debug this crash.

ptrkrysik, 2018-06-02:
This change 92e09f9ff4fa44f728da2395065a5d3be8c4cce6 should solve the issue with HackRF

closed 2018-06-02 by ptrkrysik