ptrkrysik/gr-gsm #399: Installation -> Error generating grgsm_livemon

Okeur75, 2018-04-29:
Hi all,

I'm trying to install gr-gsm on kali but make fails at "Generating grgsm_livemon" step.
Below the error:

> [ 90%] Generating grgsm_livemon
No such file or directory
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1278: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<

I'm on last Kali version, updated and everything.
root@kali:~/gr-gsm/build# uname -r
4.15.0-kali2-amd64

I followed the manual installation guide.

At first I try on Kali under Windows store and got the error, so I thought it was because of the way I used it. So now I'm on a VM under virtualbox but I got the same error at the exact same point.

If I can provide any extra information to help please ask.

Regards,

velichkov, 2018-04-29:
Hi @Okeur75,

> [ 90%] Generating grgsm_livemon
No such file or directory
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed

Could you run `make VERBOSE=1` and attach the full output as a file.

> I'm on last Kali version, updated and everything.

Also provide the full output of the following commands

```
cat /etc/debian_version
cat /etc/lsb-release  
dpkg -l gnuradio
which grcc
grcc --version
cmake --version
```

Okeur75, 2018-04-30:
Hey,

Please find the requested information below. Thank you.
`root@kali:~/gr-gsm/build# 
root@kali:~/gr-gsm/build# cat /etc/debian_version
kali-rolling
root@kali:~/gr-gsm/build# cat /etc/lsb-release 
DISTRIB_ID=Kali
DISTRIB_RELEASE=kali-rolling
DISTRIB_CODENAME=kali-rolling
DISTRIB_DESCRIPTION="Kali GNU/Linux Rolling"
root@kali:~/gr-gsm/build# dpkg -l gnuradio
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                           Version                      Architecture                 Description
+++-==============================================-============================-============================-==================================================================================================
ii  gnuradio                                       3.7.11-8                     amd64                        GNU Radio Software Radio Toolkit
root@kali:~/gr-gsm/build# which grcc
/usr/bin/grcc
root@kali:~/gr-gsm/build# grcc --version
Traceback (most recent call last):
  File "/usr/bin/grcc", line 29, in <module>
    from gnuradio import gr
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/__init__.py", line 41, in <module>
    from runtime_swig import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 17, in <module>
    _runtime_swig = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_runtime_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _runtime_swig
root@kali:~/gr-gsm/build# cmake --version
cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
root@kali:~/gr-gsm/build# `

[output-make.txt](https://github.com/ptrkrysik/gr-gsm/files/1959656/output-make.txt)

velichkov, 2018-04-30:
Hi @Okeur75,

> ImportError: No module named _runtime_swig

The workaround for this error (#391) is to create one symlink
```
ln -sf /usr/lib/x86_64-linux-gnu/libvolk.so.1.3.1 /usr/lib/x86_64-linux-gnu/libvolk.so.1.3 
```
and then try again and provide the full output

```
grcc --version
make VERBOSE=1
```

velichkov, 2018-04-30:
Hi @Okeur75,

> [ 90%] Generating grgsm_livemon
cd /root/gr-gsm/build/apps && /usr/bin/cmake -E env PYTHONPATH="/root/gr-gsm/python:/root/gr-gsm/python/misc_utils:/root/gr-gsm/python/demapping:/root/gr-gsm/python/receiver:/root/gr-gsm/python/transmitter:/root/gr-gsm/python/trx:/root/gr-gsm/build/swig" GRC_BLOCKS_PATH=/root/gr-gsm/grc **/bin/grcc** -d /root/gr-gsm/build/apps /root/gr-gsm/apps/grgsm_livemon.grc
No such file or directory

Most probably you don't have `pkg-config` and that's why CMake thinks the `grcc` full path is `/bin/grcc` and not `/usr/bin/grcc`. Please install `pkg-config` and do a clean rebuild

```bash
apt-get install pkg-config
rm -rf /root/gr-gsm/build
mkdir /root/gr-gsm/build
cd /root/gr-gsm/build
cmake ..
make VERBOSE=1
```

sashatytko, 2018-05-01:
I have the same problem.i have alŗeady installed pkg-config and i have latest version of Linux.

velichkov, 2018-05-01:
Hi @Okeur75,

> Block key "rtlsdr_source" not found

You need to install `gr-osmosdr` as well. Also have a look in the Kali.docker file and install all listed packages.

https://github.com/ptrkrysik/gr-gsm/blob/2471052cfe724699c8bd2d4fec20d7f194cf6bb9/tests/dockerfiles/Kali.docker#L4-L18

Hi @sashatytko,

> I have the same problem.i have alŗeady installed pkg-config and i have latest version of Linux.

Which of the three problems discussed in this issue, if the installation of `pkg-config`, `gr-osmosdr` and the symlink creation did not help most probably you have a different problem and in such case provide the full output of `make VERBOSE=1` attached as text file

sashatytko, 2018-05-01:
:~/gr-gsm/build$ make VERBOSE=1
/usr/bin/cmake -H/home/sasha/gr-gsm -B/home/sasha/gr-gsm/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/sasha/gr-gsm/build/CMakeFiles /home/sasha/gr-gsm/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all

make[1]: Entering directory '/home/sasha/gr-gsm/build'
make -f lib/CMakeFiles/grgsm.dir/build.make lib/CMakeFiles/grgsm.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/lib /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/lib /home/sasha/gr-gsm/build/lib/CMakeFiles/grgsm.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
make -f lib/CMakeFiles/grgsm.dir/build.make lib/CMakeFiles/grgsm.dir/build
make[2]: Entering directory '/home/sasha/gr-gsm/build'
make[2]: Nothing to be done for 'lib/CMakeFiles/grgsm.dir/build'.
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
[ 61%] Built target grgsm
make -f swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make swig/CMakeFiles/grgsm_swig_swig_doc.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/swig /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/swig /home/sasha/gr-gsm/build/swig/CMakeFiles/grgsm_swig_swig_doc.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
make -f swig/CMakeFiles/grgsm_swig_swig_doc.dir/build.make swig/CMakeFiles/grgsm_swig_swig_doc.dir/build
make[2]: Entering directory '/home/sasha/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/grgsm_swig_swig_doc.dir/build'.
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
[ 61%] Built target grgsm_swig_swig_doc
make -f swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build.make swig/CMakeFiles/_grgsm_swig_swig_tag.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/swig /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/swig /home/sasha/gr-gsm/build/swig/CMakeFiles/_grgsm_swig_swig_tag.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
make -f swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build.make swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build
make[2]: Entering directory '/home/sasha/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/_grgsm_swig_swig_tag.dir/build'.
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
[ 64%] Built target _grgsm_swig_swig_tag
make -f swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build.make swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/swig /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/swig /home/sasha/gr-gsm/build/swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
make -f swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build.make swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build
make[2]: Entering directory '/home/sasha/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/grgsm_swig_swig_2d0df.dir/build'.
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
[ 67%] Built target grgsm_swig_swig_2d0df
make -f swig/CMakeFiles/pygen_swig_67205.dir/build.make swig/CMakeFiles/pygen_swig_67205.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/swig /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/swig /home/sasha/gr-gsm/build/swig/CMakeFiles/pygen_swig_67205.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
make -f swig/CMakeFiles/pygen_swig_67205.dir/build.make swig/CMakeFiles/pygen_swig_67205.dir/build
make[2]: Entering directory '/home/sasha/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/pygen_swig_67205.dir/build'.
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
[ 70%] Built target pygen_swig_67205
make -f swig/CMakeFiles/_grgsm_swig.dir/build.make swig/CMakeFiles/_grgsm_swig.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/swig /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/swig /home/sasha/gr-gsm/build/swig/CMakeFiles/_grgsm_swig.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
make -f swig/CMakeFiles/_grgsm_swig.dir/build.make swig/CMakeFiles/_grgsm_swig.dir/build
make[2]: Entering directory '/home/sasha/gr-gsm/build'
make[2]: Nothing to be done for 'swig/CMakeFiles/_grgsm_swig.dir/build'.
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
[ 75%] Built target _grgsm_swig
make -f python/CMakeFiles/pygen_python_a4375.dir/build.make python/CMakeFiles/pygen_python_a4375.dir/depend
make[2]: Entering directory '/home/sasha/gr-gsm/build'
cd /home/sasha/gr-gsm/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sasha/gr-gsm /home/sasha/gr-gsm/python /home/sasha/gr-gsm/build /home/sasha/gr-gsm/build/python /home/sasha/gr-gsm/build/python/CMakeFiles/p [... cut ...]
		Port is not connected.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
make[2]: Leaving directory '/home/sasha/gr-gsm/build'
CMakeFiles/Makefile2:1257: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make[1]: Leaving directory '/home/sasha/gr-gsm/build'
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

sashatytko, 2018-05-01:
I have another error in gnuradio 
Param - Import(import):
	Import "from grgsm import arfcn" failed.
and my block in red

velichkov, 2018-05-01:
Hi @sashatytko,

Have you installed it - `make install`?  If yes try to run `ldconfig`

Okeur75, 2018-05-01:
1) Thank you, because it's almost solved.
2) Same error as sasha:
`root@kali:~/grgsm/gr-gsm/build/apps# ldconfig
root@kali:~/grgsm/gr-gsm/build/apps# ./grgsm_livemon
Traceback (most recent call last):
  File "./grgsm_livemon", line 30, in <module>
    from grgsm import arfcn
ImportError: No module named grgsm
`

3) Is it normal that no executable are created ? I can't find grgsm_scanner, grgsm_decode etc. ?

sashatytko, 2018-05-01:
Ldconfig solved this problem:-)
Thank you

Okeur75, 2018-05-01:
Ok forget it, it was the sdr not recognized by the VM.
Thank you for your help, it's working now !

Below is a "how-to" for install gr-gsm on a fresh Kali install for root user.
If it can help :)

````
apt update
apt upgrade
apt install git libboost-all-dev swig cmake autoconf libtool pkg-config build-essential python-docutils libcppunit-dev doxygen liblog4cpp5-dev python-scipy gnuradio-dev gr-osmosdr libosmocore-dev gr-osmosdr gnuradio  libusb-1.0-0 libusb-1.0-0-dev libfftw3-dev python-numpy
git clone https://github.com/ptrkrysik/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..

ln -sf /usr/lib/x86_64-linux-gnu/libvolk.so.1.3.1 /usr/lib/x86_64-linux-gnu/libvolk.so.1.3

make
make install

mkdir /root/.gnuradio/
echo "[grc]" > /root/.gnuradio/config.conf
echo "local_blocks_path=/usr/local/share/gnuradio/grc/blocks" >> /root/.gnuradio/config.conf

ldconfig
```

cookins, 2018-05-09:
my bad but i have same issue:
```
Block key "rtlsdr_source" not found
Validation failed:

Block - blocks_rotator_cc_0 - Rotator(blocks_rotator_cc):
	Sink - in(0):
		Port is not connected.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1320: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```
i typed before:
sudo apt-get install gr-osmosdr

and all other packages from kali.docker from my GR-GSM dir.
it was already installed before by me - so nothing was updated or removed, or upgraded.
then i deleted folder GR-GSM to start as new install and made everything from previous post, but still have same issue.... what i doing wrong? thanks!

velichkov, 2018-05-10:
Hi @cookins,

When you start the `gnuradio-companion` from the terminal it should print the directories it searches blocks into. The rtlsdr_source block (rtlsdr_source.xml) must be in one of those directories.  As you have installed it from the apt repository its should be in /usr/share/gnuradio/grc/blocks/rtlsdr_source.xml but it will be good to double check this.

Try to re-install the gnuradio and gr-osmosdr: `apt-get install --reinstall gnuradio gr-osmosdr` or `apt-get remove gnuradio gr-osmosdr && apt-get install gnuradio gr-osmosdr`

Make sure you are using the gnuradio installed from the apt repository. If you've installed one using pybombs or other methods remove it.

cookins, 2018-05-10:
here how i was installing everything at first time -
```
sudo apt-get install git
git clone https://github.com/pybombs/pybombs.git
cd pybombs
sudo python setup.py install
sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
sudo pybombs prefix init /usr/local/ -a gr-gsm
sudo pybombs -p gr-gsm install gr-gsm - WHERE I GOT TROUBLED WITH 

> configure: error: "Error: libcrypto required."

```
after i found that you recommend to type 

> sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default
after that I used usual manual from that github and that where I am now.

cookins, 2018-05-11:
Much thanks! That helped - why I not found out it myself?...

AlexeiScherbakov, 2018-06-25:
On latest x86 VM Kali  (2018.2) I got Segmentation Fault AFTER fixing bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=898544

Info about system:

```
root@kali:~# uname -r
4.15.0-kali2-686-pae
root@kali:~# cat /etc/debian_version
kali-rolling
root@kali:~# cat /etc/lsb-release
DISTRIB_ID=Kali
DISTRIB_RELEASE=kali-rolling
DISTRIB_CODENAME=kali-rolling
DISTRIB_DESCRIPTION="Kali GNU/Linux Rolling"
root@kali:~# dpkg -l gnuradio
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  gnuradio       3.7.13.2-2   i386         GNU Radio Software Radio Toolkit
root@kali:~# which grcc
/usr/bin/grcc
root@kali:~# grcc --version
Usage: grcc: [options] filename

grcc: error: no such option: --version
Segmentation fault
root@kali:~# cmake --version
cmake version 3.11.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

Log make VERBOSE=1
```
[ 89%] Generating grgsm_livemon
cd /home/gsm/gr-gsm/build/apps && /usr/bin/cmake -E env PYTHONPATH="/home/gsm/gr-gsm/python:/home/gsm/gr-gsm/python/misc_utils:/home/gsm/gr-gsm/python/demapping:/home/gsm/gr-gsm/python/receiver:/home/gsm/gr-gsm/python/transmitter:/home/gsm/gr-gsm/python/trx:/home/gsm/gr-gsm/build/swig" GRC_BLOCKS_PATH=/home/gsm/gr-gsm/grc /usr/bin/grcc -d /home/gsm/gr-gsm/build/apps /home/gsm/gr-gsm/apps/grgsm_livemon.grc
Segmentation fault
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
make[2]: Leaving directory '/home/gsm/gr-gsm/build'
CMakeFiles/Makefile2:1317: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make[1]: Leaving directory '/home/gsm/gr-gsm/build'
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

List of package install actions on Kali before compile of 
```
apt-get update
apt-get upgrade
apt-get install librtlsdr0 librtlsdr-dev
apt-get install gnuradio gnuradio-dev gr-osmosdr
apt-get install gqrx
apt-get install autoconf make cmake automake libtool gcc g++ python-dev swig pkg-config build-essential libtool shtool doxygen
apt-get libpcap0.8-dev libboost-all-dev libcppunit-dev liblog4cpp5-dev python-scipy libpcsclite-dev
apt-get install python-talloc-dev libtalloc-dev libtalloc2 python-talloc
apt-get install gnutls-bin gnutls-dev
```

On previous Kali version(2018.1 but x64 and live persistence) it compiled normally.

velichkov, 2018-06-25:
Hi @AlexeiScherbakov,

> On latest x86 VM Kali (2018.2) I got Segmentation Fault AFTER fixing bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=898544

> root@kali:~# dpkg -l gnuradio
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  gnuradio       3.7.13.2-2   i386         GNU Radio Software Radio Toolkit

There is a new gnuradio-3.7.13.3-1 in the Kali repository since today where the bug 898544 is fixed. Could you try with it?

I've tested in a docker containers (both Debian testing and Kali rolling) and in a Kali 2018.02 x86 VM with the same packages except the newer gnuradio and I'm not able to reproduce segmentation fault.

The new gnuradio does not depends on `python-gtk2` so if you see the following error you need to install it manually - `sudo apt-get install python-gtk2`. I've already reported in [bug=902371](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=902371)
```
[ 93%] Generating grgsm_livemon
Warning: Can't parse color code "#000" for domain "gr_message" 
Warning: Can't parse color code "#000" for domain "gr_stream" 
No module named gtk
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1401: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

velichkov, 2018-06-25:
Here are the logs from our Travis tests

https://travis-ci.org/velichkov/gr-gsm/builds/396540705
https://travis-ci.org/velichkov/gr-gsm/jobs/396540706#L4227-L4228
https://travis-ci.org/velichkov/gr-gsm/jobs/396540709#L4057-L4058

And the changes I've made 1496a2e54c52f73e3f0cc0d55b5b602a9272b022

closed 2018-07-06 by ptrkrysik

vannessaxxx, 2019-10-13:
[ 73%] Generating grgsm_livemon
<<< Welcome to GNU Radio Companion Compiler 3.9.0.0-git >>>

Block paths:
	/root/.grc_gnuradio
	/root/gr-gsm/grc
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /root/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /root/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:590: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

help!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

hackwebsecur, 2019-11-12:
Hello to all. 
I have installed all the dependencies and I get the same error when filling in "grgsm_livemon"

this is my operating system: 
`Linux root 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64 GNU/Linux`

this is the error I get: 

```
.......
WARNING:gnuradio.grc.core.platform.block_loader:Block with id "zeromq_sub_msg_source" loaded from
  /usr/share/gnuradio/grc/blocks/zeromq_sub_msg_source.block.yml
overwritten by
  /usr/share/gnuradio/grc/blocks/zeromq_sub_msg_source.block.yml
WARNING:gnuradio.grc.core.platform.block_loader:Block with id "zeromq_sub_source" loaded from
  /usr/share/gnuradio/grc/blocks/zeromq_sub_source.block.yml
overwritten by
  /usr/share/gnuradio/grc/blocks/zeromq_sub_source.block.yml
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/root/.grc_gnuradio
	/root/Utility/IMSI Catcher/IMSI-catcher/RISOLVERE/gr-gsm/grc
	/usr/share/gnuradio/grc/blocks
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /root/Utility/IMSI Catcher/IMSI-catcher/RISOLVERE/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /root/Utility/IMSI Catcher/IMSI-catcher/RISOLVERE/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:590: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

This is my gnuradio:

[root@root:~# dpkg -l gnuradio
Voluto=U (non noto)/I (installato)/R (rimosso)/P (rimosso totale)/H (in attesa)
| Stato=Non/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(nessuno)/R (reinstallazione richiesta) (Stato,Err: maiuscolo=grave)
||/ Nome           Versione     Architettura Descrizione
+++-==============-============-============-=================================
ii  gnuradio       3.8.0.0-5    amd64        GNU Radio Software Radio Toolkit
](url)
Thanks !!!

velichkov, 2019-11-12:
Hi @hackwebsecur,

> ii  gnuradio       3.8.0.0-5    amd64        GNU Radio Software Radio Toolkit

See https://groups.google.com/d/msg/gr-gsm/rAhicfOsqzw/_nuGdEJdAQAJ and please read all message in this thread and all then linked issues. If you just upgraded your system from gnuradio 3.7 to 3.8 then clean all your `build` directories and start from scratch.

Also in the future please post your questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

ecc001, 2020-03-13:
> Ok forget it, it was the sdr not recognized by the VM.
> Thank you for your help, it's working now !
>
> Below is a "how-to" for install gr-gsm on a fresh Kali install for root user.
> If it can help :) [...]

hi, i got similar issue while trying to install gr-gsm on Ubuntu 16.04.....pls how do i resolve this?

ecc001, 2020-03-13:
> Hi @Okeur75,
>
>
> The workaround for this error (#391) is to create one symlink
>
> ```
> ln -sf /usr/lib/x86_64-linux-gnu/libvolk.so.1.3.1 /usr/lib/x86_64-linux-gnu/libvolk.so.1.3
> ```
>
> and then try again [...]

please how do i resolve this issue when installing gr-gsm on ubuntu 16.04