ptrkrysik/gr-gsm #400: gr-gsm installation error on PyBOMBS

paktiger, 2018-05-03:
hi,

I am using kali linux **4.14.0-kali3-amd64 GNU/Linux** and trying to install gr-gsm by using PyBOMBS. However, fail to install gr-gsm. The following errors occurs after a long processing:

**PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.**
configure.ac:90: installing './config.guess'
configure.ac:90: installing './config.sub'
configure.ac:27: installing './install-sh'
configure.ac:27: installing './missing'
compiler/cpp/Makefile.am: installing './depcomp'
configure.ac: installing './ylwrap'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '0' is supported by ustar format... yes
checking whether GID '0' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for ANSI C header files... yes
[... 122 lines cut ...]
checking openssl/ssl.h usability... yes
checking openssl/ssl.h presence... yes
checking for openssl/ssl.h... yes
checking openssl/rand.h usability... yes
checking openssl/rand.h presence... yes
checking for openssl/rand.h... yes
checking openssl/x509v3.h usability... yes
checking openssl/x509v3.h presence... yes
checking for openssl/x509v3.h... yes
checking sched.h usability... yes
checking sched.h presence... yes
checking for sched.h... yes
checking wchar.h usability... yes
checking wchar.h presence... yes
checking for wchar.h... yes
checking for pthread_create in -lpthread... yes
checking for clock_gettime in -lrt... yes
checking for setsockopt in -lsocket... no
checking for BN_init in -lcrypto... no
**configure: error: "Error: libcrypto required."
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.**
Please need a detail explanation

velichkov, 2018-05-03:
Hi @paktiger,

My advice is to try to install it using the steps from the [Kali.docker](../blob/master/tests/dockerfiles/Kali.docker) file, it is a lot more easier then pybombs. See also https://github.com/ptrkrysik/gr-gsm/issues/399#issuecomment-385779646

> PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.
> configure: error: "Error: libcrypto required."

If for some reason you need to install it using `pybombs` then initialize the  prefix using the following command (From https://github.com/ptrkrysik/gr-gsm/wiki/Installation)
```
sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default
```
In the `gnuradio-default` receipt the `apache-thrift` is marked as optional so the it will be skipped.

See also:
gnuradio/gr-recipes/pull/105
gnuradio/gr-recipes/issues/88
gnuradio/gnuradio/issues/1444

velichkov, 2018-05-03:
> i am having problem with these command, sorry i am not very good at linux commands:

Then try https://github.com/ptrkrysik/gr-gsm/issues/399#issuecomment-385779646

paktiger, 2018-05-12:
bro cmake .. this command is not working on my PC while install grgsm.

apt update
apt upgrade
apt install git libboost-all-dev swig cmake autoconf libtool pkg-config build-essential python-docutils libcppunit-dev doxygen liblog4cpp5-dev python-scipy gnuradio-dev gr-osmosdr libosmocore-dev gr-osmosdr gnuradio  libusb-1.0-0 libusb-1.0-0-dev libfftw3-dev python-numpy
git clone https://github.com/ptrkrysik/gr-gsm.git
cd gr-gsm
mkdir build
cd build
**cmake ..** here it gives me error:
bash: cmake: command not found

velichkov, 2018-05-15:
Hi @paktiger,

> bash: cmake: command not found

Try to reinstall cmake with `apt-get install --reinstall cmake` and if it still fails provide the full output of 
```
dpkg -l cmake
dpkg -L cmake
dpkg --verify cmake
echo $PATH
which cmake
```

closed 2018-05-24 by ptrkrysik

ia7204280, 2019-03-15:
sudo pybombs install gr-gsm
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap

ia7204280, 2019-03-15:
i m facing such fucking error plz help...

ia7204280, 2019-03-15:
> Hi @paktiger,

root@kali:~# sudo pybombs install gr-gsm
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap
  
i m facing this issue please help me.....

Regards

velichkov, 2019-03-16:
Hi @ia7204280,

Have you read all messages in this issue before posting your message?
Why did you choose to install `gr-gsm` with `pybombs`? Are you following some tutorial? Please provide a link is so.
Have you read our [Installation instruction](https://osmocom.org/projects/gr-gsm/wiki/Installation)  before attempting to install it?
 
> PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap

It looks like your `pybombs` installation is broken as the autotools-bootstrap receipt is part of `pybombs`. See https://github.com/gnuradio/pybombs/blob/master/pybombs/recipes/autotools-bootstrap.lwr

How did you install `pybombs`? Try reinstalling it.