ptrkrysik/gr-gsm #402: grgsm/misc_utils/trx_burst_if.h error

stevebirtwistle, 2018-05-05:
Following the move of trx to the trx directory from misc_utils it would appear there is still a reference to the previous file. This is resulting in an error being thrown in making the project.

In file included from /home/user/Documents/gr-gsm/lib/trx/trx_burst_if_impl.cc:31:0:
/home/user/Documents/gr-gsm/lib/trx/trx_burst_if_impl.h:29:10: fatal error: grgsm/misc_utils/trx_burst_if.h: No such file or directory
 #include <grgsm/misc_utils/trx_burst_if.h>

pachulo, 2018-05-06:
This is the error I'm seeing:

```
[ 63%] Building CXX object lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o

In file included from /gr-gsm/lib/trx/trx_burst_if_impl.cc:31:0:
/gr-gsm/lib/trx/trx_burst_if_impl.h:29:43: fatal error: grgsm/misc_utils/trx_burst_if.h: No such file or directory
compilation terminated.

lib/CMakeFiles/grgsm.dir/build.make:1478: recipe for target 'lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o' failed
CMakeFiles/Makefile2:347: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed
Makefile:138: recipe for target 'all' failed

make[2]: *** [lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2
make: *** [all] Error 2
```

ptrkrysik, 2018-05-06:
@stevebirtwistle, thanks for reporting the issue. It was my fault, I should have tested the commit first in a separate branch. It was fixed by fd3d883d4e450.

closed 2018-05-06 by ptrkrysik