ptrkrysik/gr-gsm #403: Error compiling on Kali 2018.2

Giorgiofox, 2018-05-06:
Hi!
[ 59%] Building CXX object lib/CMakeFiles/grgsm.dir/transmitter/gen_test_ab_impl.cc.o
[ 60%] Building CXX object lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o
In file included from /root/gr-gsm/lib/trx/trx_burst_if_impl.cc:31:0:
/root/gr-gsm/lib/trx/trx_burst_if_impl.h:29:10: fatal error: grgsm/misc_utils/trx_burst_if.h: No such file or directory
 #include <grgsm/misc_utils/trx_burst_if.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

I have this error compiling from the source, 
and If I try PyBombs

PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.get_recipe - ERROR - Error fetching recipe `gr-gsm':
Package gr-gsm has no recipe file!

Can anyone please help me to get GR-GSM on Kali?
The version on main repository seem buggy 

Thanks
Giorgio

Giorgiofox, 2018-05-06:
Editing the recipe I'm now able to trying compile with PyBombs, 
but:

[ 77%] Built target _grgsm_swig
[ 79%] Built target pygen_python_bc8f2
[ 82%] Built target pygen_python_misc_utils_6b490
[ 85%] Built target pygen_python_receiver_89d28
[ 87%] Built target pygen_python_demapping_89c33
[ 89%] Generating grgsm_livemon
Traceback (most recent call last):
  File "/usr/bin/grcc", line 29, in <module>
    from gnuradio import gr
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/__init__.py", line 41, in <module>
    from runtime_swig import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 17, in <module>
    _runtime_swig = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_runtime_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _runtime_swig
apps/CMakeFiles/pygen_apps.dir/build.make:79: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1078: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-gsm:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gr-gsm. Aborting.
root@kali:~/.pybombs/recipes/gr-etcetera#

ptrkrysik, 2018-05-06:
@Giorgiofox, thank you for reporting the issue. The first one was solved by fd3d883d4e. As for the second one - try doing this:
sudo ln -sf /usr/lib/x86_64-linux-gnu/libvolk.so.1.3.1 /usr/lib/x86_64-linux-gnu/libvolk.so.1.3

It is workaround used in the test here:
https://github.com/ptrkrysik/gr-gsm/blob/development/tests/dockerfiles/Kali.docker

closed 2018-05-06 by ptrkrysik