ptrkrysik/gr-gsm #408: Ask about grgsm_decode result

jamessagan, 2018-06-01:
Dear
As I knew, Airprobe does the same function as GR-GSM, and Airprobe throws out the result like "sch.c:260 ERR: conv_decode 1
cch.c:419 error: sacch: parity error (-1 fn=2691175)
gsmstack.c:301 cannot decode fnr=0x291067 (2691175) ts=1" for the frame which could not be decoded by Airprobe. 
I want to ask about GR-GSM's result: can gr-gsm decode all the frame (burst) and print out the terminal console? Or it does not print if out the frame if it can not decode?

velichkov, 2018-06-01:
Hi @jamessagan,

Please post general questions to the [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list).

> As I knew, Airprobe does the same function as GR-GSM

Yes, `gr-gsm` is more or less a successor of `airprobe`.

> I want to ask about GR-GSM's result: can gr-gsm decode all the frame (burst) and print out the terminal console? 

Yes, you could use the `Bursts Printer` block to print the bursts.
https://github.com/ptrkrysik/gr-gsm/blob/fcb45b021a1634ef097f16f1bcae79dba7f506db/apps/grgsm_decode#L118-L119
https://github.com/ptrkrysik/gr-gsm/blob/fcb45b021a1634ef097f16f1bcae79dba7f506db/apps/grgsm_decode#L138-L139

> Or it does not print if out the frame if it can not decode?

No messages are printed when decoding a frame fails.

jamessagan, 2018-06-02:
Thank Velichkov.
I want to ask above  because I used grgsm_capture to sniff my call with sample rate 2M. After that, I decoded with grgsm_decode for BCCH channel with timeslot 0, I got immediate assignment frame with SDCCH8 and timeslot 1. I decoded SDCCH8 with timeslot 1 but I can not find any cipher mode command package.
So I want to know that which type of error that gr-gsm can not decode and does not print to console? How can I get these burst bit, frame number or burst count? Can gr-gsm support this issues?

velichkov, 2018-06-02:
Hi @jamessagan,

> I can not find any cipher mode command package.

Which messages do you find instead? Could you share the `pcap` file?

> So I want to know that which type of error that gr-gsm can not decode and does not print to console? 

As you have already seen no errors are printed to the console.

> How can I get these burst bit, frame number or burst count?

You could add some error logs in the [GSM Receiver block](https://github.com/ptrkrysik/gr-gsm/blob/development/lib/receiver/receiver_impl.cc)

jamessagan, 2018-06-02:
Dear Velichkov
_Which messages do you find instead?_ 
I don't understand which you mean.

_Could you share the pcap file?_
Sorry I cannot upload this file, it so big.

_As you have already seen no errors are printed to the console._
Do you mean that i decoded successfully for all frame and no error?

_You could add some error logs in the GSM Receiver block_
I will try

Thank you very much.

velichkov, 2018-06-02:
Hi @jamessagan,

>>> I decoded SDCCH8 with timeslot 1 but I can not find any cipher mode command package.
>>
>> Which messages do you find instead?
>
> I don't understand which you mean.

You said that you can't find Cipher Mode Command but most probably you get some SDCCH8 messages decoded, so for example do you find System Information Type 5/6 or not?

If you try to decode the example .cfile from [the wiki](https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To) are you able to decode Cipher Mode Command?

>> Could you share the pcap file?
>
> Sorry I cannot upload this file, it so big.

You could export just few messages I guess. 
Which Wireshark version you are using? 
Try with wireshark 2.2.6 or older or compile it from its latest master branch

>> As you have already seen no errors are printed to the console.
>
> Do you mean that i decoded successfully for all frame and no error?

I meant that nothing is printed when there is an error.

jamessagan, 2018-06-02:
Dear Velichkov
I can find System Information Type 5 with WireShark (ver 2.2.6). For example .cfile from the Wiki I can find Cipher Mode Command also.

These files below are results I capture by wireshark 2.2.6 for decoding BCCH timeslot 0 and SDCCH8 timeslot 1. Sample is 2000000.
[decode_bcch_timeslot0.txt](https://github.com/ptrkrysik/gr-gsm/files/2065044/decode_bcch_timeslot0.txt)
[decode_sdcch8_timeslot1.txt](https://github.com/ptrkrysik/gr-gsm/files/2065045/decode_sdcch8_timeslot1.txt)

velichkov, 2018-06-02:
> 14:11:56,922,953   ETHER
> |0   |00|00|00|00|00|00|00|00|00|00|00|00|08|00|45|00|00|43|e7|06|40|00|40|11|55|a1|7f|00|00|01|7f|00|00|01|bd|b0|12|79|00|2f|fe|42|02|04|01|01|00|00|e6|00|00|1b|a5|e0|08|14|02|bf|2f|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|2b|
>
> +---------+---------------+----------+

I can't open this with Wireshark and I'm not that good to manually decoding them. Open your big pcap file with Wireshark, go to File -> Export Specified Packages .... and export some of the SDCCH8 messages that you were able to find and upload the pcap file.

jamessagan, 2018-06-02:
Dear Velichkov
Why can you not open these files by wireshark. Because of permission? I have just change owner, so you may try to open it with wireshark again.

I also send you the link of CFILE. You can try to decode by your self. It's size is about 1.6GB:
[decode_bcch_timeslot0.txt](https://github.com/ptrkrysik/gr-gsm/files/2065109/decode_bcch_timeslot0.txt)
[decode_sdcch8_timeslot1.txt](https://github.com/ptrkrysik/gr-gsm/files/2065110/decode_sdcch8_timeslot1.txt)

grgsm_decode -c 943.6MHz_ARFCN43_sample2M_g40.cfile -f 943600000 -m BCCH -t 0 -p -s 2000000
grgsm_decode -c 943.6MHz_ARFCN43_sample2M_g40.cfile -s 2000000 -f 943.6M -m SDCCH8  -t 1 -v -p

CFILE link: https://drive.google.com/open?id=1unPSeYejYoED-57oU7fYjHqunez41Qbe

jamessagan, 2018-06-03:
Oh, I found that the channel is hopping. Is it the reason?

closed 2019-08-21 by ptrkrysik

ChillVibesMushroom, 2022-10-01:
Wireshark can't open the cfile when I try to open it what am I doing wrong?