ptrkrysik/gr-gsm #409: Improve/simplify grgsm_capture

ptrkrysik, 2018-06-02:
Grgsm-capture currently is too complicated on one hand and on the other it doesn't allow wideband capture. Remove functionality that is not absolutely needed and simplify it in order for it to allow wideband capture. Add options to select sample type (complex float, complex int16).

Something was about it was mentioned here: #216

mazen-bassiouny, 2018-06-29:
sorry what happened here ? is the wide band feature is available now ?

mazen-bassiouny, 2018-06-30:
hello Ptrkrysik,
i added the option in grgrsm_capture to use larger bandwidth , i captured 10Mhz bandwidth in a range which i am sure that has BTSs and tested that by grgsm scanner ,, now trying to decode the BCCH using grgsm_decode with the m option BCCH but nothing ,, maybe you help with some troubleshootings ,, i find ccch messages on wireshark and one of them is Bcch but no system information are there

velichkov, 2018-07-01:
Hello @mazen-bassiouny,

It would be easier if you share your changes - [fork the repo](https://help.github.com/articles/fork-a-repo/) and [push your changes to your fork](https://help.github.com/articles/pushing-to-a-remote/)

Also if possible upload somewhere the cfile you are testing with and post a link here.

closed 2019-08-21 by ptrkrysik