ptrkrysik/gr-gsm #410: not able to use this in linux mint .

jinnu92, 2018-06-05:
hi 
i ma new in this kind of stuff  . 
i have installed everything properly . 
complied successfully without error  . 
and make installed successfully .

but when i use this i got this error 

> _Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture", line 32, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: libgrgsm.so.0.41.4git: cannot open shared object file: No such file or directory_

>

velichkov, 2018-06-05:
Hi @jinnu92,

> not able to use this in linux mint 

Which version ?

> i have installed everything properly .

How did you install gnuradio - from the linux mint repository or using pybombs?

> complied successfully without error .
and make installed successfully .

How did you install gr-gsm - manually or using pybombs?

> ImportError: libgrgsm.so.0.41.4git: cannot open shared object file: No such file or directory

Try running - `sudo ldconfig`

jinnu92, 2018-06-05:
thankx for your attention.  i will do this tomorrow morning.

jinnu92, 2018-06-05:
i have installed gnu radio from mint repo. 
and complied grgsm from source.

velichkov, 2018-06-05:
Which release/version of Linux Mint you are using?  See https://linuxmint.com/download_all.php
What happens after you run - `sudo ldconfig`?

jinnu92, 2018-06-06:
thankx.  that's work fine for me. everything working now.

closed 2018-06-06 by ptrkrysik