ptrkrysik/gr-gsm #415: Error: No module named _blocks_swig0

RaNdOm0313, 2018-06-14:
Having a little issue when running grgsm_livemon

I've tried everything i could find and yet, nothing.

---ERR MSG---

Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon_headless", line 31, in <module>
    from gnuradio import blocks
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/__init__.py", line 32, in <module>
    from blocks_swig import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig.py", line 22, in <module>
    from blocks_swig0 import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 17, in <module>
    _blocks_swig0 = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 16, in swig_import_helper
    return importlib.import_module('_blocks_swig0')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _blocks_swig0

velichkov, 2018-06-14:
Hi @RaNdOm0313,

Which Linux distribution and version you are using?
How did you install gnuradio - from the official distribution repository or using pybombs, and which version?
How did you install gr-gsm - manually or using pybombs?
Are you able to run the `gnuradio-companion`?

P.S
Please note that the [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list) is the proper place to report the installation problems.

velichkov, 2018-06-15:
> gnuradio- installed it from source

Try to uninstall it and install the one from the apt repository - `apt-get install gnuradio-dev` and after that rebuild and re-install `gr-gsm`

Also have a look at  our docker files
https://github.com/ptrkrysik/gr-gsm/blob/084b14ea17f902c607aa295e18deee3ed32deb07/tests/dockerfiles/Debian_testing.docker#L1-L18

closed 2018-06-18 by ptrkrysik