ptrkrysik/gr-gsm #416: Configuring incomplete, errors occurred!

nishantparhi, 2018-06-17:
root@anonymous:~/gr-gsm/build# cmake ..
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Extracting version information from git describe...
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
CMake Warning at CMakeLists.txt:135 (find_package):
  By not providing "FindGnuradio.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gnuradio",
  but CMake did not find one.

  Could not find a package configuration file provided by "Gnuradio" with any
  of the following names:

    GnuradioConfig.cmake
    gnuradio-config.cmake

  Add the installation prefix of "Gnuradio" to CMAKE_PREFIX_PATH or set
  "Gnuradio_DIR" to a directory containing one of the above files.  If
  "Gnuradio" provides a separate development package or SDK, be sure it has
  been installed.

-- Checking for module 'volk'
--   No package 'volk' found
-- Could NOT find VOLK (missing: VOLK_LIBRARIES VOLK_INCLUDE_DIRS) 
-- Checking for module 'libosmocore'
--   No package 'libosmocore' found
-- Checking for module 'libosmogsm'
--   No package 'libosmogsm' found
-- Could NOT find libosmocore (missing: LIBOSMOCORE_LIBRARY LIBOSMOCORE_INCLUDE_DIR) 
-- Checking for module 'libosmocodec'
--   No package 'libosmocodec' found
-- Could NOT find libosmocodec (missing: LIBOSMOCODEC_LIBRARY LIBOSMOCODEC_INCLUDE_DIR) 
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:145 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/CMakeFiles/CMakeOutput.log".

what to do.

velichkov, 2018-06-17:
Hi @nishantparhi,

> GnuRadio Runtime required to compile gr-gsm

It seems you have not installed all development dependencies, you could find the full dependencies list in our docker files. 

https://github.com/ptrkrysik/gr-gsm/blob/084b14ea17f902c607aa295e18deee3ed32deb07/tests/dockerfiles/Debian_testing.docker#L4-L18

nishantparhi, 2018-06-17:
Thank you @velichkov .
It woked for me.

nishantparhi, 2018-06-17:
File "/usr/local/bin/grgsm_decode", line 30, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 17, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_grgsm_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _grgsm_swig

Now when I run grgsm_(anything) its shows this type of error.

velichkov, 2018-06-17:
Run `sudo ldconfig`

nishantparhi, 2018-06-17:
Thank you. 
It worked.

closed 2018-06-17 by nishantparhi