ptrkrysik/gr-gsm #418: Wrong MCC codes for Kosovo?

MatejKovacic, 2018-07-04:
I ran a scan in Kosovo, and I get this:

```
ARFCN:    2, Freq:  935.4M, CID: 36797, LAC:   300, MCC: 212, MNC:   1, Pwr: -64
ARFCN:    6, Freq:  936.2M, CID: 26797, LAC:   300, MCC: 212, MNC:   1, Pwr: -27
ARFCN:    9, Freq:  936.8M, CID:     0, LAC:   300, MCC: 212, MNC:   1, Pwr: -35
ARFCN:   13, Freq:  937.6M, CID: 26597, LAC:   300, MCC: 212, MNC:   1, Pwr: -33
ARFCN:   16, Freq:  938.2M, CID:     0, LAC:   300, MCC: 212, MNC:   1, Pwr: -38
ARFCN:   20, Freq:  939.0M, CID: 36997, LAC:   300, MCC: 212, MNC:   1, Pwr: -44
ARFCN:  112, Freq:  957.4M, CID:   501, LAC:   101, MCC: 293, MNC:  41, Pwr: -49
ARFCN:  114, Freq:  957.8M, CID:   522, LAC:   101, MCC: 293, MNC:  41, Pwr: -32
ARFCN:  116, Freq:  958.2M, CID:   523, LAC:   101, MCC: 293, MNC:  41, Pwr: -29
ARFCN:  118, Freq:  958.6M, CID:   311, LAC:   101, MCC: 293, MNC:  41, Pwr: -35
ARFCN:   -1, Freq:  960.6M, CID:   311, LAC:   101, MCC: 293, MNC:  41, Pwr: -44
ARFCN:  119, Freq:  958.8M, CID:   293, LAC:   101, MCC: 293, MNC:  41, Pwr: -39
ARFCN:  120, Freq:  959.0M, CID:   502, LAC:   101, MCC: 293, MNC:  41, Pwr: -45
ARFCN:  121, Freq:  959.2M, CID:   292, LAC:   101, MCC: 293, MNC:  41, Pwr: -34
ARFCN:  122, Freq:  959.4M, CID:    92, LAC:   101, MCC: 293, MNC:  41, Pwr: -37
ARFCN:  124, Freq:  959.8M, CID:    93, LAC:   101, MCC: 293, MNC:  41, Pwr: -29
```
This seems a little bit strange, since 212 is code for Monaco and 293 for Slovenia... However, I have found this:
https://en.wikipedia.org/wiki/List_of_mobile_network_operators_of_Europe#Kosovo

Is it possible that they in fact still transmit those numbers (MCC), or this is something which is hardcoded to grgsm_scanner and just needs to be updated?

For 3 frequencies I also have livemon capture for a few minutes, it it is needed...

velichkov, 2018-07-04:
Hi @MatejKovacic,

> Is it possible that they in fact still transmit those numbers (MCC), 

Yes, it is possible.
> this is something which is hardcoded to grgsm_scanner and just needs to be updated?
 
No, it's not hardcoded. You could validate this using Wireshark, caputre on the `lo` interface on port 4729 while grgsm_livemon is running and use `e212.lai.mcc` to filter only messages containing MCC
 
[image]

Here in the lower right pane you could see how 284 is encoded as 82 f4

ptrkrysik, 2018-07-26:
Hi guys. So is MCC decoded correctly after all? I remember there were some problems with it before but I'm not sure if they were fixed for good.