ptrkrysik/gr-gsm #419: Error running grgsm_capture

MatejKovacic, 2018-07-04:
`grgsm_capture.py -f 936.2e6 -c output_file_936-2M.cfile -s 2e6 -T 3 -p 35`

```
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture.py", line 257, in <module>
    if grgsm.arfcn.is_valid_downlink(options.fc, band):
TypeError: is_valid_downlink() takes exactly 1 argument (2 given)
Napaka segmentacije (izpis jedra)
```

Last line means Segmentation fault.

velichkov, 2018-07-04:
Hi @MatejKovacic,

It seems you are using an old version, this problem is fixed (#332) in current `development` and `master` branches in 1a35cb973fba5c770f7c14a769c0b1b9d329e764. My advice is to update to latest `development` branch.

closed 2018-07-12 by ptrkrysik

samnq, 2020-04-11:
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture.py", line 257, in <module>
    if grgsm.arfcn.is_valid_downlink(options.fc, band):
TypeError: is_valid_downlink() takes exactly 1 argument (2 given)

i couldn't solve this , any help ?