ptrkrysik/gr-gsm #420: Got error from printing decode result

jamessagan, 2018-07-17:
Dear GR-GSM team.
When I was decoding my cfile (which was captured by grgsm_capture) with -v and -p optional, I got some result as below:

1987121 3068179: 100000000101110101010100000000000001000111110101010100001000000100010111010100010000000000010001011101010100000000
1987122 3068212: 101010111111110101000100101010101110111101010100000000101010101111111101010001000010101010101111011101010000001000
1987123 3068245: 000101010111010101000000100000000001010101010001000010000001000101111101010100000010000000010111010101000000001000
1987124 3068278: 010100001010101110111111010100010000001010101011111111010101000010001010111111110101010100001010101010111101010101
1987121 3068179:  2f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
1987164 3067914: 101000010001111111010101010000000000000111111101010001000010100001010101011100010100100010010100010101010101000000
1987165 3067947: 101011111111110101000000100000101011111101111100010100000010101110111101110001000000000010101011011111010000011000
1987166 3067980: 000100010111110100010100101010000100011111110001000010001001010101011101010000001010000000000001010101000001000010
**_1987167 3068013: 0101000000000010111011111111000101001000001011111111111101000100101987164 3067914:  03 44 0d 06 35 05 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
001010111110010101010001001010101010101101011101_**
1987172 3068178: 100000010001110101010000000010100000000111111101010000001010000100010111010100000000101000010000010101010100000010
1987173 3068211: 101010111111111101000000101010101111111111110100000000100010111111111111010101000000001010101011011101010000001000
1987174 3068244: 000000011111010101010000100000010001010111010101000010100001010001111101010001000010000000000101110101010100000010
1987175 3068277: 000100001010101010111101110101010000000010101110111111010100010000001010101011011101010001000010001011101111010101
1987172 3068178:  03 03 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b

I think there are some problem in gr-gsm's code when it print result to console by using optional -v and -p.

Can you check it or tell my where is the code which I can check.

Thank and best regards

velichkov, 2018-07-17:
Hi @jamessagan,

See the discussion in #255

The code that prints messages calls std::cout multiple times and that's why the output from multiple threads gets mixed,

> Can you check it or tell my where is the code which I can check.

https://github.com/ptrkrysik/gr-gsm/blob/eb81b03c02833f2ebb002d3c364b7a21fdae041c/lib/misc_utils/message_printer_impl.cc#L41-L83
https://github.com/ptrkrysik/gr-gsm/blob/eb81b03c02833f2ebb002d3c364b7a21fdae041c/lib/misc_utils/bursts_printer_impl.cc#L55-L111

One possible solution is to first construct the message in std::stringstream and then print it all at once using cout. http://www.cplusplus.com/reference/sstream/stringstream/stringstream/#example

jamessagan, 2018-07-19:
Thank Velichkov.

One more thing: I have just found one branch of gr-gsm, which is https://github.com/osmocom/gr-gsm/tree/master.
What is deffirent between https://github.com/osmocom/gr-gsm/tree/master and https://github.com/ptrkrysik/gr-gsm/tree/master. osmocom/gr-gsm seems to be getting more updates than ptrkrysik/gr-gsm. Which is better than? Why are there two branch like this?

velichkov, 2018-07-19:
Hi @jamessagan 

> One more thing: I have just found one branch of gr-gsm, which is https://github.com/osmocom/gr-gsm/tree/master.

Piotr plans to move the project to osmocom and the new project home will be https://osmocom.org/projects/gr-gsm and http://git.osmocom.org/gr-gsm. The github link you posted is just a read only mirror of the osmocom repository.

> What is deffirent between https://github.com/osmocom/gr-gsm/tree/master and https://github.com/ptrkrysik/gr-gsm/tree/master. osmocom/gr-gsm seems to be getting more updates than ptrkrysik/gr-gsm.

For now both repository are functionally identical and there are no new features in the osmocom repository as far as I know. Personally I still use the `development` branch from this repository.

jamessagan, 2018-07-20:
Thank Velichkov

commit 75ae9cc3 (2018-09-11): Format the output into ostringstream and then write it at once.

This fixes the garbled output when multiple printers are used in a
flowgraph.

closes #255
closes #420

Change-Id: I1012ed26371b4c67163545652f0a1ce0f576af9e
closed 2018-09-11 by osmocom-gerrit