ptrkrysik/gr-gsm #421: grgsm_scanner duplicated and wrong results

mazen-bassiouny, 2018-07-23:
Dears,
I am getting duplicated results with the scanner application , same cell Id on different frequencies which cannot be correct.,and tested with mobile application.
For example: these are the results of scanning GSM900 band 
root@mazen-ThinkPad-T440p:/usr/local/src/gr-gsm/apps# grgsm_scanner -b GSM900

ARFCN:  995, Freq:  929.2M, CID: 26821, LAC: 36150, MCC: 262, MNC:   3, Pwr: -68
ARFCN: 1006, Freq:  931.4M, CID:  9058, LAC: 36150, MCC: 262, MNC:   3, Pwr: -62
ARFCN: 1010, Freq:  932.2M, CID:  6822, LAC: 36150, MCC: 262, MNC:   3, Pwr: -59
ARFCN: 1017, Freq:  933.6M, CID: 46821, LAC: 36150, MCC: 262, MNC:   3, Pwr: -57
ARFCN:   19, Freq:  938.8M, CID: 19401, LAC:  4291, MCC: 262, MNC:   2, Pwr: -59
ARFCN:   29, Freq:  940.8M, CID: 19401, LAC:  4291, MCC: 262, MNC:   2, Pwr: -56
ARFCN:   53, Freq:  945.6M, CID: 13522, LAC: 38722, MCC: 262, MNC:   1, Pwr: -52
ARFCN:   57, Freq:  946.4M, CID:  3322, LAC: 38722, MCC: 262, MNC:   1, Pwr: -44
ARFCN:   63, Freq:  947.6M, CID: 13522, LAC: 38722, MCC: 262, MNC:   1, Pwr: -25
ARFCN:   67, Freq:  948.4M, CID:  3322, LAC: 38722, MCC: 262, MNC:   1, Pwr: -53
ARFCN:   73, Freq:  949.6M, CID: 13522, LAC: 38722, MCC: 262, MNC:   1, Pwr: -57
ARFCN:  110, Freq:  957.0M, CID:     0, LAC:     0, MCC:   0, MNC:   0, Pwr: -44
ARFCN:  120, Freq:  959.0M, CID: 13622, LAC: 38722, MCC: 262, MNC:   1, Pwr: -38

Noting for CID :13522 ,,, it comes on different ARFCNs although the correct one is only (63)
also          CID: 19401 ,,,, it comes on 3 ARFCNs although the correct one is only (29)

can you please check ?

velichkov, 2018-07-25:
Hi @mazen-bassiouny,

Thanks for reporting this and thanks for the reminder. :wink:  Have you tried to debug it? Based on our previous discussions I think you have all the knowledge and skills to resolve this.

> ARFCN: 19, Freq: 938.8M, CID: 19401, LAC: 4291, MCC: 262, MNC: 2, Pwr: -59
ARFCN: 29, Freq: 940.8M, CID: 19401, LAC: 4291, MCC: 262, MNC: 2, Pwr: -56

> ARFCN: 53, Freq: 945.6M, CID: 13522, LAC: 38722, MCC: 262, MNC: 1, Pwr: -52
ARFCN: 63, Freq: 947.6M, CID: 13522, LAC: 38722, MCC: 262, MNC: 1, Pwr: -25
ARFCN: 73, Freq: 949.6M, CID: 13522, LAC: 38722, MCC: 262, MNC: 1, Pwr: -57

> ARFCN: 57, Freq: 946.4M, CID: 3322, LAC: 38722, MCC: 262, MNC: 1, Pwr: -44
ARFCN: 67, Freq: 948.4M, CID: 3322, LAC: 38722, MCC: 262, MNC: 1, Pwr: -53

There is 2MHz between each duplicated CID and knowing that the default sampling rate is 2e6 (=2MHz) my first guess is that you are experiencing some kind of [aliasing](https://en.wikipedia.org/wiki/Aliasing) here.

Try with 1.6e6, 2.4e6, 3.2e6, 4e6, and 8e6 sample rates and post the full output.

ptrkrysik, 2018-07-26:
I was looking into this problem and it seems that it is result of aliasing in analog fron-end of RTL-SDR. As grgsm_scanner channelizes whole available spectrum aliases can be seen as false-positives. There are probably many solutions to this problem, i.e.:
* changing frequency scanning algorithm so the scanner don't look at parts of sampled bandwidth where aliases appear,
* filtering the false-positives by confirming ARFCN with use of decoded data,
* improving speed of detecting a C0 channel (i.e. by doing frequency calibration beforehand and looking only for SCH bursts) while decreasing scanned bandwidth (i.e. to single channel)

I'm quite reluctant to do it myself at the moment due to other tasks.