ptrkrysik/gr-gsm #422: grgsm_decode  export gsm  file incomplete

ericwang321, 2018-07-24:
ubuntu 16.04 Linux bogon 4.4.0-130-generic
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
-----
#grgsm_decode BCCH ,SDCCH .function is normal.   
but when do 
#grgsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile  -s $((100000000/174)) -a 725 -m TCHF -t 5 -k 0x1E,0xF0,0x0B,0xAB,0x3B,0xAC,0x70,0x02 -o /tmp/1.gsm -v
it hangs at:
864005 1333721:  04 00 03 03 49 06 1d 9f 6d 18 10 80 00 00 00 00 00 00 00 00 00 00 00
864084 1334592:  03 82 19 13 2d 08 02 e0 90 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
864114 1333924:  01 a1 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
864136 1334624:  03 a4 11 06 0d 00 c1 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
864162 1333824:  01 73 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b

i have to press ctrl+c to stop this process.
----
at the same time ,run 
#wireshark -k -f udp -Y gsmtap -i lo
wireshark appear decode 25  packets. 
# toast -d 1.gsm
toast: incomplete frame (21 bytes missing) from 1.gsm
------
how to solve this problem ?how to find  error reason ?  
thanks.

velichkov, 2018-07-24:
Hi @ericwang321,

> it hangs at:
> i have to press ctrl+c to stop this process.

This is a known [gnuradio issue](https://lists.gnu.org/archive/html/discuss-gnuradio/2016-07/msg00319.html) and it is fixed in gnuradio/gnuradio/pull/797 which is available in the [gnuradio's next branch](https://github.com/gnuradio/gnuradio/tree/next)

>  toast -d 1.gsm
>  toast: incomplete frame (21 bytes missing) from 1.gsm

I'm not sure what `toast -d` is suppose to do but I'm able to play the `1.gsm` file as is with `vlc`
In #107 it is written that the file must have an au.gsm extension to be properly decoded.

See also #106 and https://groups.google.com/d/msg/gr-gsm/HQsvTQkIp6M/HQYD11j1CAAJ

ericwang321, 2018-08-02:
Thanks !
 

1. cannot stop【next step？】

  Thought grgsm_decode cannot stop,but it's export function is work normal.

---(how to solved problem?)
   cd gnuradio src directory ,change to new branch ,re-compile the code. 
   how to do  next  step ,is there install  instruction clealy? 
   I am worried about messing up dependencies between code
   use PyBOMBS script (github) install grgsm,the  script  is possible to update ?

2. play sound .【worked】

I read the link 106，107 carefully，the i found  I write the wrong parameter.
grgsm decode  the file with  voice coding EFR  ，not FR.

velichkov, 2018-08-03:
> cd gnuradio src directory ,change to new branch ,re-compile the code.
how to do next step ,is there install instruction clealy?

Better ask this question in the [gnuradio's mailing list](https://wiki.gnuradio.org/index.php/MailingLists)

ericwang321, 2018-08-06:
thanks! 
I will post  to mail list about install question.

closed 2018-08-06 by ericwang321