ptrkrysik/gr-gsm #425: libosmocore install fails, syntax error at /usr/local/bin/automake

YellaMehroze, 2018-08-07:
I tried to manually install libosmocore and when I execute the command "autoreconf -i -v" it throws an error. 

Output: 
=====

autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I m4 --output=aclocal.m4t
autoreconf: 'aclocal.m4' is unchanged
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake
syntax error at /usr/local/bin/automake line 986, near "$name ("
syntax error at /usr/local/bin/automake line 1147, near "$subr ("
BEGIN not safe after errors--compilation aborted at /usr/local/bin/automake line 3348.
autoreconf: automake failed with exit status: 255

YellaMehroze, 2018-08-07:
Thanks @velichkov , So shall I remove the existing packages and reinstall ? Please also note I am using Ubuntu on Windows 10 linux sub system.

velichkov, 2018-08-07:
> Please also note I am using Ubuntu on Windows 10 linux sub system.

Sorry, I'm not familiar with this Ubuntu on Windows thing.

> So shall I remove the existing packages and reinstall 

I would say install automake from the distribution package repository.

P.S.
Better report libosmocore compilation problems in the osmocom's mailing lists as this repository contains the gr-gsm project and not libosmocore

closed 2019-08-21 by ptrkrysik