ptrkrysik/gr-gsm #426: File "/usr/bin/grgsm_scanner", line 212

BRRY876, 2018-08-08:
i followd all the steps for debian and still gettingthis erro...i am using kali 2018.2

File "/usr/bin/grgsm_scanner", line 212
    // Enabling this line causes segfault with HackRF
     ^
SyntaxError: invalid syntax

velichkov, 2018-08-09:
Hi @BRRY876,

> i followed all the steps for debian and still getting this error...i am using kali 2018.2

Could you share a link to the steps you've followed. 
Which gr-gsm version do you use?

> File "/usr/bin/grgsm_scanner", line 212
// Enabling this line causes segfault with HackRF
^
SyntaxError: invalid syntax

If you are trying to comment out line 212 the correct way is with `#` and not with `//`

https://github.com/ptrkrysik/gr-gsm/blob/332e2be78f9cf82a7b6a0f053d2bfd87974b9998/apps/grgsm_scanner#L211-L213

BRRY876, 2018-08-09:
i followed the steps on wiki 
Debbian-based distributions (Debian Testing, Ubuntu 16.04+, Kali Rolling Edition)

Install all needed prerequisites with following command ('sudo' is needed on Ubuntu based distributions, on other distributions it should be omitted and commands starting with 'sudo' should be executed with superuser rights):

sudo apt-get update && \
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev

Then download the gr-gsm's source and build it with following commands:

git clone https://git.osmocom.org/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
make
sudo make install
sudo ldconfig

i wasnt not trying to comment out anything i just installed it and used grgsm_scanner in terminal and came up with the HACKRF in line 212

BRRY876, 2018-08-09:
that is what in line 212        
#self.rtlsdr_source.set_min_output_buffer(int(sample_rate*rec_len)) #this line causes segfaults on HackRF

velichkov, 2018-08-09:
Hi @BRRY876,

> Then download the gr-gsm's source and build it with following commands:

> cmake ..

> File "/usr/bin/grgsm_scanner", line 212

Most probably at some point you have installed the `gr-gsm` package from the `apt` repository as well, the package places its executables in `/usr/bin/`. The `make install` commands put the executable files in `/usr/local/bin/` by default and you have not specified any options to `cmake`.

> that is what in line 212
#self.rtlsdr_source.set_min_output_buffer(int(sample_rate*rec_len)) #this line causes segfaults on HackRF

I've just installed the gr-gsm package in a clean container and here is what's on line 212 

```
211         self.rtlsdr_source = osmosdr.source(args="numchan=" + str(1) + " " + args)
212         // Enabling this line causes segfault with HackRF
213         //self.rtlsdr_source.set_min_output_buffer(int(sample_rate*rec_len))
214         self.rtlsdr_source.set_sample_rate(sample_rate)
```
This is not a valid python.  My advice is to either remove `gr-gsm` package by running `apt-get remove gr-gsm` or manually fix the comments in `/usr/bin/grgsm_scanner` 

@petterreinholdtsen could you fix this in the package?

BRRY876, 2018-08-09:
ok... i will see if it works

BRRY876, 2018-08-10:
i got an segmentation failed error...please help....dont see the line 212 error anymore. @velichkov

velichkov, 2018-08-10:
Did you remove the `gr-gsm` package or did you fix the comment by replacing `//` with `#`?
What SDR hardware do you use?
Provide the full output of `cat $(which grgsm_scanner) | sed '210,215!d'`.
If you are familiar with `gdb` run the grgsm_scanner under `gdb` and provide the full backtrace after it crashes.

BRRY876, 2018-08-10:
@velichkov i run the cat $ command and got this 

print "Args=", args
        self.rtlsdr_source = osmosdr.source(args="numchan=" + str(1) + " " + args)
        #self.rtlsdr_source.set_min_output_buffer(int(sample_rate*rec_len)) #this line causes segfaults on HackRF
        self.rtlsdr_source.set_sample_rate(sample_rate)

        # capture half of GSM channel lower than channel center (-0.1MHz)

further on i did not replace the // with # because when i looked at line 212 it already had the # key.I am not familiar with gdb. i removed the gr-gsm that i got from the apt

velichkov, 2018-08-10:
It would be a lot easier if you answer all my questions. 

What SDR hardware do you use?

Open apps/grgsm_scanner and comment out the following two lines
```
320             os.dup2(null_fds[0], 1)
321             os.dup2(null_fds[1], 2)
```
then reinstall 
```
cd build
make install
```
then run grgsm_scanner and provide the full output

BRRY876, 2018-08-10:
sorry about that friend..never see the question.. i am using rtl-sdr 820t2 rtl2832u....going to try what you to told me to do @velichkov

BRRY876, 2018-08-10:
@velichkov after  doing what was told..coment out and then make install i got this output

rgs= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106200; UHD_3.12.0.0-0-unknown
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[INFO] Using format CF32.
Segmentation fault

velichkov, 2018-08-10:
Is there a segmentation fault when you run grgsm_livemon, grgsm_livemon_headless or grgsm_capture?
Provide the output of `rtl_test -t`

BRRY876, 2018-08-10:
all except grgsm_capture is giving same error. the output of rtl_test -t is

0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.
Reattached kernel driver

velichkov, 2018-08-10:
> all except grgsm_capture is giving same error.

That's quite strange, I would have expected all to have the same error.

Ok, let's try debug it with `gdb`. First install it if you don't have it already (`sudo apt-get install gdb`) and then in a terminal
```
gdb --args /usr/bin/python /usr/local/bin/grgsm_scanner
``` 
This will start a `gdb` session and the `(gdb)` prompt will be displayed, then execute
```
run
```
and after it crashes
```
thread apply all bt
```
press Enter few times until the `(gdb)` prompt is displayed again and then copy the full output and provide it here enclosed in ```. See https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown

BRRY876, 2018-08-10:
```
root@kali:~# gdb --args /usr/bin/python /usr/local/bin/grgsm_scanner
GNU gdb (Debian 7.12-6+b2) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/python /usr/local/bin/grgsm_scanner
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[New Thread 0x7fffe6e0a700 (LWP 1599)]
[Thread 0x7fffe6e0a700 (LWP 1599) exited]
[New Thread 0x7fffe6e0a700 (LWP 1600)]
[Thread 0x7fffe6e0a700 (LWP 1600) exited]
[New Thread 0x7fffe6e0a700 (LWP 1601)]
[INFO[New Thread 0x7fffe6609700 (LWP 1602)]
] [UHD] linux; GNU C++ version 7.3.0; Boost_106200; UHD_3.12.0.0-0-unknown
[New Thread 0x7fffe5e08700 (LWP 1603)]
[New Thread 0x7fffe5607700 (LWP 1604)]
[Thread 0x7fffe5607700 (LWP 1604) exited]
[Thread 0x7fffe5e08700 (LWP 1603) exited]
[New Thread 0x7fffe5e08700 (LWP 1605)]
[New Thread 0x7fffe5607700 (LWP 1606)]
[Thread 0x7fffe5607700 (LWP 1606) exited]
[Thread 0x7fffe5e08700 (LWP 1605) exited]
[New Thread 0x7fffe5e08700 (LWP 1607)]
[New Thread 0x7fffe5607700 (LWP 1608)]
[Thread 0x7fffe5607700 (LWP 1608) exited]
[Thread 0x7fffe5e08700 (LWP 1607) exited]
[New Thread 0x7fffe5e08700 (LWP 1609)]
[Thread 0x7fffe5e08700 (LWP 1609) exited]
[New Thread 0x7fffe5e08700 (LWP 1610)]
[Thread 0x7fffe5e08700 (LWP 1610) exited]
[New Thread 0x7fffe5e08700 (LWP 1611)]
[Thread 0x7fffe5e08700 (LWP 1611) exited]
[New Thread 0x7fffe5e08700 (LWP 1612)]
[Thread 0x7fffe5e08700 (LWP 1612) exited]
[New Thread 0x7fffe5e08700 (LWP 1613)]
[Thread 0x7fffe5e08700 (LWP 1613) exited]
[New Thread 0x7fffe5e08700 (LWP 1614)]
[Thread 0x7fffe5e08700 (LWP 1614) exited]
[New Thread 0x7fffdc12c700 (LWP 1615)]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[Thread 0x7fffdc12c700 (LWP 1615) exited]
[New Thread 0x7fffe5e08700 (LWP 1616)]
[Thread 0x7fffe5e08700 (LWP 1616) exited]
[New Thread 0x7fffe5e08700 (LWP 1617)]
[Thread 0x7fffe5e08700 (LWP 1617) exited]
[New Thread 0x7fffe5e08700 (LWP 1618)]
[New Thread 0x7fffe5607700 (LWP 1619)]
[Thread 0x7fffe5607700 (LWP 1619) exited]
[... 175 lines cut ...]
#6  0x000055555563b9e9 in PyEval_EvalCodeEx ()
#7  0x0000555555659d58 in ?? ()
#8  0x000055555567353e in ?? ()
#9  0x0000555555672f68 in ?? ()
#10 0x000055555562ba2b in ?? ()
#11 0x00005555556235e6 in PyObject_Call ()
#12 0x00005555556431d2 in PyEval_EvalFrameEx ()
#13 0x000055555563b9e9 in PyEval_EvalCodeEx ()
---Type <return> to continue, or q <return> to quit---
#14 0x000055555564331e in PyEval_EvalFrameEx ()
#15 0x000055555563b9e9 in PyEval_EvalCodeEx ()
#16 0x000055555564331e in PyEval_EvalFrameEx ()
#17 0x000055555563b9e9 in PyEval_EvalCodeEx ()
#18 0x000055555563b376 in PyEval_EvalCode ()
#19 0x000055555566f54f in ?? ()
#20 0x0000555555669ec4 in PyRun_FileExFlags ()
#21 0x0000555555669487 in PyRun_SimpleFileExFlags ()
#22 0x000055555560d1cb in Py_Main ()
#23 0x00007ffff7a2db17 in __libc_start_main (main=0x55555560cc50 <main>, 
    argc=2, argv=0x7fffffffe288, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffe278)
    at ../csu/libc-start.c:310
#24 0x000055555560cb8a in _start ()
(gdb)
```

velichkov, 2018-08-10:
> Thread 1 (Thread 0x7ffff7a06740 (LWP 1595)):
#0  0x00007fffe9e7eae6 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.1.4
#1  0x00007fffe9e050a4 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.1.4
#2  0x00007fffea1088cd in ?? ()
   from /usr/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.x86_64-linux-gnu.so

This looks a lot like #341. Are you sure that the rtl dongle is connected when you run the grgsm_scanner?

Now you need to install some debuginfo symbols and rerun in gdb and provide the output again
```
echo "deb http://debug.mirrors.debian.org/debian-debug/ testing-debug main" | sudo tee /etc/apt/sources.list.d/debian-debug.list
sudo apt-get update
sudo apt-get install gr-osmosdr-dbgsym libgnuradio-osmosdr0.1.4-dbgsym libosmosdr0-dbgsym soapysdr0.6-module-osmosdr-dbgsym
```

Also try running grgsm_scanner with `--args=rtl` argument and provide the full output
```
grgsm_scanner --args=rtl
```

And finally try plugging the dongle in a diferent USB port.

BRRY876, 2018-08-10:
@velichkov i did commands up above except for the grgsm_scanner --ards+rt1 and changed the port which the rtl was in and got this so i assumes it is working

```
root@kali:~# grgsm_scanner

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106200; UHD_3.12.0.0-0-unknown
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOlibusb: error [udev_hotplug_event] ignoring udev action bind
OOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOO
```

velichkov, 2018-08-10:
> changed the port which the rtl was in

Most probably this fixed the crash

> and got this so i assumes it is working

almost, we made a good progress but unfortunately you get a lots of  `O` in the output

> Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver

which indicates buffer overflow and that your host cant read fast enough the data coming from the dongle.

What type of USB port do you use - USB 3, 2, 1?
Share some details about your host computer - CPU, memory, etc...

velichkov, 2018-08-10:
Now you could remove the dbgsym packages and the debian-debug repo because you no longer need them.

```
sudo apt-get remove gr-osmosdr-dbgsym libgnuradio-osmosdr0.1.4-dbgsym libosmosdr0-dbgsym soapysdr0.6-module-osmosdr-dbgsym
sudo rm /etc/apt/sources.list.d/debian-debug.list
sudo apt-get update
```

BRRY876, 2018-08-10:
**CPU**-Intel® Celeron(R) CPU N2840 @ 2.16GHz x 2
**RAM**-4gb DDR3L
**GRAPHICS**- Intel Bay Trail
**BIT**- 64

velichkov, 2018-08-10:
It should be cable to handle the traffic, check that you are using usb 2 or 3 port.

> the output of rtl_test -t is
> ```
> 0: Realtek, RTL2838UHIDIR, SN: 00000001
>
> Using device 0: Generic RTL2832U OEM
> Detached kernel driver
> Found Rafael Micro R820T tuner
> Supported gain values (29): 0.0 [...]
```
Could you rerun it without the `-t` options

velichkov, 2018-08-10:
> lost at least 12 bytes

That's not good but unfortunately I'm running out of ideas. Try rebooting your the machine, a different port or a smaller sampling rate

```
rtl_test -s 1000000
grgsm_scanner -s 1000000
```

BRRY876, 2018-08-12:
@velichkov still getting a lot of zero... i give up...any suggestion?

velichkov, 2018-08-13:
Do you run it in a VM?

BRRY876, 2018-08-13:
@velichkov no run it on my hard drive. I dual boot it