ptrkrysik/gr-gsm #427: grgsm_livement generating segmentation fault

Sh0ckFR, 2018-08-09:
Hello,

I tried to use this guide on a fresh Kali rolling version : [https://osmocom.org/projects/gr-gsm/wiki/Installation](https://osmocom.org/projects/gr-gsm/wiki/Installation)

But I have an issue during the generation:

`
[ 89%] Generating grgsm_livemon
Segmentation fault
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:82: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:1173: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:127: all] Error 2
`

I removed all the osmosdr, gr-gsm packages multiple times but no way, if I try to install the debian package directly, I have the same issue than : [https://github.com/ptrkrysik/gr-gsm/issues/426](https://github.com/ptrkrysik/gr-gsm/issues/426)

So I don't really understand what is wrong, my hardware is not a hackrf but a rtl2832 dvb-t adapter.

Here is my system informations :

debian-version: kali-rolling
gnuradio version: 3.7.13.4-1
grcc path: /usr/local/bin/grcc
cmake version: 3.2.0

I also removed pybombs and the pip package gr-gsm.

velichkov, 2018-08-09:
Hi @Sh0ckFR,

Please never edit your messages as github does not send emails with the edited text.

Please use markdown formatting when pasting logs, the build logs in your message are on one line which makes them hard to understand. 

> grcc path: /usr/local/bin/grcc

You still have the `gnuradio` version installed by `pybombs`, the one in `/usr/local/bin/`. Removing the `pybombs` does not remove all packages installed by it so I guess you need to remove them manually.

When did you installed `pybombs` and `gnuradio` with it, before or after you experience the segmentation fault build failure?

>  on a fresh Kali rolling version

Because this is a fresh installation my advice is to start again from scratch by formatting and re-installing the system and on the new one **do not** install `pybombs`.

> if I try to install the debian package directly, I have the same issue than : #426

Could you copy-paste few lines around line 212 from /usr/bin/grgsm_scanner

> grcc -d /home/shock/gr-gsm/build/apps /home/shock/gr-gsm/apps/grgsm_livemon.grc
> Param - Import(import):
Import "from grgsm import arfcn" failed.To compile it manually like 

To compile it manually like this you first need to export a lot of environment variables. The easiest way to get them is by running  `make VERBOSE=1`, it's going to display the full command together with all variables. You need to copy-paste the environment variables and export  them in your shell.

> [ 89%] Generating grgsm_livemon Segmentation fault 

I'll try to reproduce this crash but it's going to take me some time.

velichkov, 2018-08-09:
Hi @AlexeiScherbakov,

Have you been able to resolve the `Segmentation fault` issue you've reported in https://github.com/ptrkrysik/gr-gsm/issues/399#issuecomment-399944362 and if yes how?

I'm still not able to reproduce the crash in a Docker container 
```
docker build -f tests/dockerfiles/Kali.docker -t gr-gsm-kali-rolling .
```
which only proves that if you start from a fresh clean Kali installation and follow the commands in the `tests/dockerfiles/Kali.docker` you could successfully install `gr-gsm`

Sh0ckFR, 2018-08-09:
Well I finally fixed the issue like that :

apt-get install pybombs -> to reinstall pybombs
pybombs remove gnuradio
apt-get install --reinstall gnuradio
cd .. && rm -rf build/ && mkdir build && cd build
cmake ..
make
make install
export PYTHONPATH=/usr/local/lib/python2.7/dist-packages/grgsm:$PYTHONPATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
grgsm_livemon

Sorry about the markdown, my keyboard is a bit broken, best way is to never use pybombs.

velichkov, 2018-08-09:
> Well I finally fixed the issue like that :

That's great :smile:  Now please close this issue.

> export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH

I'm not quite sure whether you need exporting PYTHONPATH or not,  but instead of exporting LD_LIBRARY_PATH you could run `ldconfig` after `make install`

Sh0ckFR, 2018-08-09:
Thank you for your help anyway, have a good day.

closed 2018-08-09 by Sh0ckFR

justicexxx, 2021-05-04:
97 / 5000
Çeviri sonuçları
I have a problem too, although I did everything right, I cannot find the frequencies. Please help 

[image]