ptrkrysik/gr-gsm #428: PyBOMBS Installation Fails: 'LIBTOOL' is undefined, 'RANLIB' is undefined

rytse, 2018-08-10:
I tried installing gr-gsm using PyBOMBS with `sudo pybombs install gr-gsm`. This fails while installing libosmocore with `Libtool library used but 'LIBTOOL' is undefined` and `library used but 'RANLIB' is undefined`.

I tried reinstalling libtool, automake and cmake with `sudo apt install --reinstall` to no avail.

The PyBOMBS output is as follows:

```
rytse@SEDRSSHYPRES:~$ sudo pybombs install gr-gsm
[sudo] password for rytse: 
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
The directory '/home/rytse/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Install tree:
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- airspy
   |  |
   |  +- hackrf
   |  |
   |  +- soapysdr
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  \- libosmo-dsp
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   \- libosmocore
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: libosmocore
Configuring: (100%) [============================================================================================================================================================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /usr/local/bin/automake line 3922.
src/Makefile.am:13: error: Libtool library used but 'LIBTOOL' is undefined
src/Makefile.am:13:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/Makefile.am:13:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/Makefile.am:13:   If 'LT_INIT' is in 'configure.ac', make sure
src/Makefile.am:13:   its definition is in aclocal's search path.
src/codec/Makefile.am:14: error: Libtool library used but 'LIBTOOL' is undefined
src/codec/Makefile.am:14:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/codec/Makefile.am:14:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/codec/Makefile.am:14:   If 'LT_INIT' is in 'configure.ac', make sure
src/codec/Makefile.am:14:   its definition is in aclocal's search path.
src/coding/Makefile.am:16: error: Libtool library used but 'LIBTOOL' is undefined
src/coding/Makefile.am:16:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/coding/Makefile.am:16:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/coding/Makefile.am:16:   If 'LT_INIT' is in 'configure.ac', make sure
src/coding/Makefile.am:16:   its definition is in aclocal's search path.
src/ctrl/Makefile.am:9: error: Libtool library used but 'LIBTOOL' is undefined
src/ctrl/Makefile.am:9:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/ctrl/Makefile.am:9:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/ctrl/Makefile.am:9:   If 'LT_INIT' is in 'configure.ac', make sure
src/ctrl/Makefile.am:9:   its definition is in aclocal's search path.
src/gb/Makefile.am:12: error: Libtool library used but 'LIBTOOL' is undefined
src/gb/Makefile.am:12:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/gb/Makefile.am:12:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/gb/Makefile.am:12:   If 'LT_INIT' is in 'configure.ac', make sure
src/gb/Makefile.am:12:   its definition is in aclocal's search path.
src/gsm/Makefile.am:19: error: Libtool library used but 'LIBTOOL' is undefined
src/gsm/Makefile.am:19:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/gsm/Makefile.am:19:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/gsm/Makefile.am:19:   If 'LT_INIT' is in 'configure.ac', make sure
src/gsm/Makefile.am:19:   its definition is in aclocal's search path.
src/pseudotalloc/Makefile.am:5: error: Libtool library used but 'LIBTOOL' is undefined
src/pseudotalloc/Makefile.am:5:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/pseudotalloc/Makefile.am:5:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/pseudotalloc/Makefile.am:5:   If 'LT_INIT' is in 'configure.ac', make sure
src/pseudotalloc/Makefile.am:5:   its definition is in aclocal's search path.
src/sim/Makefile.am:15: error: Libtool library used but 'LIBTOOL' is undefined
src/sim/Makefile.am:15:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/sim/Makefile.am:15:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/sim/Makefile.am:15:   If 'LT_INIT' is in 'configure.ac', make sure
src/sim/Makefile.am:15:   its definition is in aclocal's search path.
src/vty/Makefile.am:10: error: Libtool library used but 'LIBTOOL' is undefined
src/vty/Makefile.am:10:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/vty/Makefile.am:10:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/vty/Makefile.am:10:   If 'LT_INIT' is in 'configure.ac', make sure
src/vty/Makefile.am:10:   its definition is in aclocal's search path.
tests/Makefile.am:7: error: library used but 'RANLIB' is undefined
tests/Makefile.am:7:   The usual way to define 'RANLIB' is to add 'AC_PROG_RANLIB'
tests/Makefile.am:7:   to 'configure.ac' and run 'autoconf' again.
autoreconf: automake failed with exit status: 1
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make sets $(MAKE)... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
./configure: line 3966: syntax error near unexpected token `pic-only'
./configure: line 3966: `LT_INIT(pic-only disable-static)'
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package libosmocore:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package libosmocore. Aborting.
```

velichkov, 2018-08-10:
Hi @rytse,

Which Linux distribution and version do you use?

The pybombs installation method is the most problematic one so my advice is to remove everything installed by pybombs,  then remove pybombs and try to install gr-gsm using the other methods described in https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804

velichkov, 2018-08-10:
The correct link is https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

velichkov, 2018-08-10:
Have you removed everything installed by pybombs and the pybombs?

> Based on 389, I believe I have to install gr-osmosdr. However when I try to build that from source, I get error: invalid conversion from ‘bladerf_module {aka int}’ to ‘bladerf_channel_layout’

You don't have to installed it from source as according to the instructions it is installed from the apt repository
```
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev
```

Most probably you have not removed everything installed by pybombs and now there are multiple versions of gnuradio and various other libraries.

Check if `which grcc` returns `/usr/local/bin/grcc` and if yes execute `pybombs remove gnuradio` and try compiling gr-gsm again.

rytse, 2018-08-13:
After I removed all the PyBOMBS files the installation worked. Thank you @velichkov for all the help.

Also, to anyone that *must* use gr-gsm with PyBOMBS: I believe you can very easily remove gr-gsm's dependency on osmosdr stuff (the package that currently breaks with PyBOMBS). It is only used in the examples, so if you delete the apps directory and remove that target from CMakeLists, the core gr-gsm code will build. However this is a bit of a hack, so I am going with the apt package of gnuradio as velichkov suggested.

ptrkrysik, 2018-08-31:
osmosdr source is used by grgsm_livemon application, so if you disable building it with grcc :
cmake .. -DENABLE_GRCC=OFF
the build should work fine.

closed 2018-08-31 by ptrkrysik