ptrkrysik/gr-gsm #430: undefined reference to `rtlsdr_set_bias_tee'

thealmogg, 2018-08-22:
Hi,
I'm trying to install gr-gsm as goes in this guide:
Install GNU Radio GSM modules / gr-gsm
 sudo apt-get install git python-pip 
 sudo pip install PyBOMBS
 sudo pybombs prefix init /usr/local -a default_prx
 sudo pybombs config default_prefix default_prx
 sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes
 sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera
 sudo pybombs install gr-gsm
 sudo ldconfig

The process falling in this command:

 sudo pybombs install gr-gsm

with the error:

`PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.13
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-gsm
   |
   \- gr-osmosdr
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gr-osmosdr
PyBOMBS.Packager.source - WARNING - Build dir already exists: /home/parallels/src/gr-osmosdr/build
Building:    (100%) [=========================================================]
[  2%] Linking CXX shared library libgnuradio-osmosdr-0.1.5git.so
CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o: In function `rtl_source_c::rtl_source_c(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/home/parallels/src/gr-osmosdr/lib/rtl/rtl_source_c.cc:224: undefined reference to `rtlsdr_set_bias_tee'
collect2: error: ld returned 1 exit status
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:415: recipe for target 'lib/libgnuradio-osmosdr-0.1.5git.so.0.0.0' failed
make[2]: *** [lib/libgnuradio-osmosdr-0.1.5git.so.0.0.0] Error 1
CMakeFiles/Makefile2:140: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-osmosdr:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gr-osmosdr. Aborting.
`

undefined reference to `rtlsdr_set_bias_tee'?
tried and search a lot.
tried to clean any rtl-sdr from the Debian machine and install again - didn't do much.

Any ideas?
Thanks.
Using Debian 3.22.2 64bit

velichkov, 2018-08-22:
Hi @thealmogg,

> I'm trying to install gr-gsm as goes in this guide:
sudo pybombs install gr-gsm

The `pybombs` method is the most error prone and difficult method to install `gr-gsm` and I do not recommend it at all. I recommend you to remove everyting installed by pybombs 
```
sudo pybombs inv
sudo pybombs remove gnuradio  #and add every package from the output of previous command
```
and then install `gr-gsm` following [the official instructions](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)

> 3.22.2

This seems like the gnome version which is not very relevant to the gr-gsm. Provide the output of
```
cat /etc/debian_version
lsb_release -a 
```

thealmogg, 2018-08-22:
@velichkov Thanks.
Much appreciated!
Using Pybombs makes me feel like I'm debugging other devs messy spaghetti code.

closed 2018-08-22 by thealmogg