ptrkrysik/gr-gsm #431: Block keys not found ?

nturns, 2018-08-29:
Using Centos 7 and following the install instructions for Fedora 26, except used "yum install" for all missing packages. On make, got to 90% then this happened. I found fc_slider-shiftoff" in apps/grgsm_livemon.grc so I assume everything else exists, it is just not being found. I don't know what to do. During the cmake step I noticed this, Could NOT find libosmocoding (missing:  LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR). Here is the output when make failed:

`Scanning dependencies of target pygen_apps
[ 90%] Generating grgsm_livemon
Block key "variable_qtgui_range" not found
Block key "variable_qtgui_range" not found
Block key "variable_qtgui_range" not found
Block key "qtgui_freq_sink_x" not found
linux; GNU C++ version 4.8.5 20150623 (Red Hat 4.8.5-16); Boost_105300; UHD_003.010.001.000-0-unknown

Validation failed:

Block - gsm_clock_offset_control_0 - GSM Clock Offset Control(gsm_clock_offset_control):
	Param - fc(fc):
		Value "fc_slider-shiftoff" cannot be evaluated:
		name 'fc_slider' is not defined

Block - gsm_input_0 - GSM Input Adaptor(gsm_input):
	Param - fc(fc):
		Value "fc_slider-shiftoff" cannot be evaluated:
		name 'fc_slider' is not defined

Block - rtlsdr_source_0 - RTL-SDR Source(rtlsdr_source):
	Param - Ch0: Frequency (Hz)(freq0):
		Value "fc_slider-shiftoff" cannot be evaluated:
		name 'fc_slider' is not defined

Block - rtlsdr_source_0 - RTL-SDR Source(rtlsdr_source):
	Param - Ch0: Freq. Corr. (ppm)(corr0):
		Value "ppm_slider" cannot be evaluated:
		name 'ppm_slider' is not defined

Block - rtlsdr_source_0 - RTL-SDR Source(rtlsdr_source):
	Param - Ch0: RF Gain (dB)(gain0):
		Value "gain_slider" cannot be evaluated:
		name 'gain_slider' is not defined
Error during file compilation.
make[2]: *** [apps/grgsm_livemon] Error 1
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [all] Error 2
[root@c-73-25-113-164 build]#`

velichkov, 2018-08-30:
Hi @nturns,

> Using Centos 7 and following the install instructions for Fedora 26, except used "yum install" for all missing packages. 

We have a [Centos 7 Dockerfile](https://github.com/ptrkrysik/gr-gsm/blob/592ec9b7859f544b297e9352323b45ceb450a68a/tests/dockerfiles/CentOS_7.Dockerfile) that builds successfully.

> On make, got to 90% then this happened. I found fc_slider-shiftoff" in apps/grgsm_livemon.grc so I assume  everything else exists, it is just not being found. I don't know what to do. During the [...]

The `gnuradio` version in centos 7 is not too old (3.7.11-7) but unfortunately they do not distribute the Qt blocks that are used in `grgsm_livemon` because some other Qt packages are missing in centos7/epel7
https://src.fedoraproject.org/rpms/gnuradio/blob/epel7/f/gnuradio.spec#_55-57

So you have few options:

* Disable grgsm_livemon compilation - run `cmake -DENABLE_GRGSM_LIVEMON=OFF ..` from the build directory
* Use Fedora 28 :)
* Try to manually build the missing packages and then build gnuradio with enabled Qt blocks.

velichkov, 2018-08-30:
Here is the output from Fedora 28
```
$ rpm -ql gnuradio | grep blocks\/qt
/usr/share/gnuradio/grc/blocks/qtgui_ber_sink_b.xml
/usr/share/gnuradio/grc/blocks/qtgui_block_tree.xml
/usr/share/gnuradio/grc/blocks/qtgui_check_box.xml
/usr/share/gnuradio/grc/blocks/qtgui_chooser.xml
/usr/share/gnuradio/grc/blocks/qtgui_const_sink_x.xml
/usr/share/gnuradio/grc/blocks/qtgui_edit_box_msg.xml
/usr/share/gnuradio/grc/blocks/qtgui_entry.xml
/usr/share/gnuradio/grc/blocks/qtgui_freq_sink_x.xml
/usr/share/gnuradio/grc/blocks/qtgui_histogram_sink_x.xml
/usr/share/gnuradio/grc/blocks/qtgui_label.xml
/usr/share/gnuradio/grc/blocks/qtgui_number_sink.xml
/usr/share/gnuradio/grc/blocks/qtgui_push_button.xml
/usr/share/gnuradio/grc/blocks/qtgui_range.xml
/usr/share/gnuradio/grc/blocks/qtgui_sink_x.xml
/usr/share/gnuradio/grc/blocks/qtgui_tab_widget.xml
/usr/share/gnuradio/grc/blocks/qtgui_time_raster_x.xml
/usr/share/gnuradio/grc/blocks/qtgui_time_sink_x.xml
/usr/share/gnuradio/grc/blocks/qtgui_vector_sink_f.xml
/usr/share/gnuradio/grc/blocks/qtgui_waterfall_sink_x.xml
```
all those blocks are missing in Centos 7
```
$ rpm -ql gnuradio | grep blocks\/qt
$
```

nturns, 2018-08-30:
Thank you for that.

It compiled with "cmake -DENABLE_GRGSM_LIVEMON=OFF .." but the tradeoff is living without grgsm_livemon. I will probably try doing the third option you listed. Or possibly find who maintains that package for Centos and see if it can be updated.

velichkov, 2018-08-30:
You could also try installing it with [pybombs](https://osmocom.org/projects/gr-gsm/wiki/Installation#Compilation-and-installation-with-use-of-Pybombs). If you decide to do it first 
test in a VM/container because if it fails reverting/uninstalling everything installed by `pybombs` could be very difficult and having two versions (one from yum and one from `pybombs`) could also be problematic and not recommended.

Also please close this issue if you don't need more help.

velichkov, 2018-08-31:
Hi @nturns,

I remembered that few months ago I did some experiments and I built `gnuradio` for `el7` with `qt` support, the result packages are available from [my copr repository](https://copr.fedorainfracloud.org/coprs/vvvelichkov/gnuradio/)

Also I just found another [copr repository](https://copr.fedorainfracloud.org/coprs/marcusmueller/gnuradio/) that contains the newest `gnuradio` version built with `qt` for `el7`.

See also the [bugzilla issue](https://bugzilla.redhat.com/show_bug.cgi?id=1610513) where some `gnuradio` package updates are discussed.

nturns, 2018-09-01:
I tried updating with the marcusmueller-gnuradio version and I got a lot of

`--> Finished Dependency Resolution
Error: Package: gr-fcdproplus-0-0.8.20140920git1edbe523.el7.x86_64 (@epel)
           Requires: libgnuradio-runtime-3.7.11.so.0.0.0()(64bit)
           Removing: gnuradio-3.7.11-7.el7.x86_64 (@epel)
               libgnuradio-runtime-3.7.11.so.0.0.0()(64bit)
           Updated By: gnuradio-3.7.13.4-0.el7.x86_64 (marcusmueller-gnuradio)
               Not found
           Available: gnuradio-3.7.11-6.qtgui.el7.centos.x86_64 (vvvelichkov-gnuradio)
--- text cut ---
Error: Package: gr-fcdproplus-0-0.8.20140920git1edbe523.el7.x86_64 (@epel)
           Requires: libgnuradio-blocks-3.7.11.so.0.0.0()(64bit)
           Removing: gnuradio-3.7.11-7.el7.x86_64 (@epel)
               libgnuradio-blocks-3.7.11.so.0.0.0()(64bit)
           Updated By: gnuradio-3.7.13.4-0.el7.x86_64 (marcusmueller-gnuradio)
               Not found
           Available: gnuradio-3.7.11-6.qtgui.el7.centos.x86_64 (vvvelichkov-gnuradio)
               libgnuradio-blocks-3.7.11.so.0.0.0()(64bit)
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest
`
I tried this solution but it didn't work. I don't know what is going on.
https://centos.org/forums/viewtopic.php?t=59821

velichkov, 2018-09-01:
> Error: Package: gr-fcdproplus-0-0.8.20140920git1edbe523.el7.x86_64 (@epel)
Requires: libgnuradio-runtime-3.7.11.so.0.0.0()(64bit)

You have `gr-fcdproplus` on your system that is built and installed with the gnuradio from `epel` repository. You have to remove it (temporarily), update and then if you need  `gr-fcdproplus` you have to rebuild it from the src.rpm with the newer gnuradio.

ptrkrysik, 2018-09-05:
Vasil, regarding pybombs and reverting what it does, I'm starting to
consider to change pybombs installation description, so it doesn't
install to /usr/local but to a directory created by the user.

nturns, 2018-09-05:
Vasil,
I did your advice and it did compile successfully. However instead of grgsm_livemon there is a grgsm_livemon_headless. When I try to run any of the programs they return:
"ImportError: No module named grgsm"

I started a live CD of Debian and was able to compile and run grgsm_livemon straightaway just to see it work. However I did not install Debian.

I would try pybombs if it was safer.

velichkov, 2018-09-05:
Hi @nturns,

> However instead of grgsm_livemon there is a grgsm_livemon_headless.

That's more then expected as we have two versions and with the `-DENABLE_GRGSM_LIVEMON=OFF` option you've disabled compilation of the first one (that uses Qt blocks).

> When I try to run any of the programs they return:
"ImportError: No module named grgsm"

After installing gr-gsm on some distributions you need to run `sudo ldconfig`

@ptrkrysik

> so it doesn't install to /usr/local but to a directory created by the user.

This will move the c/c++ packages out of `/usr/local` which is definitely an improvement but `pybombs` installs some packages using `sudo pip` and at least on centos they goes in `/usr/lib/python2/site-packages` which in my opinion is the worst possible location.

Unfortunately the `virtualenv` support is currently broken https://github.com/gnuradio/pybombs/issues/363

I just tired a `pybombs` installation in a `centos:7` container and it fails with

```
$ pybombs prefix init -a default -R gnuradio-default ~/gnuradio/default

-- checking for module 'gmp'
--   package 'gmp' not found
-- Could NOT find GMP (missing:  GMPXX_LIBRARY GMP_INCLUDE_DIR) 
-- checking for module 'mpir >= 3.0'
--   package 'mpir >= 3.0' not found
-- Could NOT find MPIR (missing:  MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find MPLIB (missing: MPLIBXX_LIBRARY MPLIB_LIBRARY
  MPLIB_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindMPLIB.cmake:26 (find_package_handle_standard_args)
  gnuradio-runtime/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
```
Actually the `gmp` package was actually installed by `pybombs`
```
$ pybombs inv
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.5
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
Showing package state:
automake:           installed
gmp:                installed
libusb:             installed
uhd:                installed
gnuradio:           fetched
```
To fix this I had to `pybombs remove gmp` and add an rpm in the gmp.lwr (I just opened https://github.com/gnuradio/gr-recipes/pull/128)
```
PyBOMBS.install_manager - INFO - Installing package: pcsclite
fatal: unable to connect to anonscm.debian.org:
anonscm.debian.org[0: 194.177.211.202]: errno=Connection refused
anonscm.debian.org[1: 2001:648:2ffc:deb::211:202]: errno=Network is unreachable

error: Could not fetch pcsclite
PyBOMBS.Fetcher - ERROR - Unexpected error while fetching git+git://anonscm.debian.org/pcsclite/PCSC.git.
```
and after fixing the URL (https://github.com/gnuradio/gr-recipes/pull/129)
```
checking for mq_getattr in -lrt... yes
./configure: line 14507: syntax error near unexpected token `LIBSYSTEMD,'
./configure: line 14507: `	PKG_CHECK_MODULES(LIBSYSTEMD, libsystemd,,'
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pcsclite:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package pcsclite. Aborting.
```
and that's probably pybombs installed a newer `automake` in the `prefix`

This constant problems just drive me nuts :angry: