ptrkrysik/gr-gsm #432: i have problem in grgsm_scanner

antiuniv, 2018-09-05:
i installed parrot and grgsm   but if runnug grgsm_scanner i have this erro 

root@parrot]-[/root/grgsm]
#grgsm_scanner
linux GNU++ version 7.3.0; Boost_106200. UHD_003.010.003.000-0-unknown

Abandon

velichkov, 2018-09-05:
Hi @antiuniv,

> i installed parrot

Which version?

> and grgsm but if runnug grgsm_scanner i have this erro

How did you install gnuradio - from the apt repository or using pybombs?
How did you install gr-gsm - manually, from the apt repository or using pybombs?
What SDR device do you use?

To get some more output open `apps/grgsm_scanner` and comment out the following lines

```
320             os.dup2(null_fds[0], 1)
321             os.dup2(null_fds[1], 2)
```

then reintsall, test again and provide the full output.

See also #426, #336 and #273

antiuniv, 2018-09-05:
i install apt-get install gr-gsm

velichkov, 2018-09-05:
> i install apt-get install gr-gsm

Could you list all commands that you have used?

Also answer the question about the SDR device and provide the full output after commenting out lines 320 and 321.

antiuniv, 2018-09-05:
SDR 
 tuner R820T2

antiuniv, 2018-09-05:
linux; GNU C++ version 7.3.0; Boost_106200; UHD_003.010.003.000-0-unknown

Abandon
┌─[✗]─[root@parrot]─[/home/hard]
└──╼ #rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 16 bytes

antiuniv, 2018-09-05:
# put /dev/null fds on 1 and 2
            # os.dup2(null_fds[0], 1)
            # os.dup2(null_fds[1], 2)

velichkov, 2018-09-05:
> linux; GNU C++ version 7.3.0; Boost_106200; UHD_003.010.003.000-0-unknown
>
> Abandon

After commenting them you need to reinstall by executing `make install` in the build directory.

velichkov, 2018-09-05:
Also try plugging the RTL dongle in a different USB port

antiuniv, 2018-09-05:
┌─[✗]─[root@parrot]─[/home/hard/Desktop/gr-gsm/build]
└──╼ #grgsm_scanner
linux; GNU C++ version 7.3.0; Boost_106200; UHD_003.010.003.000-0-unknown

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

velichkov, 2018-09-05:
> Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOpython2: /usr/include/boost/smart_ptr/intrusive_ptr.hpp:198: T* boost::intrusive_ptr::operator->() const [with T = pmt::pmt_base]: Assertion `px != 0' failed.
Abandon

The multiple 'O' indicates buffer overflows, which means that host is not able to read the data from the RTL fast enough and the buffers get full.

Are you running it in a Virtual Machine? Provide the full output of `lsusb` and `cat /proc/cpuinfo`

Try running it with smaller sample rate
```
grgsm_scanner -s 0.8e6
grgsm_scanner -s 1.2e6
```
and provide the full output of both commands.

antiuniv, 2018-09-05:
Thx   velichkov i installed in my lapptop not vm 

─[root@parrot]─[/home/hard/Desktop/gr-gsm]
└──╼ #lsusb
Bus 001 Device 004: ID 04f2:b2a6 Chicony Electronics Co., Ltd 
Bus 001 Device 006: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 0cf3:311d Atheros Communications, Inc. 
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 002: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

──╼ #cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 54
model name	: Intel(R) Atom(TM) CPU N2600   @ 1.60GHz
stepping	: 1
microcode	: 0x10d
cpu MHz		: 1497.076
cache size	: 512 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 10
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 xtpr pdcm movbe lahf_lm dtherm arat
bugs		:
bogomips	: 3192.63
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 32 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 54
model name	: Intel(R) Atom(TM) CPU N2600   @ 1.60GHz
stepping	: 1
microcode	: 0x10d
cpu MHz		: 1306.548
cache size	: 512 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 10
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 xtpr pdcm movbe lahf_lm dtherm arat
bugs		:
bogomips	: 3192.63
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 32 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 54
model name	: Intel(R) Atom(TM) CPU N2600   @ 1.60GHz
stepping	: 1
microcode	: 0x10d
cpu MHz		: 1595.786
cache size	: 512 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 2
initial apicid	: 2
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 10
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx lm constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm arat
bugs		:
bogomips	: 3192.63
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 6
model		: 54
model name	: Intel(R) Atom(TM) CPU N2600   @ 1.60GHz
stepping	: 1
microcode	: 0x10d
cpu MHz		: 1595.825
cache size	: 512 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 3
initial apicid	: 3
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 10
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx lm constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm arat
bugs		:
bogomips	: 3192.63
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

velichkov, 2018-09-05:
> model name : Intel(R) Atom(TM) CPU N2600 @ 1.60GHz

Maybe this Atom processor is not powerful enough. If you have another machine with more powerful CPU try on it.

Have you tried with a smaller sample rate ?

antiuniv, 2018-09-05:
By the time he was working plain

antiuniv, 2018-09-05:
The aplication grgsm_scanner  were working on the same machine

velichkov, 2018-09-05:
Sorry I somehow still don't get it. I've asked if you could try on a more powerful machine (if you have one) and if you have tried with a smaller sample rate and you respond with "The aplication grgsm_scanner were working on the same machine".

antiuniv, 2018-09-05:
Thank you. I will try it on another machine

velichkov, 2018-09-05:
Please try with a smaller sample rate
```
grgsm_scanner -s 0.8e6
grgsm_scanner -s 1.2e6
```
and provide the full output of both commands.

antiuniv, 2018-09-08:
Thx @velichkov 

i install linux parrot 4.4.0 Same device

─[root@parrot]─[/home/alex/Desktop/gr-gsm]
└──╼ #lsusb
Bus 001 Device 004: ID 04f2:b2a6 Chicony Electronics Co., Ltd
Bus 001 Device 006: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 0cf3:311d Atheros Communications, Inc.
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 002: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

──╼ #cat /proc/cpuinfo
processor : 0
vendor_id : GenuineIntel
cpu family : 6
model : 54
model name : Intel(R) Atom(TM) CPU N2600 @ 1.60GHz
stepping : 1
microcode : 0x10d
cpu MHz : 1497.076
cache size : 512 KB
physical id : 0
siblings : 4
core id : 0
cpu cores : 2
apicid : 0
initial apicid : 0
fdiv_bug : no
f00f_bug : no
coma_bug : no
fpu : yes
fpu_exception : yes
cpuid level : 10
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 xtpr pdcm movbe lahf_lm dtherm arat
bugs :
bogomips : 3192.63
clflush size : 64
cache_alignment : 64
address sizes : 36 bits physical, 32 bits virtual
power management:

processor : 1
vendor_id : GenuineIntel
cpu family : 6
model : 54
model name : Intel(R) Atom(TM) CPU N2600 @ 1.60GHz
stepping : 1
microcode : 0x10d
cpu MHz : 1306.548
cache size : 512 KB
physical id : 0
siblings : 4
core id : 0
cpu cores : 2
apicid : 1
initial apicid : 1
fdiv_bug : no
f00f_bug : no
coma_bug : no
fpu : yes
fpu_exception : yes
cpuid level : 10
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 xtpr pdcm movbe lahf_lm dtherm arat
bugs :
bogomips : 3192.63
clflush size : 64
cache_alignment : 64
address sizes : 36 bits physical, 32 bits virtual
power management:

processor : 2
vendor_id : GenuineIntel
cpu family : 6
model : 54
model name : Intel(R) Atom(TM) CPU N2600 @ 1.60GHz
stepping : 1
microcode : 0x10d
cpu MHz : 1595.786
cache size : 512 KB
physical id : 0
siblings : 4
core id : 1
cpu cores : 2
apicid : 2
initial apicid : 2
fdiv_bug : no
f00f_bug : no
coma_bug : no
fpu : yes
fpu_exception : yes
cpuid level : 10
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx lm constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm arat
bugs :
bogomips : 3192.63
clflush size : 64
cache_alignment : 64
address sizes : 36 bits physical, 48 bits virtual
power management:

processor : 3
vendor_id : GenuineIntel
cpu family : 6
model : 54
model name : Intel(R) Atom(TM) CPU N2600 @ 1.60GHz
stepping : 1
microcode : 0x10d
cpu MHz : 1595.825
cache size : 512 KB
physical id : 0
siblings : 4
core id : 1
cpu cores : 2
apicid : 3
initial apicid : 3
fdiv_bug : no
f00f_bug : no
coma_bug : no
fpu : yes
fpu_exception : yes
cpuid level : 10
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx lm constant_tsc arch_perfmon pebs bts nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm arat
bugs :
bogomips : 3192.63
clflush size : 64
cache_alignment : 64
address sizes : 36 bits physical, 48 bits virtual
power management:

==========================
apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y cmake autoconf libtool pkg-config build-essential python-docutils libcppunit-dev swig doxygen liblog4cpp5-dev  python-gtk2 python-scipy     gnuradio-dev    gr-osmosdr 
    libosmocore-dev

git clone https://github.com/ptrkrysik/gr-gsm.git
cd gr-gsm
 mkdir build
cd build
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ && 
make -j $(nproc) 
make install 
ldconfig 
make CTEST_OUTPUT_ON_FAILURE=1 test

agrgsm_scanner  and gegsm_livemon  worked good

velichkov, 2018-09-11:
> grgsm_scanner and grgsm_livemon worked good

When testing `grgsm_scanner` did you specify smaller sampling rate and does it work with the default 2M ?

antiuniv, 2018-09-16:
Thx @velichkov 

uname -a
Linux parrot 4.14.0-parrot13-686-pae #1 SMP Parrot 4.14.13-1parrot13 (2018-01-21) i686 GNU/Linux

─[root@parrot]─[/home/alex/Desktop]
└──╼ #grgsm_scanner

ARFCN:   xx, Freq:  xxx.xM, CID:     0, LAC: xxxxx, MCC: xxx, MNC:   x, Pwr: -xx

problem not for version parrot 

i see problem in make ? my be 

bcz i installed gr-gsm with out  nproc  () i have error in grgsm_scanner 

but i installed agian with make -j $(nproc)  workked good

velichkov, 2018-09-17:
> Linux parrot 4.14.0-parrot13-686-pae #1 SMP Parrot 4.14.13-1parrot13 (2018-01-21) i686 GNU/Linux
> problem not for version parrot

This is the Linux's kernel version and not the Parrot OS version. It seems you are running an older version ([probably 3.11](https://blog.parrotsec.org/parrot-3-11-release-notes/)) as the [4.2 version](https://blog.parrotsec.org/parrot-4-2-release-notes/) is with kernel 4.18.6. Also you are running the 32bit version (i686).

Provide the full output of 
```
lsb_release -a
cat /etc/lsb-release
cat /etc/apt/sources.list
cat /etc/apt/sources.list.d/parrot.list
apt-cache policy
```

>i see problem my be in proc
>
>nproc 4
>
>make -j $(nproc)

I doubt this is the problem as it specify only how many CPUs will be used during compilation and does not make any difference when executing the compiled programs.

antiuniv, 2018-09-17:
└──╼ #lsb_release -a
No LSB modules are available.
Distributor ID:	Parrot
Description:	Parrot 3.11
Release:	3.11

└──╼ #cat /etc/lsb-release
DISTRIB_ID=Parrot
DISTRIB_RELEASE=3.11
DISTRIB_CODENAME=stable
DISTRIB_DESCRIPTION="Parrot 3.11"

└──╼ #cat /etc/apt/sources.list
# 

# deb cdrom:[Debian GNU/Linux 3.11 _Parrot_ - Official Snapshot i386 LIVE/INSTALL Binary 20180127-23:30]/ parrot contrib main non-free

#deb cdrom:[Debian GNU/Linux 3.11 _Parrot_ - Official Snapshot i386 LIVE/INSTALL Binary 20180127-23:30]/ parrot contrib main non-free

└──╼ #cat /etc/apt/sources.list.d/parrot.list
## stable repository
deb http://deb.parrotsec.org/parrot stable main contrib non-free
#deb-src http://archive.parrotsec.org/parrot stable main contrib non-free

──╼ #apt-cache policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
1001 http://deb.parrotsec.org/parrot stable/non-free i386 Packages
     release o=Parrot,a=stable,n=parrot,l=Parrot Stable Repository,c=non-free,b=i386
     origin deb.parrotsec.org
1001 http://deb.parrotsec.org/parrot stable/contrib i386 Packages
     release o=Parrot,a=stable,n=parrot,l=Parrot Stable Repository,c=contrib,b=i386
     origin deb.parrotsec.org
1001 http://deb.parrotsec.org/parrot stable/main i386 Packages
     release o=Parrot,a=stable,n=parrot,l=Parrot Stable Repository,c=main,b=i386
     origin deb.parrotsec.org
Pinned packages:

velichkov, 2018-09-17:
Ok, as suspected you are running 3.11 and not 4.1 as you've told me in https://github.com/ptrkrysik/gr-gsm/issues/432#issuecomment-418711702 . You could try to upgrading your system `sudo apt full-upgrade` and risk breaking `grgsm_scanner` or keep it as is.

duvaldt, 2019-01-04:
I have the same problem here, using Ubuntu and have tried with both the compiled version and distributed packages.

I can tune in on radio stations usin gqrx, but I have never seen anything using grgsm_scanner or grgsm_livemon. When grgsm_livemon starts, I get no output in the terminal. 

**Output from grgsm_scanner looks like this:** 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Args= 

**Output from rtl_test looks like this:**
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...

Any ideas?

duvaldt, 2019-01-04:
When I start gqrx I see this: 
mattias@mdsw apps (master) $ gqrx 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/mattias/.config/gqrx/bookmarks.csv
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Found Rafael Micro R820T tuner
getDeviceList : Available input devices:
    0 : "Realtek RTL2838UHIDIR SN: 00000001"
    1 : "RFSPACE SDR-IQ Receiver"
    2 : "RFSPACE SDR-IP Receiver"
    3 : "RFSPACE NetSDR Receiver"
    4 : "RFSPACE Cloud-IQ Receiver"
    5 : "hw:HDA Intel PCH,0"
    6 : "default"
    7 : "Generic RTL2832U OEM :: 00000001"
    8 : "RTL-SDR Spectrum Server"
    9 : "Red Pitaya Transceiver Server"
    10 : "Complex Sampled (IQ) File"

If I use device 7 instead of 0 nothing will work, but if I use 7 I can listen to the radio. When I use grgsm_scanner, it says it's using device 0, and a "Generic RTL... device". Unable to specify anything else though.

velichkov, 2019-01-07:
Hi @duvaldt,

> getDeviceList : Available input devices:
> 0 : "Realtek RTL2838UHIDIR SN: 00000001"
> 7 : "Generic RTL2832U OEM :: 00000001"

How many RTL SDR devices do you have connected to your system - one or two?

I'm not familiar with `gqrx` project but it seems [they are using `gr-osmosdr`](https://github.com/csete/gqrx/blob/master/src/qtgui/ioconfig.cpp#L237-L242) as well.

When running `grgsm_scanner` try to specify the rtl device in the `--args` parameter. Try the following
```
grgsm_scanner --args=rtl=0
grgsm_scanner --args=rtl=1
grgsm_scanner --args=rtl=7
```
If you have multiple RTL devices you could try to change their serial numbers as well with `rtl_eeprom -d 0 -s 00000002`
 
https://ptrkrysik.github.io/#capturing-gsm-downlink-and-uplink-with-multi-rtl
https://osmocom.org/projects/gr-osmosdr/wiki#RTL-SDR-Source

duvaldt, 2019-01-07:
Hi,

I only have one device connected, that is why it's so strange =/

I've tried changing it to something else but nothing other than 0 will work.

Could it be possible that there are some problems with the hardware? I can
listen to the radio but I never find any GSM-stations.

Best regards,
Mattias

velichkov, 2019-01-07:
> I only have one device connected, that is why it's so strange =/

Yes, it's really strange and I don't know why gqrx reports two devices.

> I've tried changing it to something else but nothing other than 0 will work.

Could you upload somewhere (google drive, dropbox, pastbing ...) the whole debug output when you start it with 0 and 1.

> I can listen to the radio but I never find any GSM-stations.  

Have you tried a different band? Maybe there is no any GSM cells in the default GSM900 band where you live.
```
-b BAND, --band=BAND  Specify the GSM band for the frequency. Available
                        bands are: GSM900, DCS1800, GSM850, PCS1900, GSM450,
                        GSM480, GSM-R
```

duvaldt, 2019-01-07:
Thanks! I'll give it a try =)

Do you know if it's possible to get debug messages with the prebuild
distribution package?

velichkov, 2019-01-07:
> Do you know if it's possible to get debug messages with the prebuild distribution package?

No, I've added the `--debug` flag recently and it's still not available in the packages. You have to comment out the following two lines in `/usr/bin/grgsm_scanner`
```
320             os.dup2(null_fds[0], 1)
321             os.dup2(null_fds[1], 2)
```

velichkov, 2019-01-07:
When you start `gqrx` is shows the following dialog box where you need to select a device

[image]

and when you select one the `Device string` is changed automatically.

What is the `Device string` when you select device `#7` and what is when you select `#0`?

duvaldt, 2019-01-07:
Good idea!
Device 0:
available=Yes,driver=rtlsdr,manufacturer=Realtek,product=RTL2838UHIDIR,rtl=0,serial=00000001,soapy=2,tuner='Rafael
Micro R820T'
Device 7: rtl=0

Thanks again!

velichkov, 2019-01-07:
> soapy=2

Most probably the device is found twice because it's detected by both soapy (device `#0`) and directly by the `rtl-sdr` (device `#7`). In my opinion `grgsm_sacnner --args=rtl=0` should work so try scanning all bands and always wait until it finishes as it usually takes some time. You could also try scanning with [kalibrate-rtl](https://github.com/steve-m/kalibrate-rtl)

How did you install gr-osmosdr and which Ubuntu version do you use?

duvaldt, 2019-01-07:
rtl0 - https://pastebin.com/QknqM7pj
rtl1 - https://pastebin.com/3K2tF06z
rtl7 - https://pastebin.com/Qbmg4Xjz

apt-get install gr-osmosdr

Ubuntu 18.04.1 LTS
4.15.0-43-generic

Thanks, I'll give it a try again!

duvaldt, 2019-01-07:
Does this give you any clues?
https://pastebin.com/5VAtC3Hj

velichkov, 2019-01-08:
> rtl0 - https://pastebin.com/QknqM7pj

Everything looks good here.

> rtl1 - https://pastebin.com/3K2tF06z
> rtl7 - https://pastebin.com/Qbmg4Xjz

>> FATAL: Wrong rtlsdr device index given.

This error is more then expected as as you have only one RTL dongle.

> Does this give you any clues? https://pastebin.com/5VAtC3Hj

>> [R82XX] No valid PLL values for 1807525000 Hz!
>> r82xx_set_freq: failed=-1

I don't have much experience with the RTL devices but this doesn't look good. It fails to tune to the requested frequency, maybe these dongles do not support 1800MHz.

Have you tried to search with kalibrate-rtl?

Does your dongle support bias tee? There was [a report](https://groups.google.com/forum/#!searchin/gr-gsm/bias%7Csort:date/gr-gsm/Xfs8jEcmYaI/JChJOv3YAQAJ) that it has to be disabled.

milankubin, 2019-12-08:
Interesting, I'm running into the same problems with exactly the same dongle.
The dongle is certainly limited.
kalibrate gives me at least a result :)

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
kal: Scanning for GSM-850 base stations.
GSM-850:
    chan:  231 (889.8MHz - 8.467kHz)    power:   25694.55
^C.chan 244

velichkov, 2019-12-09:
Hi @milankubin,

> Interesting, I'm running into the same problems with exactly the same dongle.

What is the exact problem you are having as at least 3 problems are being discussed in this issue?
Do you have a suitable antenna attached to the device?
What OS do you use and which version?
How did you install gnuradio and gr-gsm and which versions?

> kalibrate gives me at least a result :)

kalibrate-rtl is known to report false positives.

> kal: Scanning for GSM-850 base stations.

Where do you live?

The `GSM-850` band is used in North America and in USA there is no longer GSM network in this band as AT&T [stopped their 2G network](https://about.att.com/innovationblog/2g_sunset) back in 2017.

For the higher GSM bands (1800 and 1900 MHz) you need RTL dongle with Elonics tuner like [this](https://www.nooelec.com/store/sdr/sdr-receivers/nesdr-xtr-rtl2832u-e4000.html) one or another device that supports these frequencies.

botherder, 2020-07-01:
Sorry for reviving this ticket, but I stumbled upon it search for the issue I'm having.

```
Args= 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 434, in <module>
    main()
  File "/usr/local/bin/grgsm_scanner", line 431, in main
    options.ppm, options.gain, options.args, prn = printfunc, debug = options.debug)
  File "/usr/local/bin/grgsm_scanner", line 329, in do_scan
    ppm=ppm, gain=gain, args=args)
  File "/usr/local/bin/grgsm_scanner", line 212, in __init__
    str(grgsm.device.get_default_args(args)))
  File "/usr/local/lib/python2.7/dist-packages/grgsm/device.py", line 55, in get_default_args
    raise RuntimeError("Unable to find any supported SDR devices")
RuntimeError: Unable to find any supported SDR devices
```

grgsm_scanner does not seem to be able to identify a supported sdr, but it lists the following:

```
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

label='RFSPACE SDR-IQ Receiver',sdr-iq=/dev/ttyUSB0
label='RFSPACE SDR-IP Receiver',sdr-ip=127.0.0.1:50000
label='RFSPACE NetSDR Receiver',netsdr=127.0.0.1:50000
cloudiq=127.0.0.1:50000,label='RFSPACE Cloud-IQ Receiver'
label='RTL-SDR Spectrum Server',rtl_tcp=localhost:1234
label='Red Pitaya Transceiver Server',redpitaya=192.168.1.100:1001
file=/path/to/your/file,freq=100e6,label='Complex Sampled (IQ) File',rate=1e6,repeat=true,throttle=true
```
I am testing one of these:
https://www.amazon.de/NooElec-NESDR-Smart-RTL2832U-R820T2-basierte/dp/B01HA642SW/

If I try to specify any rtl index, it doesn't seem to help.

cesare3338, 2020-11-02:
> Ci scusiamo per aver rianimato questo biglietto, ma mi sono imbattuto nella ricerca del problema che sto riscontrando.
> 
> ```
> Args= 
> [INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
> 
> RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.
> 
> Traceback (most recent call last):
>   File "/usr/local/bin/grgsm_scanner", line 434, in <module>
>     main()
>   File "/usr/local/bin/grgsm_scanner", line 431, in main
>     options.ppm, options.gain, options.args, prn = printfunc, debug = options.debug)
>   File "/usr/local/bin/grgsm_scanner", line 329, in do_scan
>     ppm=ppm, gain=gain, args=args)
>   File "/usr/local/bin/grgsm_scanner", line 212, in __init__
>     str(grgsm.device.get_default_args(args)))
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/device.py", line 55, in get_default_args
>     raise RuntimeError("Unable to find any supported SDR devices")
> RuntimeError: Unable to find any supported SDR devices
> ```
> 
> grgsm_scanner non sembra essere in grado di identificare un sdr supportato, ma elenca quanto segue:
> 
> ```
> [INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
> 
> RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.
> 
> label='RFSPACE SDR-IQ Receiver',sdr-iq=/dev/ttyUSB0
> label='RFSPACE SDR-IP Receiver',sdr-ip=127.0.0.1:50000
> label='RFSPACE NetSDR Receiver',netsdr=127.0.0.1:50000
> cloudiq=127.0.0.1:50000,label='RFSPACE Cloud-IQ Receiver'
> label='RTL-SDR Spectrum Server',rtl_tcp=localhost:1234
> label='Red Pitaya Transceiver Server',redpitaya=192.168.1.100:1001
> file=/path/to/your/file,freq=100e6,label='Complex Sampled (IQ) File',rate=1e6,repeat=true,throttle=true
> ```
> 
> Sto testando uno di questi:
> https://www.amazon.de/NooElec-NESDR-Smart-RTL2832U-R820T2-basierte/dp/B01HA642SW/
> 
> Se provo a specificare un indice rtl, non sembra aiutare.
HAI RISOLTO ?

ptrkrysik, 2026-11-09:
The reports in this thread were hardware/environment related (e.g. R820T dongles cannot tune 1800 MHz, no SDR device found, outdated Parrot package); please open a new issue with the current release and the output of grgsm_scanner --debug if you still see problems.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik