ptrkrysik/gr-gsm #435: osr hard coded in grgsm_livemon

baddemanax, 2018-11-21:
hello,

I was trying to get grgsm-livemon running on a raspberry PI, but as mentioned in other post it did buffer over run quite often.
So I tried to decreased the OSR as explained in other post to a value of 2 , with no effect.
So I found out that OSR value was hardcoded to 4 in "GSM input adaptor" as in "GSM Receiver" blocks.
After the changes I got better performances on the Pi3, but it over runs again after a while.
So I tried with osr = 1 ( so no oversampling) but the amount of data decoded is really lower than with a higher oversampling rate, so missing a lot of datas. (:
So I wonder how it would be possible to improve performances on the Pi ( which has a 4 cores processor)
As I am new to this field, could you provide me with some hints where start and maybe what to try... 

my config is :
Rafael Micro R820T tuner
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
 GNU C++ version 6.2.0 20161010
raspbian stretch kernel 4.14.80-v7+

thanks in advance

velichkov, 2018-11-22:
Hi @baddemanax,

> I was trying to get grgsm-livemon running on a raspberry PI

Which model?

> , but as mentioned in other post it did buffer over run quite often.
>
> So I tried to decreased the OSR as explained in other post to a value of 2 , with no effect.
> So I found out that OSR value was hardcoded to 4 in "GSM input adaptor" as in "GSM Receiver" blocks.

Yes, it's hardcoded in these two blocks and this have to be fixed, thanks for reporting it.
>
> After the changes I got better performances on the Pi3, but it over runs again after a while.
> So I tried with osr = 1 ( so no oversampling) but the amount of data decoded is really lower than with a higher oversampling rate, so missing a lot of datas. (:
That's to be expected
>
> So I wonder how it would be possible to improve performances on the Pi ( which has a 4 cores processor)

Have you tried with a smaller sampling rate: 1M or 0.8M? Smaller value probably won't work because of the used offset which you could try to remove as well, see below.

Have you tried grgsm_livemon_headless and if yes does it keep up with the load?
>
> As I am new to this field, could you provide me with some hints where start and maybe what to try...

If any of the above does not help you will eventually need to profile the grgsm code in order to find parts that consumes a lot of CPU and then try to optimize them. The fact that part of the code is python and you're running it on RPi that currently handle the load makes the task a lot more difficult.

A good articular about profiling tools and RPi is https://www.softwarecoven.com/software-profiling-tools-for-raspberry-pi/
>
> my config is :
> Rafael Micro R820T tuner

If your RTL dongle does not have a (too big) DC spike on the center frequency, you could try to remove the Rotator block and set the shiftoff variable to 0.
>
> gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
> GNU C++ version 6.2.0 20161010
> raspbian stretch kernel 4.14.80-v7+

How did you install grgsm on RPi? Have you followed the steps from our [wiki](https://github.com/ptrkrysik/gr-gsm/wiki/Installation-on-RaspberryPi-3) and is it up to date?

baddemanax, 2018-11-22:
Hi ,

Thanks for the fast answer.
I am using Rpi 3 B+
I tried also a lower sample rate, The minimum allowed is 900001 Hz, lower one gives error. (maybe from the driver, I still need to investigate why)
I am using the headless version, as I am missing a screen for the Rpi.
I installed stretch Lite on the Pi and followed the wiki. There was just a small issue with Gnutls missing to compile libosmocore. So I added the package :
 
`sudo apt-get install libneon27-gnutls-dev`

To run the grgsm code you also need to specify  --args rtl, otherwise you get error message, compare to desktop version where this is not needed.

I will give a try with your proposal ( check Dc and trying to remove the Rotator block and profile the code on the pi and come back to you)
So far I just found that the OSR function was using only 1 cpu core on the Pi and it couldn't handle it properly leading to buffer over run. Setting OSR to 1 seemed to work for the pi, but comparing with same setup on my main computer I see a huge difference in the data processed : 

on the pi I run 
`grgsm_livemon_headless --args rtl -f 926.6M -g 40 -s 1M --osr 1`
I see only a few data packets once and a while

on the desktop i run
`grgsm_livemon_headless -g 40 -f 926.6M` 
 I see a constant flow of datas

I have 2 rtl dongles ( same brand) placed close to each other, So I am expecting to record same datas.

I also tried on the Rpi with 
`grgsm_livemon_headless --args rtl -f 926.6M -g 40 -s 1M --osr 2`
Then I could see a huge improvement in the datas received but I got few buffer over runs and after a while it hangs, and only showed buffer over run like

```
O 2d 06 3f 10 0c c3 d6 73 dc 47 00 00 c0 0e e4 43 2b 2b 2b 2b 2b 2b 2b
 55 06 19 8d eb 0d f5 af fc 42 ff ff 00 00 00 00 00 00 00 b9 00 00 2b
 2d 06 3f 10 0c c3 d6 73 dc 47 00 00 c0 0e e4 43 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 41 06 21 00 05 f4 10 ea 2a 33 17 05 f4 15 e3 de 29 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
O 49 06 1b 81 5a 02 f6 02 05 17 c8 02 28 14 65 45 b9 00 00 80 00 40 1b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 2d 06 3f 13 0c c3 d6 7f dd e9 00 00 42 31 30 34 40 51 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
O 25 06 21 00 05 f4 35 d6 b9 d3 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
```
I will search further and let you know

Zarados, 2019-03-13:
@velichkov @baddemanax 
When I run grgsm_scanner on a raspberry Pi, the result always less than I run this script on the Kali.
(I set the process and memory as same as raspberry pi, In fact the result of pi usually is zero.....)
Did you have any idea?

velichkov, 2019-03-13:
Hi @Zarados,

> Did you have any idea?

The RPi is not powerful enough to run grgsm_scanner and grgsm_livemon, someone needs to optimize the code. See slso [Compiler optimisation for RPi 3.](https://groups.google.com/d/topic/gr-gsm/FcUi-vg7ZBI/discussion)

jsa91, 2019-11-03:
Been running some test regarding this subject with a Tinker Board. If I try running gr-gsm in headless mode without any flags I get a serious amount of overflow. But with these setting things are acctually running pretty smooth:

` grgsm_livemon_headless --args rtl -f 953.0M -g 30 -s 1.0M --osr 3 -o 0`

Even if the data received still lags from time to time, overflow is reduced to almost zero.

What is the lowest sample rate recommended to use before losing to much information from the channel?

ptrkrysik, 2026-11-09:
OSR is now a configurable parameter of the receiver and input blocks, so the hardcoded value was fixed; Raspberry Pi performance is limited by the hardware, please open a new issue if you still see buffer overruns with the current release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik