ptrkrysik/gr-gsm #437: grgsm for raspberry pi 3B+ stretch

bferrell, 2018-12-11:
Using the install proceedure for raspberry pi 3 on a 3B+ I find I get 

corrupted double-linked list

On all the graphic utilities.

Any thoughts?

velichkov, 2018-12-11:
Hi @bferrell,

See https://github.com/ptrkrysik/gr-gsm/issues/435#issuecomment-440987611. Do you specify the `--args ` parameter when you run the graphic utilities or not? And what SDR device do you use?

ptrkrysik, 2026-11-09:
This was reported against an old Raspbian/GNU Radio 3.7 setup and the requested details were never provided; please open a new issue with the current release if it still occurs.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik