ptrkrysik/gr-gsm #441: grgsm_capture arguments

prkrmx, 2019-01-02:
Hello
I have two receivers in my system name=RX_1 and name=RX_2
I found that I cant launch grgsm_capture for the selected receiver
Every time the RX_1 receiver starts working
grgsm_capture -a 89 --args="name=RX_2" -T 10

GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.12.0.0-59-g523994c7
gr-osmosdr v0.1.4-99-gc98be5dd (0.1.5git) gnuradio 3.7.12.0
ubuntu 16

thanks

velichkov, 2019-01-02:
> product: B205mini

Try adding `uhd,` before the `name` - e.g. `grgsm_capture -a 89 --args="uhd,name=RX_2" -T 10`
See also https://osmocom.org/projects/gr-osmosdr/wiki#UHD-Source-Sink

prkrmx, 2019-01-02:
Thanks for the quick response  @velichkov 
It works as expected

velichkov, 2019-01-02:
> Thanks for the quick response

You are welcome. 

> It works as expected

Could you please close the issue?

closed 2019-01-02 by prkrmx