ptrkrysik/gr-gsm #442: grgsm_livemon Error: Address already in use

chungtra, 2019-01-09:
After executing grgsm_livemon command, on all frequencies, I'm receiving this error. 
Somebody please guide to resolve the issue.

grgsm_livemon -f 946.4M
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx2_64_mmx_orc
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 270, in <module>
    tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
  File "/usr/bin/grgsm_livemon", line 157, in __init__
    self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", "127.0.0.1", "4729", 10000)
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 1042, in make
    return _blocks_swig5.socket_pdu_make(type, addr, port, MTU, tcp_no_delay)
RuntimeError: bind: Address already in use

velichkov, 2019-01-09:
Hi @chungtra,

> self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", "127.0.0.1", "4729", 10000)
> RuntimeError: bind: Address already in use

Most probably you have started another instance of grgsm_livemon, grgsm_livemon_headless or grgsm_decode and that's why it can't open and bind to the server port. You could find the other process by executing `sudo netstat -upln | grep 4729` and then you need to stop it.

chungtra, 2019-01-10:
this resolved my issue. Thanks 👍

closed 2019-01-10 by chungtra