ptrkrysik/gr-gsm #443: Having problem when installing pybombs install gr-gsm

ImSrabon, 2019-01-22:
sudo pybombs install gr-gsm
[INFO] Prefix Python version is: 2.7.15
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] Executing command with elevated privileges: `sudo -H pip2 install PythonQwt'
Requirement already satisfied: PythonQwt in /usr/local/lib/python2.7/dist-packages (0.5.5)
Requirement already satisfied: NumPy>=1.3 in /usr/lib/python2.7/dist-packages (from PythonQwt) (1.15.4)
Install tree:
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- airspy
   |  |
   |  +- hackrf
   |  |
   |  +- uhd
   |  |
   |  +- soapysdr
   |  |
   |  +- gnuradio
   |  |  |
   |  |  +- uhd
   |  |  |
   |  |  +- apache-thrift
   |  |  |
   |  |  \- pyqwt5
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  +- libosmo-dsp
   |  |  |
   |  |  \- gnuradio
   |  |     |
   |  |     +- uhd
   |  |     |
   |  |     +- apache-thrift
   |  |     |
   |  |     \- pyqwt5
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   \- gnuradio
      |
      +- uhd
      |
      +- apache-thrift
      |
      \- pyqwt5
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: pyqwt5
[INFO] Install python-apt to speed up apt processing.
[WARNING] Build dir already exists: /usr/local/src/pyqwt5/configure
Configuring: (100%) [===================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
Command line options:
{'debug': False,
 'disable_numarray': False,
 'disable_numeric': False,
 'disable_numpy': False,
 'excluded_features': [],
 'extra_cflags': [],
 'extra_cxxflags': [],
 'extra_defines': [],
 'extra_include_dirs': ['/usr/local/include/qwt5/'],
 'extra_lflags': [],
 'extra_lib_dirs': ['/usr/local/lib/', '/usr/local/lib64'],
 'extra_libs': [],
 'jobs': '',
 'module_install_path': '/usr/local/lib/python2.7/site-packages/',
 'modules': [],
 'qt': 4,
 'qwt_sources': '../qwt-5.2',
 'sip_include_dirs': [],
 'subdirs': [],
 'timelines': [],
 'trace': ''}

Found SIP-4.19.13.
Found 'posix' operating system:
2.7.15+ (default, Nov 28 2018, 16:27:22) 
[GCC 8.2.0]
Do not get upset by error messages in the next 3 compiler checks:
Check if 'size_t' and 'unsigned int' are the same type:
size_t_check.cpp:3:6: error: no declaration matches ‘void a::f(unsigned int)’
 void a::f(unsigned int) {};
      ^
size_t_check.cpp:2:24: note: candidate is: ‘void a::f(size_t)’
 class a { public: void f(size_t); };
                        ^
size_t_check.cpp:2:7: note: ‘class a’ defined here
 class a { public: void f(size_t); };
       ^
NO
Check if 'size_t' and 'unsigned long' are the same type:
YES
Check if 'size_t' and 'unsigned long long' are the same type:
size_t_check.cpp:3:6: error: no declaration matches ‘void a::f(long long unsigned int)’
 void a::f(unsigned long long) {};
      ^
size_t_check.cpp:2:24: note: candidate is: ‘void a::f(size_t)’
 class a { public: void f(size_t); };
                        ^
size_t_check.cpp:2:7: note: ‘class a’ defined here
 class a { public: void f(size_t); };
       ^
NO
Failed to import numarray: PyQwt will be build without support for numarray.

Failed to find Numeric2: PyQwt will be build without support for Numeric.

Found NumPy-1.15.4.

g++ -c -pipe -O2 -Wall -W -D_REENTRANT -DQT_NO_DEBUG -DQT_CORE_LIB -DQT_SHARED -I../../../../share/qt4/mkspecs/linux-g++ -I. -I../../../../include/qt4/QtCore -I../../../../include/qt4 -I../qwt-5.2/src -I/usr/local/include/qwt5/ -I../../../../include/python2.7 -I../../../../lib/python2.7/dist-packages/numpy/core/include -I. -o qwt_version_info.o qwt_version_info.cpp
g++ -Wl,-O1 -o qwt_version_info qwt_version_info.o    -L/usr/lib/x86_64-linux-gnu -lQtCore -lpthread 
Found Qwt-5.2.1.

Extended command line options:
{'debug': False,
 'disable_numarray': False,
 'disable_numeric': False,
 'disable_numpy': False,
 'excluded_features': ['-x HAS_PYQT_031105',
                       '-x HAS_NUMARRAY',
                       '-x HAS_NUMERIC',
                       '-x HAS_QWT4'],
 'extra_cflags': [],
 'extra_cxxflags': [],
 'extra_defines': ['HAS_NUMPY'],
 'extra_include_dirs': ['/usr/local/include/qwt5/',
                        '/usr/include/python2.7',
                        '/usr/lib/python2.7/dist-packages/numpy/core/include'],
 'extra_lflags': [],
 'extra_lib_dirs': ['/usr/local/lib/', '/usr/local/lib64'],
 'extra_libs': [],
 'iqt': 'iqt5qt4',
 'iqt_sipfile': '../sip/iqt5qt4/IQtModule.sip',
 'jobs': '',
 'module_install_path': '/usr/local/lib/python2.7/site-packages/',
 'modules': ['iqt', 'Qwt5'],
 'qt': 4,
 'qwt': 'qwt5qt4',
 'qwt_sipfile': '../sip/qwt5qt4/QwtModule.sip',
 'qwt_sources': '../qwt-5.2',
 'sip_include_dirs': [],
 'subdirs': ['iqt5qt4', 'qwt5qt4'],
 'timelines': ['-t Qwt_5_2_0'],
 'trace': ''}

The following modules will be built: ['iqt', 'Qwt5'].

Setup the PyQwt package build.
sip invokation:
'/usr/bin/sip -b tmp-iqt5qt4/iqt5qt4.sbf -c tmp-iqt5qt4   ../sip/iqt5qt4/IQtModule.sip'
sh: 1: /usr/bin/sip: not found
SIP failed to generate the C++ code.
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyqwt5:
Configuration failed
[ERROR] Error installing package pyqwt5. Aborting.

velichkov, 2019-01-22:
Hi @ImSrabon,

> sudo pybombs install gr-gsm
[INFO] Prefix Python version is: 2.7.15
[INFO] PyBOMBS Version 2.3.4a0

In my opinion the `pybombs` installation method is the most difficult and error prone. Have you tried the other installation methods described [here](https://osmocom.org/projects/gr-gsm/wiki/Installation).? 

It looks you are using Debian or Ubuntu or another distribution based on them so I recommend you to uninstall everything installed by `pybombs`, then uninstall `pybombs` and try either 
* [Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804](https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804)
* [Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)

> Setup the PyQwt package build.
sip invokation:
'/usr/bin/sip -b tmp-iqt5qt4/iqt5qt4.sbf -c tmp-iqt5qt4 ../sip/iqt5qt4/IQtModule.sip'
sh: 1: /usr/bin/sip: not found

In your case installation of PyQwt fails because `/usr/bin/sip` is not found  and here is how to find which package provides it and then install it.
```
$ sudo apt-get install apt-file
$ sudo apt-file update
$ apt-file search "/usr/bin/sip"
sip-dev: /usr/bin/sip
$ sudo apt-get install sip-dev
```
If you choose to use `pybombs` then it's good to report this issue in the [gr-recipes](https://github.com/gnuradio/gr-recipes/) repository as well

marwon34, 2019-01-29:
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyqwt5:
Configuration failed
[ERROR] Error installing package pyqwt5. Aborting.

velichkov, 2019-01-29:
Hi @marwon34 

Please read my previous message https://github.com/ptrkrysik/gr-gsm/issues/443#issuecomment-456466771

In my opinion the `pybombs` installation method is the most difficult and error prone. Have you tried the other installation methods described [here](https://osmocom.org/projects/gr-gsm/wiki/Installation).? 

If you choose to use `pybombs` then report this issue in the [gr-recipes](https://github.com/gnuradio/gr-recipes/) repository.

marwon34, 2019-01-29:
When I try to install packages As described Les show this 
make: * No targets specified and no makefile found.  Stop.

velichkov, 2019-01-29:
> When I try to install packages As described Les show this

Could you be more specific and list all commands (including all arguments) that you've executed and the their whole output.

Also which Linux Distribution do you use and which version?

marwon34, 2019-01-29:
root@kali:~/gr-gsm/build# make
make: * No targets specified and no makefile found.  Stop.
I use  Linux

velichkov, 2019-01-29:
> I use Linux
> root@kali

From your host name it looks that your are using Kali Linux but which version?

> GnuRadio Runtime required to compile gr-gsm
> -- Configuring incomplete, errors occurred!

It seems you have not installed all dependencies. Please follow the exact instructions from
https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition and run

```
sudo apt-get update && \
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev
```
> root@kali:# git clone https://git.osmocom.org/gr-gsm
fatal: destination path 'gr-gsm' already exists and is not an empty directory.
root@kali:# cd gr-gsm
root@kali:/gr-gsm# mkdir build
mkdir: cannot create directory ‘build’: File exists

You don't have to create folders and clone the repository multiple times, do it only once.

Have you removed everything installed by `pybombs` and `pybombs` itself?

marwon34, 2019-01-29:
I use Kali Linux 2019.1
Can I communicate with you? I am Marwan from Libya

velichkov, 2019-01-29:
> I use Kali Linux 2019.1

Then the easiest way to install `gr-gsm` is from the package repository by running  `sudo apt-get install gr-gsm` See https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804

marwon34, 2019-01-29:
pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default
[INFO] Prefix Python version is: 2.7.15
[INFO] PyBOMBS Version 2.3.4a0
[WARNING] There already is a prefix in `/root/pybombs/bladeRF'.
Continue using this path Y/[N]? y
[INFO] Prefix Python version is: 2.7.15
Alias `bladeRF' already exists, overwrite Y/[N]? y
[INFO] Prefix Python version is: 2.7.15
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] Executing command with elevated privileges: `sudo -H pip2 install PythonQwt'
Requirement already satisfied: PythonQwt in /usr/local/lib/python2.7/dist-packages (0.5.5)
Requirement already satisfied: NumPy>=1.3 in /usr/lib/python2.7/dist-packages (from PythonQwt) (1.15.4)
Install tree:
|
\- gnuradio
   |
   +- uhd
   |
   +- apache-thrift
   |
   \- pyqwt5
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: pyqwt5
[INFO] Install python-apt to speed up apt processing.
[WARNING] Build dir already exists: /root/pybombs/bladeRF/src/pyqwt5/configure
Configuring: (100%) [=========================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
Command line options:
{'debug': False,
 'disable_numarray': False,
 'disable_numeric': False,
 'disable_numpy': False,
 'excluded_features': [],
 'extra_cflags': [],
 'extra_cxxflags': [],
 'extra_defines': [],
 'extra_include_dirs': ['/root/pybombs/bladeRF/include/qwt5/'],
 'extra_lflags': [],
 'extra_lib_dirs': ['/root/pybombs/bladeRF/lib/',
                    '/root/pybombs/bladeRF/lib64'],
 'extra_libs': [],
 'jobs': '',
 'module_install_path': '/root/pybombs/bladeRF/lib/python2.7/site-packages/',
 'modules': [],
 'qt': 4,
 'qwt_sources': '../qwt-5.2',
 'sip_include_dirs': [],
 'subdirs': [],
 'timelines': [],
 'trace': ''}

Found SIP-4.19.13.
Found 'posix' operating system:
2.7.15+ (default, Nov 28 2018, 16:27:22) 
[GCC 8.2.0]
Do not get upset by error messages in the next 3 compiler checks:
Check if 'size_t' and 'unsigned int' are the same type:
size_t_check.cpp:3:6: error: no declaration matches ‘void a::f(unsigned int)’
 void a::f(unsigned int) {};
      ^
size_t_check.cpp:2:24: note: candidate is: ‘void a::f(size_t)’
 class a { public: void f(size_t); };
                        ^
size_t_check.cpp:2:7: note: ‘class a’ defined here
 class a { public: void f(size_t); };
       ^
NO
Check if 'size_t' and 'unsigned long' are the same type:
YES
Check if 'size_t' and 'unsigned long long' are the same type:
size_t_check.cpp:3:6: error: no declaration matches ‘void a::f(long long unsigned int)’
 void a::f(unsigned long long) {};
      ^
size_t_check.cpp:2:24: note: candidate is: ‘void a::f(size_t)’
 class a { public: void f(size_t); };
                        ^
size_t_check.cpp:2:7: note: ‘class a’ defined here
 class a { public: void f(size_t); };
       ^
NO
Failed to import numarray: PyQwt will be build without support for numarray.

Failed to find Numeric2: PyQwt will be build without support for Numeric.

Found NumPy-1.15.4.

g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_CORE_LIB -I. -I../qwt-5.2/src -I/root/pybombs/bladeRF/include/qwt5/ -isystem /usr/include/python2.7 -I/usr/lib/python2.7/dist-packages/numpy/core/include -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o qwt_version_info.o qwt_version_info.cpp
g++ -Wl,-O1 -o qwt_version_info qwt_version_info.o   -lQt5Core -lpthread 
Found Qwt-5.2.1.

Extended command line options:
{'debug': False,
 'disable_numarray': False,
 'disable_numeric': False,
 'disable_numpy': False,
 'excluded_features': ['-x HAS_PYQT_031105',
                       '-x HAS_NUMARRAY',
                       '-x HAS_NUMERIC',
                       '-x HAS_QWT4'],
 'extra_cflags': [],
 'extra_cxxflags': [],
 'extra_defines': ['HAS_NUMPY'],
 'extra_include_dirs': ['/root/pybombs/bladeRF/include/qwt5/',
                        '/usr/include/python2.7',
                        '/usr/lib/python2.7/dist-packages/numpy/core/include'],
 'extra_lflags': [],
 'extra_lib_dirs': ['/root/pybombs/bladeRF/lib/',
                    '/root/pybombs/bladeRF/lib64'],
 'extra_libs': [],
 'iqt': 'iqt5qt4',
 'iqt_sipfile': '../sip/iqt5qt4/IQtModule.sip',
 'jobs': '',
 'module_install_path': '/root/pybombs/bladeRF/lib/python2.7/site-packages/',
 'modules': ['iqt', 'Qwt5'],
 'qt': 4,
 'qwt': 'qwt5qt4',
 'qwt_sipfile': '../sip/qwt5qt4/QwtModule.sip',
 'qwt_sources': '../qwt-5.2',
 'sip_include_dirs': [],
 'subdirs': ['iqt5qt4', 'qwt5qt4'],
 'timelines': ['-t Qwt_5_2_0'],
 'trace': ''}

The following modules will be built: ['iqt', 'Qwt5'].

Setup the PyQwt package build.
sip invokation:
'/usr/bin/sip -b tmp-iqt5qt4/iqt5qt4.sbf -c tmp-iqt5qt4   ../sip/iqt5qt4/IQtModule.sip'
sip: Deprecation warning: ../sip/iqt5qt4/IQtModule.sip:32: %Module version number should be specified using the 'version' argument
0 file(s) lazily copied.

Setup the PyQwt package build.
sip invokation:
'/usr/bin/sip -I /usr/share/sip/PyQt4 -b tmp-qwt5qt4/qwt5qt4.sbf -c tmp-qwt5qt4   -x VendorID -t WS_X11 -x PyQt_NoPrintRangeBug -t Qt_4_8_6 -x Py_v3 -g -x HAS_PYQT_031105 -x HAS_NUMARRAY -x HAS_NUMERIC -x HAS_QWT4 -t Qwt_5_2_0 ../sip/qwt5qt4/QwtModule.sip'
sip: Deprecation warning: ../sip/qwt5qt4/QwtModule.sip:32: %Module version number should be specified using the 'version' argument
sip: Unable to find file "QtCore/QtCoremod.sip"
SIP failed to generate the C++ code.
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyqwt5:
Configuration failed
[ERROR] Error installing package pyqwt5. Aborting.
root@kali:~#

marwon34, 2019-01-29:
I want to solve this problem The use of BladeRF 2.0 micro

velichkov, 2019-01-30:
> sip: Unable to find file "QtCore/QtCoremod.sip"

Try installing the package that provides the missing file: `sudo apt-get install python-qt4-dev`

ptrkrysik, 2019-08-23:
This is topic for the mailing list

ptrkrysik, 2019-08-23:
Please go there

closed 2019-08-23 by ptrkrysik