ptrkrysik/gr-gsm #445: Internal Compiler Error

arifkyi, 2019-02-07:
Hello ,

Need your kindly help, i'm experiencing like this after run make and/or make -j $nproc
of course i tried PyBOMBS but not worked.

Looking forward for the resolution of problem below.

[ 87%] Built target pygen_python_trx_a57ee
[ 87%] Built target pygen_python_receiver_4f558
[ 89%] Built target pygen_swig_33fc6
[ 89%] Built target man_grgsm_livemon.rst
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
swig/CMakeFiles/_grgsm_swig.dir/build.make:68: recipe for target 'swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o] Error 4
CMakeFiles/Makefile2:664: recipe for target 'swig/CMakeFiles/_grgsm_swig.dir/all' failed
make[1]: *** [swig/CMakeFiles/_grgsm_swig.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

velichkov, 2019-02-07:
Hi @arifkyi,

> c++: internal compiler error: Killed (program cc1plus)
> file:///usr/share/doc/gcc-5/README.Bugs for instructions.

This seems like a compile bug and the gcc 5 is a quite old version.
What OS and version do you use? 
Try upgrading the OS or compiler.

arifkyi, 2019-02-21:
Hello Vasil,

Thank you so much for your response:

Im using Ubuntu 16.04  ,
however the problem is resolved by using this command below that i just aware there is easier way to install with Ubuntu, instead using the PyBOMBS

Install grgsm  For older Debian and Ubuntu :

sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm
sudo apt update
sudo apt install gr-gsm

closed 2019-02-21 by arifkyi