ptrkrysik/gr-gsm #446: grgsm_livemon not working

tekelok, 2019-02-08:
hello,
operation Ubuntu 18.04.1 LTS
card rtl

grgsm_livemon -f 938.2M
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.14.0.0-125-g4d658fc5
Using device #0 Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 361, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 349, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 101, in __init__
    self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + args )
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1170, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: Failed to open rtlsdr device.

velichkov, 2019-02-08:
Hi @tekelok,

The output you've posted contains a suggestion how to fix it.

> Using device #0 Generic RTL2832U OEM
> usb_open error -3
> Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

Have you done this? How did you install `rtl-sdr`- manually or from the repository?

closed 2019-08-23 by ptrkrysik

darkhacknet, 2022-05-22:
> Hi @tekelok,
>
> The output you've posted contains a suggestion how to fix it.
>
>
> Have you done this? How did you install `rtl-sdr`- manually or from the repository?

Hey my friend could you please help me with a little issue with livemon command ? the issue is that the command are not displaying any info. Basically never sniff packets ! Im using hackrf and dragonOS.

thanks in advance !