ptrkrysik/gr-gsm #447: I have problem in grgsm_livemon

UrubuCarniceiro, 2019-02-19:


velichkov, 2019-02-19:
Hi @UrubuCarniceiro,

Which OS are you using and which version?
How did you install gnuradio - from repository, using pybombs or manual?
Which `gr-gsm` installation method are you following? See https://osmocom.org/projects/gr-gsm/wiki/Installation

velichkov, 2019-02-19:
Also provide the full output of `make VERBOSE=1` executed from the `build` directory.

UrubuCarniceiro, 2019-02-21:
> Hi @UrubuCarniceiro,
> 
> Which OS are you using and which version?
> How did you install gnuradio - from repository, using pybombs or manual?
> Which `gr-gsm` installation method are you following? See https://osmocom.org/projects/gr-gsm/wiki/Installation

Hi velichkov 
I'm using Parrot sec and in Parrot GNUradio is a native program, I did not installed, but i made the upgrade all over programs.

UrubuCarniceiro, 2019-02-21:
> make VERBOSE=1

(/gsm/gr-gsm-master/build) I go to the folders and use the command (make VERBOSE=1) but still not work he continue goes to 90% and  the answer is :
[ 90%] Generating grgsm_livemon
cd /gsm/gr-gsm-master/build/apps && /usr/bin/cmake -E env PYTHONPATH="/gsm/gr-gsm-master/python:/gsm/gr-gsm-master/python/misc_utils:/gsm/gr-gsm-master/python/demapping:/gsm/gr-gsm-master/python/receiver:/gsm/gr-gsm-master/python/transmitter:/gsm/gr-gsm-master/python/trx:/gsm/gr-gsm-master/build/swig" GRC_BLOCKS_PATH=/gsm/gr-gsm-master/grc /usr/local/bin/grcc -d /gsm/gr-gsm-master/build/apps /gsm/gr-gsm-master/apps/grgsm_livemon.grc
Segmentation fault
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[2]: Leaving directory '/gsm/gr-gsm-master/build'
make[1]: *** [CMakeFiles/Makefile2:1323: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make[1]: Leaving directory '/gsm/gr-gsm-master/build'
make: *** [Makefile:141: all] Error 2

UrubuCarniceiro, 2019-02-21:
And I did use the method that you advised me (https://osmocom.org/projects/gr-gsm/wiki/Installation) but the same problem happened.

[ 87%] Generating __init__.pyc, udp_link.pyc, ctrl_if.pyc, ctrl_if_bb.pyc, radio_if.pyc, radio_if_uhd.pyc, radio_if_lms.pyc, radio_if_grc.pyc, transceiver.pyc, dict_toggle_sign.pyc
[ 89%] Generating __init__.pyo, udp_link.pyo, ctrl_if.pyo, ctrl_if_bb.pyo, radio_if.pyo, radio_if_uhd.pyo, radio_if_lms.pyo, radio_if_grc.pyo, transceiver.pyo, dict_toggle_sign.pyo
[ 89%] Built target pygen_python_trx_39b58
Scanning dependencies of target pygen_apps
[ 90%] Generating grgsm_livemon
Segmentation fault
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:1323: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

velichkov, 2019-02-25:
Hi @UrubuCarniceiro,

> I'm using Parrot sec and yes I used pybombs and in Parrot, GNUradio is a native program,  I did not installed, but I made the upgrade all over programs.

The PyBOMBS always installs gnuradio from source and most probably you ended with two versions installed on your system - one from the parrot's repository in `/usr/bin/` and a second one in `/usr/local/bin` installed by PyBOMBS. Having two versions could be source of many problems so I recommend you to remove the version installed by PyBOMBS (and everything installed by it), remove the PyBOMBS, remove the `build` directory and start from scratch.

```
pybombs inv
pybombs remove apache-thrift uhd gnuradio # add the names of other packages listed by `pybombs inv`
pip uninstall pybombs
cd /gsm/gr-gsm-master/
rm -rf build
mkdir build
cd build
cmake ..
make
```

Parrot is based on Debian testing so you could follow https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

> And I'm using (https://www.youtube.com/watch?v=ZGVwRxwhv-c&index=5&list=PLRovDyowOn5F_TFotx0n8A79ToZYD2lOv). sorry to use that video, I know that is not the most correct form to learn to do this but I liked the way he taught

I haven't watched the whole video but I think it's a good tutorial except the recommendation to not build from the `master` branch.

> (/gsm/gr-gsm-master/build) I go to the folders and use the command (make VERBOSE=1) but still not work he continue goes to 90% and the answer is :

The `make VERBOSE=1` is not suppose to fix the problem but only to give us some more information, like the exact commands with all parameters that are being used.

> ... /usr/local/bin/grcc -d /gsm/gr-gsm-master/build/apps /gsm/gr-gsm-master/apps/grgsm_livemon.grc
Segmentation fault

The `grcc` in `/usr/local/bin` is being used, it's installed by PyBOMBS and it's failing with Segmentation fault. If you are c++ programmer you could try to debug it with `gdb` or you could try reporting this in the gnuradio mailing list.

velichkov, 2019-02-25:
> Parrot is based on Debian testing so you could follow https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

And also because it's based on debian testing you could install gr-gsm from the parrot's apt repository `sudo apt-get install gr-gsm`

ciphercc, 2019-05-14:
i m trying to install gr-gsm in kali linux with pybombs install gr-gsm this command during the installation process i m facing this issue can any on help how to resolve this

[ 29%] Generating python docstrings for blocks_swig10_doc
Traceback (most recent call last):
  File "/usr/local/src/gnuradio/docs/doxygen/swig_doc.py", line 328, in <module>
    make_swig_interface_file(di, swigdocfilename, custom_output=custom_output)
  File "/usr/local/src/gnuradio/docs/doxygen/swig_doc.py", line 251, in make_swig_interface_file
    blocks = di.in_category(Block)
  File "/usr/local/src/gnuradio/docs/doxygen/doxyxml/base.py", line 140, in in_category
    self.confirm_no_error()
  File "/usr/local/src/gnuradio/docs/doxygen/doxyxml/base.py", line 206, in confirm_no_error
    self.check_parsed()
  File "/usr/local/src/gnuradio/docs/doxygen/doxyxml/base.py", line 203, in check_parsed
    self._parse()
  File "/usr/local/src/gnuradio/docs/doxygen/doxyxml/doxyindex.py", line 43, in _parse
    self._root = index.parse(os.path.join(self._xml_path, 'index.xml'))
  File "/usr/local/src/gnuradio/docs/doxygen/doxyxml/generated/index.py", line 71, in parse
    doc = minidom.parse(inFilename)
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 1918, in parse
    return expatbuilder.parse(file)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 924, in parse
    result = builder.parseFile(fp)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 211, in parseFile
    parser.Parse("", True)
xml.parsers.expat.ExpatError: no element found: line 1, column 0
make[2]: *** [gr-blocks/swig/CMakeFiles/blocks_swig10_swig_doc.dir/build.make:200: gr-blocks/swig/blocks_swig10_doc.i] Error 1
make[1]: *** [CMakeFiles/Makefile2:2757: gr-blocks/swig/CMakeFiles/blocks_swig10_swig_doc.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

velichkov, 2019-05-15:
Hi @ciphercc,

> i m trying to install gr-gsm in kali linux with pybombs install gr-gsm this command during the installation process i m facing this issue can any on help how to resolve this

I hate to say it but installing with pybombs is the most difficult and error prone method. My advice is to first try installing gr-gsm using the method describe [here](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)

> [ 29%] Generating python docstrings for blocks_swig10_doc
Traceback (most recent call last):
File "/usr/local/src/gnuradio/docs/doxygen/swig_doc.py", line 328, in 

It fails while building gnuradio. Better report this error in the [gnuradio mailing list](https://wiki.gnuradio.org/index.php/MailingLists) or in the [pybombs issues](https://github.com/gnuradio/pybombs/issues)

ciphercc, 2019-05-15:
> Hi @ciphercc,
>
>
> I hate to say it but installing with pybombs is the most difficult and error prone method. My advice is to first try installing gr-gsm using the method describe [...]

thanks Velichkov   now its work

ciphercc, 2019-05-15:
how to find correct GSM frequency for india this there any data available or any tools for that

velichkov, 2019-05-15:
> how to find correct GSM frequency for india this there any data available

Try searching with google for "India gsm bands" or "India gsm frequencies" 

> or any tools for that

Try `grgsm_scanner` with all supported bands - `-b GSM900`, `-b DCS1800`, ....

chandu-77, 2019-11-10:
If you are using a old low specific hardware then don't use pybomb to build the program. The master branch perfectly works without any errors in Parrot OS code name jolly Roger , and result is unbelievable . Also the pybomb build is works great in Ubuntu distribution but try to use manual installation in other than Ubuntu . 
You can see my results here 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[Second web](http://www.thejobnotice.com) 

You can visit my website at
[First website](https://governmentjobs.politics.blog)

chandu-77, 2019-11-10:
Don't clone from osmocomm . Clone from this repository

chandu-77, 2019-11-10:
Actually don't follow the YouTube videos blindly they will mislead you. As you can see I was able to see the clear text SMS , also able to catch the type 0 SMS , but unfortunately that screenshots not showing  here

wilmirram, 2020-07-10:
> Na verdade, não siga os vídeos do YouTube cegamente, pois eles o enganarão. Como você pode ver, eu consegui ver o SMS em texto não criptografado, também consegui capturar o SMS do tipo 0, mas infelizmente as capturas de tela não são exibidas aqui

@chandu-77 
I am almost 2 months trying and always with some error.
Can you help?
Who knows, we can exchange emails until we adjust.
If it doesn't work out, I'll quit because I've run out of options.

douglasvilelavilar@gmail.com

ptrkrysik, 2026-11-09:
This was an environment problem (GNU Radio/pybombs build on Parrot/Kali) with the old build system; please follow the current build instructions for GNU Radio 3.10 and open a new issue if the build fails.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.