ptrkrysik/gr-gsm #448: grcc grgsm_livemon_headless.grc failed

arifkyi, 2019-02-21:
Hello Dev team and friends,

[grgsm_livemon_headless.txt](https://github.com/ptrkrysik/gr-gsm/files/2888613/grgsm_livemon_headless.txt)

i would like to raise case:

**what:**  'grcc grgsm_livemon_headless.grc' yield the error
**What error:** see below in details
**file Grcc:** attached ( i renamed it from .grc to .txt to be able to attached)
**Os :** ubuntu 16.04.2

Error:

~/gr-gsm/apps$ grcc grgsm_livemon_headless.grc 
>>> Error: Connection between gsm_clock_offset_control_0(ctrl) and gsm_input_0(ctrl_in) could not be made.
	source key 'ctrl' not in source block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 409, in import_data
    source_port = verify_and_get_port(source_key, source_block, 'source')
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 390, in verify_and_get_port
    raise LookupError('%s key %r not in %s block keys' % (dir, key, dir))
LookupError: source key 'ctrl' not in source block keys
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Validation failed:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
	Param - Cell allocation(cell_allocation):
		Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
		downlink2arfcn() takes exactly 2 arguments (1 given)
Error during file compilation.

velichkov, 2019-02-25:
Hi @arifkyi,

From https://github.com/ptrkrysik/gr-gsm/issues/445#issuecomment-465925837

> Im using Ubuntu 16.04 , 
> however the problem is resolved by using this command below that i just aware there is easier way to install with Ubuntu, instead using the PyBOMBS 
> 
> Install grgsm For older Debian and Ubuntu :
>
> sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm
> sudo apt update
> sudo apt install gr-gsm

The version you have installed from the @ptrkrysik's ppa repository is quite old and it seems it's not compatible with the `gnuradio` installed on your system. I do not recommend using this ppa repository and I do not recommend PyBOMBS either. 

My recommended method for 16.04 is [Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)

We are using Docker to test installation on different Linux distributions, the docker file is [Ubuntu_16_04.docker](https://github.com/ptrkrysik/gr-gsm/blob/master/tests/dockerfiles/Ubuntu_16_04.docker) and I just tested it successfully. 

```
cd gr-gsm
docker build -f tests/dockerfiles/Ubuntu_16_04.docker .
```

arifkyi, 2019-02-27:
Hello Velichkov,

Thank you so much we can close the issue,
 i  delete and install Ubuntu 16.04 LTS, so install in clean Ubuntu
and did 100% as your recommendation from this link [https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition] and found to be working 100% for the grgsm_livemon_headless.

Thank you very much again Velichkov for your expert support.

closed 2019-02-27 by arifkyi