ptrkrysik/gr-gsm #450: grgsm_decode gnuradio issue decoding voice traffic

prkrmx, 2019-03-11:
Hello
I know that this is a stupid question, but I have not found a solution.
this is a known problem: when we decode voice traffic the grgsm_decode does not stop.
we have already discussed this issue https://github.com/ptrkrysik/gr-gsm/issues/422 . it was in 2016. This was a known gnuradio issue and it is fixed in gnuradio/gnuradio/pull/797 
but I still have this problem. all components were installed thru grgsm one month ago
what should I do?  is it fixed by gnuradio or not?
thanks for the help and understanding

GNU Radio Companion 3.7.9
Ubuntu 16.04.5 LTS
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

velichkov, 2019-03-11:
Hi @prkrmx,

> this is a known problem: when we decode voice traffic the grgsm_decode does not stop.
> we have already discussed this issue #422 . it was in 2016.

#422 is from July 2018 and not 2016.

> This was a known gnuradio issue and it is fixed in [gnuradio/gnuradio/pull/797](https://github.com/gnuradio/gnuradio/pull/797)

It was fixed and merged into gnuradio's `next` branch back on Jul 1, 2016, but a stable gnuradio version that contains this fix has not been released yet. They have switched to development in the `master` branch since then so it's available in their `master` as well. The fix will be available in the upcoming `3.8` major release. 

> but I still have this problem. all components were installed thru grgsm 

What do you mean by "installed thru grgsm"?

> one month ago

It doesn't matter when you installed it but how and from where.

> is it fixed by gnuradio or not?

Yes, it is.

> GNU Radio Companion 3.7.9
> Ubuntu 16.04.5 LTS
> linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Most probably the `3.8` version won't be released for `16.04 LTS` in the official Ubuntu's repositories ever.

> what should I do? 

One possible solution is to install `gnuradio` from its `master` branch - you could compile it manually (and then manually recompile all other packages that uses gnuradio) or try installing it using `pybombs`. 

If you choose the `pybombs` way, first try installing it in a VM or container as you could messed up your system quite easily and also try installing it not in `/usr/local/` but in a different prefix (`$HOME/gnuradio-3.8` for example), see https://github.com/gnuradio/pybombs#prefixes

P.S.
More appropriate place for such questions is the [gr-gsm's malling list](https://github.com/ptrkrysik/gr-gsm/#mailing-list)
Also consider posting `gnuradio` and `pybombs` related questions to the [gnuradio's mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio)

prkrmx, 2019-03-12:
hi @velichkov 
Thank you very much for your answer

> What do you mean by "installed thru grgsm"?

i mean, all the prerequisites have been installed from [here ](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition) and I thought that gnuradio from this list has already received this patch.

any way, I will try to compile `gnuradio` manually, because sometimes `pybombs` it’s a very painful installation.
thanks

closed 2019-03-12 by prkrmx