ptrkrysik/gr-gsm #451: grgsm_decode decoding TCHH

prkrmx, 2019-03-11:
Hello All
when I run the grgsm_decode through the TCHH , I get these errors. Is this something critical for decoding speech?
all components were installed thru grgsm one month ago
thanks

GNU Radio Companion 3.7.9
Ubuntu 16.04.5 LTS
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

grgsm_decode -b GSM900_93_953600000_1000000.bursts.bz2 -k f853654d039c9b61 -t 2 --sub-channel=1 --multi-rate=20151737d0 -m TCHH  -a 93 -o test.au.gsm

multi_rate configuration: 20151737d0
Enabled AMR Codecs:
4,75 kbit/s codec rate: is part of the subset
6,90 kbit/s codec rate: is part of the subset
7,40 kbit/s codec rate: is part of the subset
Error! frame_nr:360848 mod26:20 fn_is_odd:1 length:-1
Error! frame_nr:360852 mod26:24 fn_is_odd:0 length:-1
Error! frame_nr:360857 mod26:3 fn_is_odd:1 length:-1
Error! frame_nr:360861 mod26:7 fn_is_odd:0 length:-1
Error! frame_nr:360865 mod26:11 fn_is_odd:1 length:3
Error! frame_nr:360870 mod26:16 fn_is_odd:0 length:3
Error! frame_nr:360874 mod26:20 fn_is_odd:1 length:-1
Error! frame_nr:360878 mod26:24 fn_is_odd:0 length:-1
Error! frame_nr:360883 mod26:3 fn_is_odd:1 length:3
Error! frame_nr:360887 mod26:7 fn_is_odd:0 length:-1
Error! frame_nr:360891 mod26:11 fn_is_odd:1 length:-1
Error! frame_nr:360896 mod26:16 fn_is_odd:0 length:-1
Error! frame_nr:360900 mod26:20 fn_is_odd:1 length:-1
Error! frame_nr:360904 mod26:24 fn_is_odd:0 length:-1
Error! frame_nr:360909 mod26:3 fn_is_odd:1 length:-1
Error! frame_nr:360913 mod26:7 fn_is_odd:0 length:-1
Error! frame_nr:360917 mod26:11 fn_is_odd:1 length:-1
Error! frame_nr:360922 mod26:16 fn_is_odd:0 length:-1
Error! frame_nr:360926 mod26:20 fn_is_odd:1 length:3
Error! frame_nr:360930 mod26:24 fn_is_odd:0 length:-1
Error! frame_nr:360935 mod26:3 fn_is_odd:1 length:-1
Error! frame_nr:360939 mod26:7 fn_is_odd:0 length:-1
Error! frame_nr:360943 mod26:11 fn_is_odd:1 length:-1
Error! frame_nr:360948 mod26:16 fn_is_odd:0 length:-1
Error! frame_nr:360952 mod26:20 fn_is_odd:1 length:-1
Error! frame_nr:360956 mod26:24 fn_is_odd:0 length:-1
Error! frame_nr:360961 mod26:3 fn_is_odd:1 length:-1
Error! frame_nr:360965 mod26:7 fn_is_odd:0 length:-1
Error! frame_nr:360969 mod26:11 fn_is_odd:1 length:3
Error! frame_nr:360974 mod26:16 fn_is_odd:0 length:-1
...

velichkov, 2019-03-11:
Hi @prkrmx,

> when I run the grgsm_decode through the TCHH , I get these errors. Is this something critical for decoding speech?
> all components were installed thru grgsm one month ago

Again like in #450, what do you mean by "installed thru grgsm" ?

> grgsm_decode -b GSM900_93_953600000_1000000.bursts.bz2 -k f853654d039c9b61 -t 2 --sub-channel=1 --multi-rate=20151737d0 -m TCHH -a 93 -o test.au.gsm
>
> multi_rate configuration: 20151737d0
> Enabled AMR [...]

If the output file is not empty you could safely ignore these errors.

You could try adding the `--voice-boundary` option but also make sure that your capture contains the begging of the call as when this option is specified grgsm_deocde expects one of the Alerting, Progress, Connect or Connect Acknowledge messages before starting to decode voice frames. Without this option it tries to decodes all frames on the specified timeslot including the frames before the call had started and after it had ended and when it fails it prints this error. Note that event with this option it prints few such errors.

```
    --voice-boundary    Enable voice boundary detection for traffic channels.
                        This can help reduce noice in the output.
```

prkrmx, 2019-03-12:
hi @velichkov 
thanks for the answer
Yes, when I set this option, errors disappear.
thanks

closed 2019-03-12 by prkrmx

phamduythai92mta, 2019-04-01:
Hi all. 
i have a question like that. where can i get the parameter (multi-rate).   in this case -multi-rate=20151737d0.
thanks in advance!

phamduythai92mta, 2019-04-02:
sorry! i am still a little bit confused.
this is my Assignment command. Where is the --multi-rate ?. and this is my command to try decoding. but i did not get the audio output file. 
grgsm_decode -s 1e6 -f 940.4M -c 940.4_s1e6_voice.cfile -m TCHH --sub-channel 1 -t 5 --multi-rate 0306201d
multi_rate configuration: 0306201d
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_decode", line 396, in <module>
    print_bursts=options.print_bursts, ppm=options.ppm)
  File "/usr/local/bin/grgsm_decode", line 107, in __init__
    self.tch_h_decoder = grgsm.tch_h_decoder(tch_h_channel, multi_rate, enable_voice_boundary_detection)
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 2052, in make
    return _grgsm_swig.tch_h_decoder_make(sub_channel, multi_rate, boundary_check)
RuntimeError: Multirate speech version

[image]

velichkov, 2019-04-02:
> multi_rate configuration: 0306201d

Try with `201d17371204`, you need to skip the first two bytes - the Element ID and Length

phamduythai92mta, 2019-04-02:
i have tried that. and the output file is about 6 byte and there is no sound. in addition the program does not stop. 
could you tell me what i was wrong?
thanks!

velichkov, 2019-04-02:
> in addition the program does not stop.

That's a known issue, see #450 and #422 

> could you tell me what i was wrong?

Could you share the cfile? Are you sure that there is no encryption?

phamduythai92mta, 2019-04-03:
that is my cfile.
yes. there is no encryption
https://drive.google.com/open?id=1HhTQgBZPNYDowK6fdtyOACrSHfmQk9gE

velichkov, 2019-04-03:
> https://drive.google.com/open?id=1HhTQgBZPNYDowK6fdtyOACrSHfmQk9gE

```
404. That’s an error.

The requested URL was not found on this server. That’s all we know.
```

nullvsi, 2023-02-13:
I have the same problem with decoding TCHH and getting audio file. Output file is 6 bytes lenght.

```CLI
sudo grgsm_decode -a34 -s1e6 -cvoice2.cfile -mBCCH -t0
```

[image]

```CLI
sudo grgsm_decode -a34 -s1e6 -cvoice2.cfile -mSDCCH8 -t1
```

[image]

```CLI
sudo grgsm_decode -a34 -s1e6 -cvoice2.cfile -mSDCCH8 -t1 -e3 -k'9DF1CFDF89141185'
```

[image]

```CLI
sudo grgsm_decode -a34 -s1e6 -cvoice2.cfile -mTCHH -t2 -e3 -k'9DF1CFDF89141185' -dFR -ovoice.au.gsm --sub-channel=1 -v --multi-rate=20151737d0
multi_rate configuration: 20151737d0
Enabled AMR Codecs:
4,75 kbit/s codec rate: is part of the subset
5,90 kbit/s codec rate: is part of the subset
7,40 kbit/s codec rate: is part of the subset
2013545 3109313:  01 73 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
2013623 3110177:  01 41 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
2013595 3109305:  05 01 03 03 2d 06 1e 00 d9 52 f5 60 34 83 93 fc 00 22 13 2b 2b 2b 2b
2013699 3109369:  05 01 03 03 49 06 1d 10 00 00 00 15 00 00 00 00 00 00 2a 82 00 00 00
2013907 3109497:  06 01 03 03 2d 06 1e 00 d9 52 f5 60 34 83 93 fc 00 22 13 2b 2b 2b 2b
2014307 3111273:  03 60 09 03 0f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
2014323 3111801:  09 01 03 03 49 06 1d 10 00 00 00 15 00 00 00 00 00 00 2a 82 00 00 00
2014427 3111865:  09 01 03 03 49 06 06 9f 00 34 00 00 00 00 00 00 00 06 40 00 00 00 00
2014531 3111929:  0a 01 03 03 2d 06 1e 00 d9 52 f5 60 34 83 93 fc 00 22 13 2b 2b 2b 2b
2014635 3111993:  0a 01 03 03 49 06 1d 10 00 00 00 15 00 00 00 00 00 00 2a 82 00 00 00
2014739 3112057:  0b 02 03 03 49 06 06 9f 00 34 00 00 00 00 00 00 00 06 40 00 00 00 00
2014947 3112185:  0b 02 03 03 49 06 1d 10 00 00 00 15 00 00 00 00 00 00 2a 82 00 00 00
2015051 3112249:  0b 02 03 03 49 06 06 9f 00 34 00 00 00 00 00 00 00 06 40 00 00 00 00
2015155 3112313:  0c 02 03 03 2d 06 1e 00 d9 52 f5 60 34 83 93 fc 00 22 13 2b 2b 2b 2b
2015259 3112377:  0c 01 03 03 49 06 1d 10 00 00 00 15 00 00 00 00 00 00 2a 82 00 00 00
2015363 3112441:  0c 01 03 03 49 06 06 9f 00 34 00 00 00 00 00 00 00 06 40 00 00 00 00
2015467 3112505:  0d 01 03 03 2d 06 1e 00 d9 52 f5 60 34 83 93 fc 00 22 13 2b 2b 2b 2b
2015720 3114482:  03 62 15 03 25 02 ea 90 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
2015675 3113049:  0d 01 03 03 49 06 06 9f 00 34 00 00 00 00 00 00 00 06 40 00 00 00 00
2015755 3113953:  01 81 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
2015772 3114514:  03 84 19 03 2a 08 02 ea 90 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
2015883 3113177:  0d 02 03 03 49 06 1d 10 00 00 00 15 00 00 00 00 00 00 2a 82 00 00 00
2015997 3113545:  03 86 0d 06 0d 00 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
```