ptrkrysik/gr-gsm #452: [ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libbladeRFSupport.so)

hakanonymos, 2019-03-14:
i have install all packages with 
sudo apt-get update && \
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev

git clone https://git.osmocom.org/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
make
sudo make install
sudo ldconfig

after installing

root@kali:~# grgsm_livemon 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libbladeRFSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libbladeRFSupport.so: undefined symbol: bladerf_set_gain_mode
[INFO] Using format CF32.
Erreur de segmentation
root@kali:~# 

how to resolve this probleme ?

velichkov, 2019-03-16:
Hi @hakanonymos,

What SDR device do you use?

> root@kali:# grgsm_livemon
> gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp [...]

Try upgrading or reinstalling the `libbladerf1` package
```
apt-get update
apt-get upgrade
apt-get install --reinstall libbladerf1
```
If the above does not help provide the full output of the following commands
```
find / -name libbladeRFSupport.so
ldd  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libbladeRFSupport.so
objdump -TC  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libbladeRFSupport.so | grep bladerf_set_gain_mode
objdump -TC /usr/lib/x86_64-linux-gnu/libbladeRF.so.1 | grep bladerf_set_gain_mode
```

ptrkrysik, 2019-08-21:
This is installation issue. If you would like to continue, do it on the mailing and point to this issue.

closed 2019-08-21 by ptrkrysik